luc-github/Repetier-Firmware-4-Davinci #116: Z Axis Offset

The-Experimentalist, 2015-10-15:
Hi

Now I am up and running with the firmware on the DaVinci AiO I need to calibrate the bed. After homing all axis the extruder nozzle is somewhere around 2.8mm below the bed. Is there an EEPROM setting I can configure to get this closer before using the screws to adjust the bed or do I need to lose all this using the adjustment screws?

Can I use 'Z home pos' for this?

I know this isn't really an issue but not sure where else to post questions specifically about low level firmware details?

luc-github, 2015-10-15:
yes this is the function of the screws, when homing is done,  this is normal process to do bed levelling

The-Experimentalist, 2015-10-15:
Thanks for the quick response. Just to be clear make the whole adjustment with the screws? Is this distance of approximately 3mm quite normal?

luc-github, 2015-10-15:
you mean nozzle is too far or too close ?
normal position for home is z=0

The-Experimentalist, 2015-10-15:
With all homed the nozzle is 3mm below the bed height. The bed retaining screw in that corner is very close to the extruder carriage but not touching. It is not a problem to use the screws I just wanted to understand if this is the best way. 

Is there a document anywhere that explains what each of the EEPROM settings is used for?

Thanks for your time.

luc-github, 2015-10-15:
I suggest you to read configuration.h every parameter is documented
no other doc also Davinci has more EEPROM settings than original repetier as I wanted to avoid recompilation for flexibility

luc-github, 2015-10-16:
If you feel good you can push a documentation for all parameters on wiki - I am sure several people will be interested :smile: 

About Z offset any other issue of we close topic ?

The-Experimentalist, 2015-10-16:
Sorry been away from the PC today so only just seen your message. Got my first print off last night so all good with the Z offset now. If I get some time and understand the parameters then I may try and look at publishing that on the wiki. Pretty flat out at the minute so that may not be soon. Thanks for your help so far, I really appreciate it.

luc-github, 2015-10-20:
Wiki is now up but empty - so ready to be feed - https://github.com/luc-github/Repetier-Firmware-0.92/wiki

luc-github, 2015-10-28:
OK I think issue can be closed now if no issue

closed 2015-10-28 by luc-github