luc-github/Repetier-Firmware-4-Davinci #118: Semi-Auto bed leveling

Zyphuris55, 2015-11-06:
The manual leveling wizard felt too slow (and I kept burning my hand). The Auto leveling was good, but with each test taking around 3 minutes, then guessing at how much to turn the knob so each test point is at 1.0 +/- 0.01 mm; the process ends up taking around 20 minutes.

I ended up making the semi-auto bed leveling wizard. It's in a rough state (menu text is hard coded in the functions still), but it's all there. I haven't done all the real-world testing, but there is a good chance it won't crash your print head. You'd most likely end up in a weird loop if anything, which power cycling the printer fixes. My printer hasn't fallen into any yet.

I didn't see where I could email the developer (Luc) the code...so I zipped it, uploaded it to my dropbox, and here is the link. https://dl.dropboxusercontent.com/u/65968034/SemiAutoMod.7z
The project is based on the 15-09-22.1 source.

The reason I am uploading it now in it's rough state still is that I will be leaving the state (US) for awhile and I can't take my printer with me. Hopefully the code goes to good use

luc-github, 2015-11-07:
I am in trip now and seems I cannot access to the content of your Dropbox - can you describe what your process is doing compare to manual and auto levelling ?
Thanks

Zyphuris55, 2015-11-10:
= Manual
- moves head to multiple spots, pauses, and lets user adjust print bed

= Auto
- moves to each of the 3 probing spots, probes, shows the results at the end (and asks if the user wants to save the adjustment)

= Semi-auto
- moves to each spot (or specific spot)
- probe a spot and stays in the probing menu till the user is done with that spot. Switches between needing the bed to move up/ down.
-- Moving bed up: a notification sound goes off when the bed is high enough to touch the probe
-- moving bed down: tells the user how much to rotate the knob and then press ok to re-test.
- shows final results and asks to save adjustments (all 3 spots must have been probed)
- quick probe option is available, which tests all the remaining spots then shows results

There is more detail in each steps, but that's the majority. By using the semi-auto, the calibration only needs to run once. I haven't timed how long a full (worst case) calibration is, but on auto (with adjusting the bed to 1.00) is at least 20 mins.

Here is a google drive link, https://drive.google.com/file/d/0BwcaftRIn4d2RF80N1hDbkRPMFU/
Both links have the file as private and in a 7-zip format.

luc-github, 2015-11-10:
Well Autolevel is not really accurate on Davinci because sensor position may not be the same between all Davinci  (it is just a screw) - there is some variable distance between metal pad and glass on each corner ,and the glass is not flat on all printers, so ideally should have some calibration of Autolevel distances before doing autolevel, but it is not easy.

Manual Level need some practice but you can get very good results - for me it take usually 2 rounds + 1 for verification, so around 2 min, every 2 month

doing calibration in one round means your bed is already nearly leveled as when you up(or down) one corner you also change a little the position of the 2 others in down(or up) direction.

I have a quick look at the code and have several interesting changes

limited660, 2016-01-21:
I don't see how you can just stop at each probing spot and work at it until you get a good number because you are changing the height of all locations with one adjustment due to the location of the adjustments on the bed versus where the head is probing the bed.

I can't find a good picture to show it but the first probing location (rear) is quite far from the rear adjustment screw which is nearly centered on the bed.  Both front probing locations are also quite far from their respective adjustment screws.  Any adjustment on the front screws will move the rear probing location quite a bit so you may end up chasing your tail unless some math can be added to the process.

luc-github, 2016-04-18:
Auto level has been deeply rewritten for 0.92.9 - I need to rewrite the interactive part now and add menu for new option to calculate the average Z offset 

So I close it - let's open again when new bed leveling is working

closed 2016-04-18 by luc-github