luc-github/Repetier-Firmware-4-Davinci #119: Arduino 1.6.6 is out and Due 1.6.6 is also out - need test

luc-github, 2015-11-07:
as both release are available - need to test to evaluate impact and if can work

celticht32, 2015-11-14:
for 1.6.6 of the arduino software the directories are :

USBCore.ccp = C:\Users\<user ID>\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\cores\arduino\USB
Variant.ccp    = C:\Users\<user ID>\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.6\variants\arduino_due_x

This is on windows 10..

celticht32, 2015-11-14:
when you compile however you get the following:
Arduino: 1.6.6 (Windows 10), Board: "Arduino/Genuino Uno"
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\chris\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\chris\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\chris\Documents\Arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path "C:\Users\chris\AppData\Local\Temp\buildce1dceec2a6587cd5dd12d87cc5b6e54.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Work\Repetier\Repetier-Firmware-0.92-master\src\ArduinoDUE\Repetier\Repetier.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\chris\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\chris\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\chris\Documents\Arduino\libraries" -fqbn=arduino:avr:uno -ide-version=10606 -build-path "C:\Users\chris\AppData\Local\Temp\buildce1dceec2a6587cd5dd12d87cc5b6e54.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Work\Repetier\Repetier-Firmware-0.92-master\src\ArduinoDUE\Repetier\Repetier.ino"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\chris\AppData\Local\Temp\buildce1dceec2a6587cd5dd12d87cc5b6e54.tmp\sketch\Repetier.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI" "C:\Users\chris\AppData\Local\Temp\buildce1dceec2a6587cd5dd12d87cc5b6e54.tmp\sketch\Repetier.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI" "C:\Users\chris\AppData\Local\Temp\buildce1dceec2a6587cd5dd12d87cc5b6e54.tmp\sketch\Repetier.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI" "C:\Users\chris\AppData\Local\Temp\buildce1dceec2a6587cd5dd12d87cc5b6e54.tmp\sketch\Repetier.ino.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI" "C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\SPI.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI" "C:\Users\chris\AppData\Local\Temp\buildce1dceec2a6587cd5dd12d87cc5b6e54.tmp\sketch\Commands.cpp"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -M -MG -MP -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10606 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-I [... cut ...]
C:\Users\chris\AppData\Local\Temp\buildce1dceec2a6587cd5dd12d87cc5b6e54.tmp\sketch\Commands.cpp:2097:54: note: in expansion of macro 'READ'

                             if(EEPROM::busesensor && READ(FIL_SENSOR1_PIN))uid.executeAction(UI_ACTION_NO_FILAMENT,true);

                                                      ^

Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
exit status 1
'NVIC_EncodePriority' was not declared in this scope

luc-github, 2015-11-14:
You did not selected Due but Uno :
```Arduino: 1.6.6 (Windows 10), Board: "Arduino/Genuino Uno"```
also please do not post all the output but eventually a link to file containing the output, this is unreadable

celticht32, 2015-11-14:
ok will do that going forward.. sorry =(

celticht32, 2015-11-14:
ok Selected due in the board... 
copied the two files  to the above directories.
compiled / verified the code.. and the attached error occured..
[error.txt](https://github.com/luc-github/Repetier-Firmware-0.92/files/34495/error.txt)

luc-github, 2015-11-14:
looks like old usbcore.cpp is not compatible with new module, may be variant.cpp neither
need to see the difference, 

for compilation test they are not necessary but variant.cpp is mandatory for usage, usbcore.cpp is just an additional safety if no watchdog enabled

celticht32, 2015-11-14:
well going through I am compiling and seeing that the following differs so far:
 bool USBD_ClassInterfaceRequest(Setup& setup) - Different definition
int USBD_SendInterfaces(void) - different definition
static bool USBD_SendDescriptor(Setup& setup) - different definition and definition has no parms..

Thats as far as I got...

luc-github, 2015-11-14:
you may give a try without modifying usbcore.cpp, it is a patch I did if you unplug usb when a connection is active, without this patch printer freeze waiting for usb reconnect.
but if watchdog is enabled it is not necessary, printer will freeze, then triger the watchdog then reset

celticht32, 2015-11-14:
also found that the setup structure name has changed to
USBSetup and think the structure has changed as well.. working through that now... will try and put the old usbcore and other back momentarily

celticht32, 2015-11-14:
ok it compiles without the modifications to usbcore.cpp and variant.cpp.
What were the changes in variant.cpp or were they part of the same changes in usbcore.cpp?

luc-github, 2015-11-14:
variant.cpp add the correct pins for davinci, if not used or adjusted you will have black bars and printer won 't work

celticht32, 2015-11-14:
ok put variant.cpp back and it compiled just fine...
no errors...
Guess the next thing is to flash it ..

celticht32, 2015-11-15:
Flashed and printing... only issue is with the usbcore.cpp file... Thanks

luc-github, 2015-11-16:
Thank you,
I have modified the usbcore.cpp accordingly but did not tested it yet
https://github.com/luc-github/Repetier-Firmware-0.92/commit/dbcacba033895478fccfb569fd648823c563821a

chrismidgley, 2016-01-12:
I have tried Arduino 1.6.7 / Due 1.6.6 with this latest firmware (Due 2.0a, Mac OS X, Simplify3D).  At first I had a bunch of USB problems (inability to connect or would connect and disconnect after a few messages) and I switched to the original usbcore.cpp and things appear to have gotten better, but I'm not 100% of my testing to know that the updated usbcore.cpp was faulty (I was doing all my tests remotely over a VPN so I wasn't sure what was really happening on the printer).  When I returned home, I noticed those occasional connect failures were associated with a firmware reboot (guessing usb lockup and watchdog?).   

With this configuration I have about 20 hours print time (about half ABS and half PLA) and so far with good results.  I did notice that extruder filament unload positioning was odd - it put the left extruder directly on the wiper on the left bin while heating (was worried the heat might melt that plastic so I didn't let it continue and did the filament unload by hand).  

Thanks for your efforts, luc!

cmidgley, 2016-01-13:
I have been experiencing problems with 1.6.7 / Due 1.6.6 and latest .sync firmware (see comment above for config).  Not sure where the problem is, but had prints failing part way through.  Firmware was just saying "wait" and not operating any more of the print, or was getting USB disconnects and unable to reconnect without power down/up.  Note that when using USBCore I couldn't maintain a USB connection worth anything, whereas with stock USBCore I was printing.  I did around 30 hours total of printing, and while smaller prints worked (1-2 hour), larger prints would often fail to finish (3-8 hour).

I went back to original 1.6.5 combo and things appear to be working again, but since it doesn't happen every time I'll need more printing time before gaining confidence that the issue was really firmware related vs. some new problem in my USB/computer/whatever setup.  

I'll give the new .sync firmware a shot using 1.6.5 in a bit (after maybe another 20-30 hours of print time to try to understand the stability of what I'm running now) - hopefully to help see if the problem might be in there, or in the 1.6.7/1.6.6 kit.

If you have any recommendations on what to try or what I should be looking at when a failure occurs please let me know.  I don't need the new firmware - just thought I could lend a hand at testing it.

luc-github, 2016-01-14:
Thanks for your feedback - usbcore.cpp patch is just avoiding to get stuck when usb connection is open and you unplug cable.
If you have issue with the patch when connected it means your USB connection is very bad and yes you will loose a lot of packet with the patch because retry number is not enough to recover.

currently I just tested 0.92.6-sync with 1.6.5 and short print 

Can you explain more :
>I did notice that extruder filament unload positioning was odd - it put the left extruder directly on the wiper on the left bin while heating 

Thanks for your help

cmidgley, 2016-01-14:
Re:usbcore, yes that matches what I had read, but thanks for confirming.

I'm totally willing to accept I have some issues with my hardware (I've ordered a new high quality USB cable to see if that happens to change anything) - just did find it interesting how drastically it changed in behavior based on firmware rev with no hardware change.  I'll do more testing across the firmware revisions and see if it is consistent/reproducible or not.

You can ignore the unload thing, especially as it relates to the new firmware.  I've always done unloads "by hand" and I used the menu for the first time in the new firmware.  Turns out it behaves the same in the prior firmware as well.  Basically it places the left extruder head directly on the wipe bar rather than off of it for a couple millimeters - which as it heats I was worried it may melt the plastic of the wipe bar just sitting there (Duo 2.0A).  Maybe like the other pull request for the 2.0A wiping this is another adjustment to make for 2.0A in that printers.cpp (or whatever that file is) ?  I'll take a look at that this weekend.

luc-github, 2016-01-26:
1.6.7 is now out for a while so this issue is no need anymore

Thanks for the feedback

closed 2016-01-26 by luc-github