luc-github/Repetier-Firmware-4-Davinci #120: Sync 0.92.6 version

luc-github, 2015-11-15:
0.92.6 is ready and now it is time to sync (starting this 16th of Nov) hope to done it quickly.
I have created a branch that will use travis-ci to compile all models of DAVINCI using Arduino 1.6.5 and DUE module 1.6.4 to have a global view - first commits will likely be failed but it will help to track the sync.

FYI the test/adjustment for Arduino 1.6.6 and DUE 1.6.6 will come after - no need to ask in advance, as first target is to make 0.92.6 working in an environment which is known to be working = IDE 1.6.5/DUE 1.6.4

Several bug fixes (better native usb support, autolevel,etc..) and enhancement ( watchdog, multi language at runtime, better support for Jam sensor) 
Will try to integrate fully the "wizard mode" and ensure it will work with Davinci as before it was only working with encoder and overlaped Davinci already existing feature like load/unload and out of filament sensor detection.

luc-github, 2015-11-20:
Translation part bring more work than expected - Repetier has 266 items for one language but Davinci has 409 items until now.... but I expect to add some more to cover everything, still some adjustement to be done to keep same UI
UK is done and will do FR and SP / IT later - but for DE/NL/SE/PT/CZ/PL I will put english wording for missing parts and disable them

Next step will be add the new menu entries and see how the overlap for X, Y, Z offsets
Next step will be the new coating entries integration
Next step will be wizard mode integration and see how it can live with existing filament detection and load/unload functions

I am on the 2 last files for sync but not the easiest ones uimenu.h and ui.cpp (in addition of uilang.h / uilang.cpp for all translations)

luc-github, 2015-11-24:
Sync is done for English - and can be flashed - light test seems Ok
Screen : Ok
Key pad : Ok
Heat bed and extruders : Ok  
Need to do some adjustment in menu as there are some typo mistake - 
Now will work on new menu items integration :
-  how the overlap for X, Y, Z offsets
- Integrate the new coating entries in menu
- integrate language selection (will add french first -  I did an excel file to help https://github.com/luc-github/Repetier-Firmware-0.92/blob/0.92.6-sync/Translate.xlsx)

Would be great to have test case check list to verify functionalities, but I have no time for this... 
So will test/adjust wizards (no filament/Jam) functions experimentally later

luc-github, 2015-11-24:
seems some important commit have been done after 0.92.6 like `Arduino IDE 1.6.6 compatible, STARTUP_GCODE`, so will do a new sync after to be sure to get all updates

Doudou14, 2015-12-01:
In french:
Salut Luc, ça faisait longtemps que je n'ai pas touché au firmware de la Davinci. Est-ce que la dernière version fonctionne bien ?

luc-github, 2015-12-01:
Salut Edouard,
la 0.92.6 est encore en test preliminaire, je suis a la bourre, j'espere pouvoir la passer en alpha d'ici ce week end
Je n'ai meme pas fait un test d'impression encore...

Doudou14, 2015-12-02:
D'accord pas de soucis, je n'ai plus le temps de travailler sur le firmware mais je suis toujours tes travaux :)

luc-github, 2015-12-08:
Got delay working on it - I am currently packing process to move to France and few free time for coding, sorry.
I had also to add several new "translated lines" for main screen and it is actually a lot of copy paste due to 10 languages even not translated...

I should be able to integrate all tomorrow to allow some pre-alpha tests - but leaving Friday for Xmas vacations so no access to printer

luc-github, 2015-12-09:
Translation / UI adjustment should be OK for UK/FR, I have only enabled spanish in addition of UK/FR
all others unfinished translated languages are disabled to limit memory usage and can be enabled anytime.
Filament / Door Sensors seems ok on my 2.0
I did a remote print using ESP8266 wifi module without filament and so far so good.
Emergency stop also is fine.
Still need to review:
1 -every menu item one by one to check if Ok 
2 - pause behavior during printing
3 - long print without filament

What I won't test before 2016 because lack of time :
1 - Jam sensor
2 - out of filament behavior during printing 
3 - auto bed leveling - my probe is not in correct position after I clean my nozzles - need to reposition it
4 - long print with filament

luc-github, 2015-12-30:
Doing a new sync as several fixes have been implemented upstream

luc-github, 2016-03-30:
I am late due to my moving that keep me busy more than expected -  now need 0.92.8 sync

luc-github, 2016-04-12:
I back on it - seems a lot of typo update and big rewrite of autoleveling part on 0.92.8 
so I will have to review carefully to rewrite and review the changes impact on menu function

luc-github, 2016-04-16:
first part of sync is done https://github.com/luc-github/Repetier-Firmware-0.92/commit/c0f7bde2e9e818d8cb96ad187fa798de7806d197
Autolevel is inactive in menu but working in GCODE - until now the code was duplicated one for GCODE and one for menu with more interactivity - with the big rewrite of the autolevel in GCODE part menu part is outdated - to avoid to maintain it I will try to split GCODE one in several part - easy to get original FW update and avoiding to duplicate it for menu version - will do next week 

Current code is compiling but not tested

luc-github, 2016-04-16:
will update travis-ci to reflect the usage of ide 1.6.8 and Due 1.6.7

luc-github, 2016-04-18:
0.92.9 is compiling and can be flashed - autolevel menu is disabled but other items looks fine

joelmeijering, 2016-05-22:
I checked out the 0.92.6-sync branch (which contains the 0.92.9 release of you're currently merging, as I understand).

Compilation on Arduino 1.6.5 (Due 1.6.4) went fine. Unfortunately, after uploading and restarting my printer, it only displayed black rectangles on the display (it always does this when starting up normally but with 0.92.9 it just kept showing the rectangles forever)

I flashed the old firmware back and now it is working again.

This was on DaVinci 1.0. I set the printer type in the configuration file before flashing.

Is there any way I can get some debugging information from the printer? I assume the printer encounters an error while initializing repetier.

luc-github, 2016-05-22:
usually black bar issue is due to bad variant.cpp or copy to wrong location

actually I use 1.6.9 ide and 1.6.8 due module now but should be ok for previous versions

when you said you use previous fw version , you compile it ? or you flash a binary file already compiled?

joelmeijering, 2016-05-22:
For both previous firmware and 0.92.9, I compiled the files with Arduino and then used the Upload button. 

For both the old and new firmware it succesfully compiles and uploads. However, with the new firmware, I get the black bars problem.

I will try with Arduino 1.9.6 and Due 1.6.9 and report back to you.

luc-github, 2016-05-22:
Ok this is weird -  thanks for your feedback

joelmeijering, 2016-05-22:
With Arduino 1.6.9 and Due 1.6.8 I get the same issue. I have tried both with and without the supplied variant.cpp and USBCore.cpp from the "Arduino - 1.6.8 -Due 1.6.7" folder.

There is no folder for 1.6.9 though, did you forget to add this to git, or are the custom files not needed for Arduino 1.6.9?

I also tried disabling setting EEPROM_MODE to 0 (this disables eeprom) in configuration.h to make sure old config values weren't interfering with the newer version of the firmware. Still no luck.

Ill flash it back to the old firmware now.

luc-github, 2016-05-23:
That is my mistake - I will push the variant.cpp /usbcore.cpp today when back home - sorry

joelmeijering, 2016-05-23:
Ok, great.

I don't think it matters much though. Today I flashed back the old firmware and I used the new arduino (1.6.9) version with the "wrong" custom USBCore.cpp etc. files from a different version (1.6.8 I believe). It's working fine. So I don't think it's what's causing the black bars on the new firmware. But I'll still try it with the right USBCore.cpp, just to be sure.

luc-github, 2016-05-23:
I pushed the files - https://github.com/luc-github/Repetier-Firmware-0.92/commit/7406a9607ca93cf0133ea525d5c825436f1490fb

no big change with previous

luc-github, 2016-06-05:
updated with 0.92.10 and added Zmin calculation menu - still autolevel menu do be done

luc-github, 2016-06-07:
Autolevel is implemented as well as several fixes -  changing to alpha - ready for test

luc-github, 2016-06-07:
https://github.com/luc-github/Repetier-Firmware-0.92/tree/0.92.6-sync will be merge to master when several feedback will be received

kryts, 2016-07-04:
any one else having issue with the load and unload in this version?
I haven't been able to get these to work in this version. The hot end moves to the load/unload position and heats up but then never loads or unloads. I can cancel the loading or unloading but then no commands that are sent to the printer do anything.

I can load/unload  fine by using the manual extrude controls

luc-github, 2016-07-04:
Just tested and seems fine on my printer - is your filament sensor on or off ?

luc-github, 2016-07-04:
Did you do a M502/M500 after flashing ? now load/unload speed and distance are in EEPROM for better modification

kryts, 2016-07-04:
my filament sensor is off I don't have the board attached any more
yes I did do the save . I did it from the menue instead of g command

luc-github, 2016-07-04:
You do not have the message to press ok when temperature is reached ?

kryts, 2016-07-05:
I do but then nothing happens

luc-github, 2016-07-05:
can you export your EEPROM settings?

gamecompiler, 2016-07-07:
The firmware compiles fine under Linux, 
the JAM Sensors works great, saved my extruder.
When Homeing the Z-Axis, the message will not dissapear as long a new one will be written.
The Homeing of the X and Y Axis works better than before ("smoother")
The menu reacts faster while printing ? (Im not sure)

luc-github, 2016-07-07:
Thanks for sharing ^_^ and you are lucky - german translation should be OK thanks @SebastianKnaus

kryts, 2016-07-07:
sure. sorry have been busy 

[Current aio 7 7 16.txt](https://github.com/luc-github/Repetier-Firmware-0.92/files/353132/Current.aio.7.7.16.txt)

kryts, 2016-07-07:
looking over the file I notice that there is a load unload distance and for some reason that is set at 0 I think that might be the issue.

luc-github, 2016-07-07:
yes 
```
<epr pos="1179" type="3" value="0.000">Load Feed Rate:</epr>
<epr pos="1183" type="3" value="0.000">Unload Feed Rate:</epr>
<epr pos="1187" type="3" value="0.000">Unload/Load Distance:</epr>
```
these values should not be 0
should be
```
//Davinci specific
// Loading / Unloading Filament value
#define UI_SET_PRESET_LOADING_FEEDRATE  2 
#define UI_SET_PRESET_UNLOADING_FEEDRATE  4
#define UI_SET_PRESET_UNLOAD_LOAD_DISTANCE  60 
```

luc-github, 2016-07-07:
normaly M502/M500 load and save these values to EEPROM

gamecompiler, 2016-08-17:
You could have wrote in the Readme that the EEPROM settings are now stored to sdcard.
I was very confused why Zmin is always moving back to firmwares default ones. Even when saving to eeprom.

Id wonder half hours why my printer is homeing to -22mm (my firmware settings) without SDCard connected ^^.
Nice feature, but if not known - very confusing Oo!

luc-github, 2016-08-17:
It is not new, It has always been like this - Davinci board does not have EEPROM, EEPROM is actually a virtual EEPROM file stored on SDCard

juri74, 2016-08-23:
Hello again!
i think i did a sufficient to good traslation to italian, it's very difficult to put 10 letter words in 3-4 letter space XD
i think there are some point where the new user (the one never used a 3d printer) could have some problems, but regular user shouldn't have problems at all.
i do not know how to pass the file, i do not have Microsoft Office but Openoffice, i had zipped the ODS file but Github do have a bug still telling me:
"Unfortunately, we don’t support that file type. Try again with a PNG, GIF, JPG, DOCX, PPTX, XLSX, TXT, PDF, or ZIP."
even if i try to send a ZIP file!!

Edit: i found a workaround, save the attached PNG file then rename it's extension to ZIP :)
[image]

luc-github, 2016-08-23:
Hi thanks a lot
I have a question: in french I changed On/Off by Yes/No because also short and meaning is same- what about italian ? Ok = Si , Off = No

juri74, 2016-08-23:
hello, in italian "on" and "off" can be leaved as is because is a common use, however the exact italian translation would be On = Acceso , Off = Spento

luc-github, 2016-08-23:
Ok got it - thanks a lot - I will integrate it this week

luc-github, 2018-01-27:
Since repetier is now in 1.X I think better to close this issue

closed 2018-01-27 by luc-github