luc-github/Repetier-Firmware-4-Davinci #121: Documentation request: dual extruder offset calibration

zendes, 2015-11-18:
In your readme, step 12, you mention "Next you can calibrate your filament as usual, and second extruder offset if you have."

I've tried finding the correct procedure for calibrating the second extruder offset, but nothing useful popped up. Can you point me to a wiki page/add a snippet on how to calibrate this step?

luc-github, 2015-11-18:
I did not find any but use some models for offset  calibration 
[image]

I used the dual extrusion models from thingiverse :
http://www.thingiverse.com/thing:21773
and
http://www.thingiverse.com/thing:12069

you have several others like http://www.thingiverse.com/thing:533814

but I found the traffic cone (http://www.thingiverse.com/thing:21773) is the best one IMHO, it use more material compare others, but it is more accurate, no side compression, and easy to check, no measurement just simple touch until it give smooth contact all around.

Hope it help

luc-github, 2015-11-19:
ho sorry I understood you were talking about the specific calibration of second extruder.
for the step calibration of every extruder, the process is well known, you can find many tutorial on internet:
http://www.instructables.com/id/How-to-calibrate-the-Extruder-on-your-3d-Printer/
http://reprap.org/wiki/Triffid_Hunter's_Calibration_Guide
https://www.youtube.com/watch?v=JZGdMc2ebPo

I use personnaly the video method, did a tube of 100mm, it is more easy to mesure filament itself and use caliper only for delta measurement

zendes, 2015-11-19:
Sorry about the confusion. Yes, I was talking about calibrating the offset between the two extruders and not calibrating the filament in my first question. It's just that step 12 mentioned both filament and dual extruder and neither had any docs, so when I found some steps for the first one I thought I'd share them. I'm going to try and print the traffic cone tonight and see how that goes.

luc-github, 2015-11-19:
ho ok, thank you

luc-github, 2015-11-24:
Any update or I close issue ?

zendes, 2015-11-24:
you can close it. from what i've read so far there isn't a standard procedure to align both extruders, so the question can't really be answered.

closed 2015-11-24 by zendes