luc-github/Repetier-Firmware-4-Davinci #123: Need translators for new 0.92.6 based version

luc-github, 2015-11-26:
new feature come with new version, now FW can change language at runtime and come with 10 languages, but lot of translation are missing due to Davinci UI, here the list:
https://github.com/luc-github/Repetier-Firmware-0.92/blob/0.92.6-sync/Translate.xlsx
Anyone who want to contribute is welcome, there is a tab for each language and only one column to update (B), currently all missing translations are replaced by english.
Here the status:
English (ok),
Deutsch (50%),
Nederlandse(50%),
Portugues(50%),
Italiano(50%),
Espanol(50%),
Svenska(50%),
Francais (100%),
Cestina(50%),
Polski(50%)

Thanks in advance,

voorter, 2015-12-12:
Hi,
I've translate Dutch for you.
Kind regards,

J. Voorter
[Translate.xlsx](https://github.com/luc-github/Repetier-Firmware-0.92/files/60096/Translate.xlsx)

luc-github, 2015-12-12:
excellent will integrate it and enable Dutch this week end
Thanks a lot for your help

voorter, 2015-12-12:
No problem. I'll try to do the German the next days.
If I want to upgrade to latest version to test, canI just reflash it or do I have to wipe the board again?

KR,

J.

luc-github, 2015-12-12:
no need to wipe, just flash
but after do not forget to load fail safe or send m502/m500 commands

first boot will ask for language, then when done press back to home menu

luc-github, 2015-12-13:
@voorter I have integrate your translation - just as reminder Davinci screen display 16 chars, and for menu it is 15 or 13 depending the kind of menu to display,need space for  selection char and menu arrow.
I just check some lines like text for bed leveling and some are 23 long, which will make an ugly result as long lines are stripped or animated to show all.
Now you can test it, but I guess you will want to give a second run after seeing the result.

This screen is really a challenge to get nice UI, french was not easy but I guess NL or GE will be harder as they have very long words.

Thanks again for your help and support

voorter, 2015-12-13:
@luc-github I did not think about the limitations of the display.
In attach adapted translation for Dutch.

Kind regards,
[Translate.xlsx](https://github.com/luc-github/Repetier-Firmware-0.92/files/60439/Translate.xlsx)

luc-github, 2015-12-13:
@voorter update implemented : https://github.com/luc-github/Repetier-Firmware-0.92/commit/8173d6788f0dbf044a950741f8bc1ee347873850

Thanks a lot

luc-github, 2015-12-30:
<b>English (100%)</b>
Deutsch (50%),
<b>Nederlandse(100%),</b> thanks @vooter
Portugues(50%),
Italiano(50%),
Espanol(50%),
Svenska(50%),
<b>Francais (100%)</b>
Cestina(50%),
Polski(50%)
Turkish(0%) - not even implemented yet

SebastianKnaus, 2016-02-19:
Hi Luc,

i`ve tried it. Can you please explain me the meaning of the red marked lines?

kind regards
Seb

[Translate_ger.xlsx](https://github.com/luc-github/Repetier-Firmware-0.92/files/138007/Translate_ger.xlsx)

luc-github, 2016-02-19:
```#define UI_TEXT_EXTRUDER_COLD_EN "E%Ne is too cold!"```
when doing load/unload or change filament position in  extruder it need minimum temperature to work, if temperature is too low - extruder is too cold to do extrusion.
this is the message to say Extruder temperature is too cold to allow extrusion

```#define UI_TEXT_LIGHT_OFF_AFTER2A_EN "Dis. After:%il"```
Light will be off after some time if no activity 

```#define UI_TEXT_KEEP_LIGHT_ON_EN "Print w/LED:%lk"```
this is the complement of previous setting, if you do not need the light when printing, set to Off,  it will power off after time defined by previous setting - if need always light when printing set to On

luc-github, 2016-02-19:
```

SebastianKnaus, 2016-02-22:
Look at this

I gave my best, lets see on the display in real :-D

[Translate_ger.xlsx](https://github.com/luc-github/Repetier-Firmware-0.92/files/141097/Translate_ger.xlsx)

luc-github, 2016-02-22:
Thanks - I will add it this week end - but cannot test right now- my printer is not arrived yet - need to wait begining of March :disappointed:

luc-github, 2016-03-23:
Got delay sorry - will do ASAP

luc-github, 2016-04-16:
GE update integrated in https://github.com/luc-github/Repetier-Firmware-0.92/commit/c0f7bde2e9e818d8cb96ad187fa798de7806d197 but not tested yet

luc-github, 2016-04-18:
GE is now enabled in  0.92.9 based version and typo are fixed to allow GE selection

ShRTee, 2016-04-26:
Hello and thank you all for the great job with this firmware, it transformed my Da Vinci 1,0A to a new printer, so the lest i can do is to contribute is translation.
[SE-Translate.xlsx](https://github.com/luc-github/Repetier-Firmware-0.92/files/236141/SE-Translate.xlsx)
 
I have now translated the Swedish translation completely.
File is attached..

luc-github, 2016-04-26:
Thanks a lot -  unfortunalty if you look at https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/lcd-4x16-bn.pdf page 13 the character are limited to a-z A-Z
there are no ä, ö etc -  I am not sure of the result but should be weird -  it is same in French 
Is it ok to change not using special char but a o instead ?

ShRTee, 2016-04-26:
Hello ..

OK, will change it tonight to letter without the dots...
Think most Swedish will understand..
Date: Tue, 26 Apr 2016 01:58:58 -0700

ShRTee, 2016-04-26:
Hello again.
Just looked at the table of characters and there is äö in the smaller format.
See in HHHH, HHHHL and HHHHH crossings..

Mats 

From: shortnder@hotmail.com
To: reply@reply.github.com
Subject: RE: [luc-github/Repetier-Firmware-0.92] Need translators for new 0.92.9 based version (#123)
Date: Tue, 26 Apr 2016 11:50:57 +0000

Hello ..

OK, will change it tonight to letter without the dots...
Think most Swedish will understand..
Date: Tue, 26 Apr 2016 01:58:58 -0700

luc-github, 2016-04-26:
Ok I will push as you gave and you can review correct when integrated if necessary

ShRTee, 2016-04-26:
Hello.

Ok, will check anyway tonight, maybe i can translate to different words using less åäö..
THX for great job..

Mats

Date: Tue, 26 Apr 2016 05:50:26 -0700

luc-github, 2016-04-26:
Hi I just integrate your translation and it is as expected - need a second round
Please note screen is 16 char long so line must be 16 if pure text and 15 if it menu item to let space for ">" and 13 if it submenu to left space for ">" and for "->"
I have fixed 2 typo issues due to  extra "
to use the  ä or ö use code like "Store \176 EEPROM"
you can do your changes in uilang.h starting    : https://github.com/luc-github/Repetier-Firmware-0.92/blob/0.92.6-sync/src/ArduinoDUE/Repetier/uilang.h#L3437
when ok just copy the section in Excel as it is :smile: I keep same format

SebastianKnaus, 2016-04-26:
Hi Luc,

in my german version are also some "ü". You can easy replace them by _x_e.

For example:

Düse -> Duese
Ähre -> Aehre

hope thats ok so.

luc-github, 2016-04-26:
Hi @SebastianKnaus actually I did not noticed it in what you sent - did you flashed with latest version to see the result ? My german is close to 0 :smile:  just know the basic I learned when I was teenager on holidays ( drink eat sleep...)

ShRTee, 2016-04-29:
Hello again Luc.
Now i did a second round to the Swedish translation.
I do not know how to get it in my printer to test it out...
Is there any program for Windows or mac to simulate the functions off line so to say..

I do not know much programing but is willing to learn..

So please tell me if i can help or do something...

Best regards
Mats Evander

Date: Tue, 26 Apr 2016 12:03:58 -0700

luc-github, 2016-04-29:
Just modify excel column then select the content of this column and copy it

Then go to uilang. h and replace text from line I mentioned earlier by selecting all lines refering to swedish and past the text you copied from excel

Then compile and flash

Very easy

luc-github, 2016-06-05:
new menus for AiO scanner and for Zmin calculation

juri74, 2016-08-13:
Hello!
i think i translated around 90% of italian, however i would test it to see how it works on the display, there are lots of text like "blablabla **%nX**" with this kind of codes, and i need to view it. how can i put it into the firmware? working with 16char len is very difficult!

luc-github, 2016-08-13:
well the only test is to modify the FW and flash it
First go to configuration.h and enable italian, I suggest to disable another language also to keep same memory footprint and avoid overload
https://github.com/luc-github/Repetier-Firmware-0.92/blob/devt/src/ArduinoDUE/Repetier/Configuration.h#L1955

then go to language.h and change lines for italian
https://github.com/luc-github/Repetier-Firmware-0.92/blob/devt/src/ArduinoDUE/Repetier/uilang.h#L2510-L3000

Then just push compile and flash - you know the proces now ^_^

Grazie mille, fa 25 anni che io non ho parlato italiano, è un po difficile da ricordare

juri74, 2016-08-13:
Infinite grazie a te Luc per l'aiuto!
only a little question however: how many characters do occupy the encoded strings? (**%xx**, **\nnn** and so on) this would be useful to optimize text
grazie ancora, ciao!
Juri

luc-github, 2016-08-13:
Well that the tricky things every variable can have different size.

juri74, 2016-08-13:
ok, i feared this reply... ok testing time
thanks again

luc-github, 2016-08-29:
Italian implemented https://github.com/luc-github/Repetier-Firmware-0.92/commit/b9e31ef3639ee4bcb11f933371337c98e38e7a98

alexws54tk, 2016-11-04:
@luc-github, I added Russian translation.
Take it from my gist [here](https://gist.github.com/alexws54tk/1cc3b23fdc558f8cb869085c82e58366).

luc-github, 2016-11-04:
Hi alexander - thanks a lot - did you tested it ? I see a lot of cyrilic char - I am surprised it can be displayed  the char table seems having any :
[image]

alexws54tk, 2016-11-04:
Hi Luc. I not testing this, because I not have printer.
This machine is shows Latin characters only?
Sounds bad. Grief trouble.
Well, at least, it is possible to pull translation to the main project.

luc-github, 2016-11-04:
yes sure - check here : https://github.com/repetier/Repetier-Firmware/blob/development/src/ArduinoDUE/Repetier/uilang.h there are actually  less words to translate than Davinci version :smile:

alexws54tk, 2016-11-04:
Ok. Later, I merge my translation of these two projects.
Currently, I write font with Cyrillic characters for Repeater Firmware.
Same as in «Marlin».

Pascalmarin, 2018-05-29:
Hi,
here is the complete (I think so) Spanish translate ;-D

[Translate.xlsx](https://github.com/luc-github/Repetier-Firmware-4-Davinci/files/2048443/Translate.xlsx)

luc-github, 2018-05-31:
@Pascalmarin Thanks a lot ^_^

closed 2019-07-11 by stale[bot]