luc-github/Repetier-Firmware-4-Davinci #124: Dosent Pre Heat and Bed Dosent move away when finished.

Whopper3DPrints, 2015-12-17:
First when i start a print on the machine it dosent calibrate or heat up and i cant even make it print with repetier. When print is finished bed dosent move away from hot end. Please help thanks

luc-github, 2015-12-17:
These actions are done by your slicer in custom start GCODE and end GCODE.
not by FW.
In Stock FW you have no control but in repetier you have FULL control - it means the FULL control is given to GCODE provided by your slicer like any other printer.
Have a look to the [FAQ](https://github.com/luc-github/Repetier-Firmware-0.92/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+) and especially [What GCODE are supported ?](https://github.com/luc-github/Repetier-Firmware-0.92/issues/105)

Whopper3DPrints, 2015-12-17:
Thanks for that but I also noticed that my Repetier will crash and that fails my print is there anyway to just send the code to the printer so it can print stand alone.

jamesarm97, 2015-12-18:
It's called, print from a SD card. Does the board you use have and SD? Most do on the LCD or on the board.

luc-github, 2015-12-18:
if you use repetier you can send the generate GCODE file to SD then launch the print from repetier or printer, but transfert speed is very slow, it is better to use a wifi SD card which make file transfert almost immediat.

Whopper3DPrints, 2015-12-19:
ok well i solved my repetier crash problem but now my gcode wont tell the extruder to heat it tells the bed to heat but not the extruder and i f i want the extruder to heat i must command it on the printer itself.

luc-github, 2015-12-19:
check what gcode you used, you may used the gcode asking to reach bed temperature before starting to heat extruder. check your generated gcode - the problem is there not in FW actually

luc-github, 2015-12-21:
if no update I close issue

closed 2015-12-21 by luc-github