luc-github/Repetier-Firmware-4-Davinci #125: Bad bug, X stepper wont stop moving on home

gamecompiler, 2015-12-20:
Hi
if the X-Axis is in parking postion (Y and Z not, cause my hose lost power),
and i press home all,  the X-Axis stepper wont stop moving and dont detect that he is already in parking position.

Please somebody test if he have the same problems.
Thanks in advance.

luc-github, 2015-12-20:
I am 10K km away from my printer so cannot test until end of December, sorry.
Can you confirm : no issue if all is already homed and you home all again ?
You can check the 0.92.6 (https://github.com/luc-github/Repetier-Firmware-0.92/tree/0.92.6-sync) to see if you still can reproduce the issue - it is still in alpha but compile and homing is fine I remember there were some fixes for homing.

gamecompiler, 2015-12-20:
I cant flash the firmware.
I feel me bullied from my PC ^^.
The uploading succeed, and then nothing happens....
The Printer is restarted a lot of times.

The current master branch works well(But with the bug inside).
(Arduino IDE 1.6.6)

luc-github, 2015-12-20:
do not use 1.6.6 : use Ide 1.6.5 and module Due 1.6.4

luc-github, 2015-12-20:
1.6.6 ide and module need some modifications to make fw working - not done yet, it is on TODO list

gamecompiler, 2015-12-21:
Id got it compiling only with IDE 1.5.8 but it works.
(Awesome with the translation feature !)
(Also with the Bed coating thing, what i can do with that Oo?, im using glue all the time without that feature)

The bug is not fixed in the sync release.
Id have the same problem. If usefull i can make a video of the bug.

gamecompiler, 2015-12-21:
Just trying out, do you use a e3D v6 or another thermistor ?
Because now the thermistor table works correct for me.

Thank you very much !

luc-github, 2015-12-21:
ok will check when back to home, as I need the printer to get debug info

No my 2.0 is full original but FW of course ^_^

coating is new feature of 0.92.6, it add some offset according coating, if you do not see any advantage just left none and ignore it ;)

luc-github, 2015-12-29:
ok - back to home and cannot reproduce the issue at least with 0.92.6
but home flags are not consistent - will redo a test first with 0.92.3 as I have already noticed boolean 
variables seems not giving correct value or it is wrongly interpreted as true  even in false state.

Also, can you give a step by step process with also a status of end stops state (check page 2 of main screen) process to reproduce it?
like what I did :
1- home X (end stop is on/black square ) no issue,  put Y at +10 (end stop is off/white square),  put Z at +10 (end stop is off/white square)  = only X is homed
2 - power printer down 
3 - power printer on, X end stop is on/black square , Y end stop is off/white square,Z end stop is off/white square
4 - Home all, no issue, X end stop is on/black square , Y end stop is off/white square (because I have #define ENDSTOP_Y_BACK_ON_HOME 7), Z end stop is  on/black square

Doudou14, 2015-12-29:
If I make a home X no worries
If I make a home all, the X axis run in the back and hit on then stop. Once all the axes in home position, the X axis shifts slightly towards on the bed.

Otherwise good job, you have the patience to do all that!
This printer how to sell in France, will have one dedicated forum

luc-github, 2015-12-29:
Edouard can you tell me since when you have these issues (Hit end stop / little shift) ? it has always been there or it is new on latest release only?

What is the state of printer before doing Home all ? just power on ?

thanks

Doudou14, 2015-12-29:
Since I've spent the last Version 0.92.6 (I had a version of April 2014)

I have this problem when I do a home all axis on repetier or on the LCD or when I gcode printing makes this command

luc-github, 2015-12-29:
Ok so happen on 0.92.6 branch not on master branch then
Thanks

luc-github, 2015-12-30:
flashed 0.92.3 (master branch) with IDE 1.6.5 / DUE 1.6.4 as Davinci 1.0 and cannot reproduce any issue
I can home all or separately - behavior is correct 
End stop status at home for 1.0 at parking/home position is 
X: -33.00 mm black square= on end stop 
Y: -12.00 mm black square= on end stop 
Z: 0.00 mm black square= on end stop 

So need more details to reproduce issue

for 0.92.6 I will open another issue related to boolean management - that may bring this behaviour but I was not able to reproduce yet

luc-github, 2015-12-30:
@Doudou14  FYI Shift issue is now fixed in 0.92.6 but cannot reproduce the second issue

luc-github, 2015-12-30:
@gamecompiler pls provide more information to reproduce issue and try again 0.92.6 to see if issue is still valide 
Thanks

Doudou14, 2015-12-30:
Great job, the homing X axis works well :)
Thank

gamecompiler, 2016-01-02:
Id found the problem. I added a selfmade carriage for my e3d v6. 
The end stop flag was to short. 

Im sorry.

Thanks :smile:

closed 2016-01-02 by gamecompiler