luc-github/Repetier-Firmware-4-Davinci #130: I need some help.

firedrake150, 2016-01-05:
So, I followed the instructions in the Arduino files to flash the printer, and jumped JP1, with Davinci 3 set and Model 0 since it's not an "A" Model Device.  I can upload the firmware just fine, but the printer doesn't display anything except black bars.  Am I Missing something small?

luc-github, 2016-01-05:
what version of IDE are you using ?

firedrake150, 2016-01-05:
Arduino 1.5.8 Repetier Firmware 0.92.  Arduino says it successfully
uploads.

Indie Developer,  mobile.   This communication may contain content that is
confidential or personeel sensitive.   If you have recieved this
communication in error,  please delete all copies and notify the sender.
have a lovely day.

luc-github, 2016-01-05:
did you modified the variant.cpp as requested ? black bar are usually the sign the file is not set properly

firedrake150, 2016-01-05:
I did that on a few machines.   each one the modified files made sure
Arduino would not load up.  I'll give it another shot and see.

Indie Developer,  mobile.   This communication may contain content that is
confidential or personeel sensitive.   If you have recieved this
communication in error,  please delete all copies and notify the sender.
have a lovely day.

luc-github, 2016-01-05:
in 1.5.8 the variant.cpp is in program files so be sure system allowed you to modify files

luc-github, 2016-01-05:
you wrote you did on several printers, it is happening on all printers you flash or only on some ?

firedrake150, 2016-01-05:
Mean't PC's with different OS's.   A Mac,  A win 7 pro,  A win 10 Pro,  A
Win 8 Pro.   All did the same thing yeah.

Indie Developer,  mobile.   This communication may contain content that is
confidential or personeel sensitive.   If you have recieved this
communication in error,  please delete all copies and notify the sender.
have a lovely day.

luc-github, 2016-01-05:
I can flash from OSX / win7 / Win 10 so OS is not the issue

You may try arduino 1.6.5  / due 1.6.4 

you used JP1 and set to DUO this is what I have (just one fan instead of 2) so I am confident it is working.

if flash is working but black bar it is always the variant.cpp per my experience

Try to erase with Jumper again just in case - and power off the printer few sec

luc-github, 2016-01-05:
also to check it is not the printer you can use the binary I provided for 0.91 it is old but work on 2.0
https://github.com/luc-github/Repetier-Firmware/releases/tag/v1.0-RC1

using bossac :
bossac -p <your COM like COM16>  -R -e -w -v -b DaVinci-2.0DualFans-2014-12-12-1.3.bin

firedrake150, 2016-01-05:
I got the printer to beep loudly and now it's no longer displaying black
bars.   I ended up redownloading the entire archive and replacing the file
itself rather then a folder,  And ONLY variant,  not the two thr vid I
watched had replaced.   I'll have to give it Some time to verify I have
everything working but so far we've made progress.   Thank you for taking
the time to help.   ^w^

luc-github, 2016-01-05:
ok I suggest to do same with USBcore.cpp it is not mandatory but it is an additionnal safety if you use USB to print

luc-github, 2016-01-05:
it is always the variant.cpp copy issue :smile: 
happy printing

closed 2016-01-05 by luc-github