luc-github/Repetier-Firmware-4-Davinci #132: PID M301

gamecompiler, 2016-01-08:
Hi, id made a PID calibration today with:
    M303 E-0 S230 C8

The results were for my e3d v6:
    12:04:12.573 : Kp: 19.45
    12:04:12.573 : Ki: 1.57
    12:04:12.573 : Kd: 60.37

To save the settings i wanted to enter:
    M301 P19.45 I1.57 D060.37
    M500

But at M301 .... the printer says:
    12:07:46.891 : Unknown command:N1197 M301  P19 I1.57

If i enter only M301:
    12:10:58.621 : Unknown command:N1387 M301

luc-github, 2016-01-08:
Yes no M301 according to repetier.ino 
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Repetier.ino

the command is M204

jamesarm97, 2016-01-08:
FYI, you can add an X1 to the end and it will save the PID values in EEPROM for you after the tune is done. I have some gcode scripts for common temperatures on my website for the Dup I3 users to download and run and that is what it does to make it simple for them.

M303 P0 S230 R8 X1 ; (Extruder0, 230c Reps:8 Autosave)

closed 2016-01-08 by luc-github