luc-github/Repetier-Firmware-4-Davinci #133: No support for M17/M18 Gcodes

henry-id, 2016-01-09:
Pretty common command, and quite useful.

There's already a menu option to do it, so I really can't figure out why it can't support these Gcodes.

I'm currently looking into enabling it for myself, but this coding is a level above me right now.

luc-github, 2016-01-09:
what do these commands ?
they are not in repetier FW
https://github.com/repetier/Repetier-Firmware/blob/work092/src/ArduinoDUE/Repetier/Repetier.ino

luc-github, 2016-01-09:
Are the reprap command for enable / disable motor  ? repetier is M84

luc-github, 2016-01-11:
Any update or  can I close issue ?

closed 2016-01-13 by luc-github

clowrey, 2017-05-04:
I would appreciate M18 support too as Simplify3D uses this command for its "disable motors" button in machine control panel, which does nothing on repetier. But manual M84 does work, so maybe the M84 command could also be used for M18?

luc-github, 2017-05-04:
no need it is,  just wrong default setting in S3D which can be modified
[image]

clowrey, 2017-05-04:
Ahh I see, thanks for the hint on that. Is there any easy way to replace the M17 "enable motor" command with an equivalent in Repetier?

luc-github, 2017-05-04:
https://github.com/luc-github/Repetier-Firmware-0.92/issues/105
repetier automaticaly enable motors when used so there is no enable motor command

clowrey, 2017-05-04:
I see.. might be nice to have enable if you just want them to hold position without moving. But I understand it is not 100% necessary

luc-github, 2017-05-04:
in that case you just send command to move of 0 position it should do the trick

clowrey, 2017-05-04:
If anyone has edit privileges on this page http://reprap.org/wiki/G-code the section on M18 for Repetier could be updated to say "use M84" like Marlin has, instead of "No" like currently.