luc-github/Repetier-Firmware-4-Davinci #135: DaVinci 1.0A stops mid print

mkvetny, 2016-01-20:
I recently flashed my DaVinci 1.0a with repetier 0.92 firmware, and am running repetier host 1.6.0. There were no problems with the installation, or the manual calibration. The problem occurs when printing.  The print starts normally and will run for about 20 min or so, after which the printer stops. One command is listed as held in buffer, the menu on the front panel can be navigated. The printer is completely unresponsive to repetier host. I looked through the g-code and didnt see anything out of the ordinary about the last command sent. If you have any questions please feel free to ask. Any help is greatly appreciated.

luc-github, 2016-01-20:
Looks like same issue as https://github.com/luc-github/Repetier-Firmware-0.92/issues/38
the problem is linked to usb Davinci port sometimes got noise and loose connection
I fix should be available in test branch - https://github.com/luc-github/Repetier-Firmware-0.92/tree/0.92.6-sync but as I do not have a printer right now to test the code until march - I cannot validate the code
You can give a try 
Or you can print from SD which is always more reliable than USB, ( I never print from USB)

luc-github, 2016-01-26:
If no update I will close as duplicate as https://github.com/luc-github/Repetier-Firmware-0.92/issues/38

closed 2016-01-26 by luc-github

mkvetny, 2016-01-26:
I haven't had an opportunity to try it just yet. Ill let you know.