luc-github/Repetier-Firmware-4-Davinci #137: Extruder names used for manual heating through LCD are incorrect

limited660, 2016-01-27:
When using the LCD to heat the extruders manually, Extruder 1 heats the left and Extruder 2 heats the right.

luc-github, 2016-01-27:
weird I used it on my duo and was ok
extruder 1 is left and extruder 2 is right

limited660, 2016-01-27:
Maybe I am confused then as most everything else in the firmware refers to the right extruder (when looking at the front) as 0 or 1.

luc-github, 2016-01-27:
right is T1
left is T0

luc-github, 2016-01-27:
why T0 is left?
because when in home position, the selected extruder is T0, so if select T1 mouvement is possible
So in home position T0 is at position 0,0 , ready to be used,

limited660, 2016-01-27:
When homed, the printer is at purge box for extruder 2?

luc-github, 2016-01-27:
yes

limited660, 2016-01-27:
Ok I will need to fix the code for M100 because I thought the homed extruder was 0.  Is it possible to clarify the original area to say left and right like the Load/Unload rather than 1 and 2?

luc-github, 2016-01-27:
sure, but on phone now, will check later
you right drop box should refer to right and left not 1 and 2, it is a miss, and confusing

limited660, 2016-01-27:
I've found the relevant code.  Is there a reason to have the ability to select extruders on the page where the heating is done?  Continuing on below the "SwitchOff" shows options to select each one and you can't do anything with that from what I could see.  Also do you have a problem changing "SwitchOff" to "Disable", it allows "Left" and "Right" to fit nicely on the heating page.

limited660, 2016-01-27:
Pull request up, French entries will need changed if they desire.

closed 2016-01-27 by limited660

luc-github, 2016-01-27:
for selecting extruder menu I did it on purpose, but need to remember the user case as I do not have my printer until March. to visualize the menu

limited660, 2016-01-27:
These were on the menu used to turn heating on.