luc-github/Repetier-Firmware-4-Davinci #139: DaVinci 1.0A Stepper Motors getting too hot

MasterGsus, 2016-01-27:
I have a DaVinci 1.0A with repetier 0.92.6 an E3D v6, Simplify3d and Octprint on a Raspberry Pi B.
after 3 hours printing my Stepper Motors (all of it) getting really hot an then this is going happen (see my video)
https://www.youtube.com/watch?v=BeEHgqOaZfQ

this is what i have try to solve this 
- reflash with new 0.92.6-synch
- checked #define ALWAYS_CHECK_ENDSTOPS 0
- changed X/Y Accelleration from 1000 to 900
- print without Octopi, directly from Simplify3D
- check cable to X/Y Stepper motor. no break
but with no success

now have no more ideas...
what can i do?

luc-github, 2016-01-27:
what is your printing speed ?

MasterGsus, 2016-01-27:
i started with 50mm/s reduced to 40mm/s an now 30mm/s but no difference.

luc-github, 2016-01-27:
looks correct speed
There is not current control on Davinci, normal printer drivers have digpot or manual potentiometer but on Davinci all is hardwired.
Some points to check: 
- check if motors are cold when printer is Idle 
- be sure your axis are well lubricated with oil
- it is happening with all models ? even with cubes ?
- anyway motors are always hots when running during hours, I do some prints for more than 8 hours but never got this issue, you can put a fan blowing on motherboard and add heatshinks on motors them self

MasterGsus, 2016-01-27:
Many thanks for your fast repley

- check if motors are cold when printer is Idle
yes they are cold
- be sure your axis are well lubricated with oil
yes 
- it is happening with all models ? even with cubes ?
they getting hot on every model. but the "wiggle" is only on circles an curves
-anyway motors are always hots when running during hours, I do some prints for more than 8 hours but never got this issue, you can put a fan blowing on motherboard and add heatshinks on motors them self
i have orderd now heatshinks and a fan for motherboard and will test it.

luc-github, 2016-01-27:
for your first tests you can just use a normal fan, just open the back door and let the fan blow on motherboard

MasterGsus, 2016-01-29:
so now i am going to test a longer print with fan on the back. but i have see another problem with 0.92.6-sync.
i use simplify3d and there i have a z-offset -0.30mm, but my printer say -0.08. i have checked the g-code and there is -0.30mm
also my layer-high in g-code and simplify3D is 0.2mm, but my printer shows each layer a difference vom 0.1mm.
i think i am going back to 0.92

MasterGsus, 2016-02-01:
now i have tested with fan blowing on the board, but changed nothing.
then i have placed the heatshinks on the stepper motors and a test print with 4 hours works well.
and i have noticed when i print circels with outline 50% of printspeed (25mm/s outline speed) this is too slow and the motors make this like in the video. but when i make outline speed minimum 40mm/s it makes normal circles without the "wiggle".

luc-github, 2016-02-01:
in 3D printing it is always to find the right setting => too slow or too fast = Not good

About heat seems it is not drivers issue but motors only per your experimentation - I read on some forum people add heatshinks sometimes even on stock FW as motor getting hot - but it is not so common, but now you have the solution so it is good

thevisad, 2016-02-01:
No one asked the important question, what is "too hot" and what is the temperature of the motors? A no touch thermometer or a thermoprobe will tell you that. My Davinci has a completely insulated chamber (http://puu.sh/mRQEQ/36634ca9c4.jpg), which has increased the internal temperature of the build chamber by 20 degrees Celsius. I run roughly 55-65 degrees inside the build chamber.  I can tell you my motors are a lot hotter then what you are experiencing and I have never had a single issue.

luc-github, 2016-02-26:
if no update on this I close issue

closed 2016-02-26 by luc-github