luc-github/Repetier-Firmware-4-Davinci #140: Extruders read 29C at room temp

limited660, 2016-02-05:
Both extruders read 29C at room temp (~20C) with DaVinci stock thermistors and the 100k thermistors provided by E3D.  I've tried bypassing the entire control board but nothing changes.  I'm not sure if the electronics may be different for 2.0A but do you have any suggestions for improving thermistor table or finding what the issue is?

limited660, 2016-02-05:
If the temp ADC value could be printed over serial I might be able to make a new thermistor table but I'm not sure if that's the issue.

luc-github, 2016-02-05:
I am surprised you have same  value with stock and 100k, davinci sensor is not a 100k thermistor per all feedback

luc-github, 2016-02-05:
tables for 1.A, 2.A and AiO have been done by 1.A owner so cannot help to say what is the shift

limited660, 2016-02-05:
This is the closest I've been able to get just by modifying the table and testing temperatures against my Fluke 87v thermocouple.  It's within 1-2C for most of the printing range but I can't test above 245C on the one and it seems to be less accurate at the low end.
```

{294,2383},
{375,2231},
{461,2109},
{526,2025},
{601,1949},
{674,1895},
{811,1781},
{925,1704},
{1071,1628},
{1219,1560},
{1445,1453},
{1673,1369},
{1902,1293},
{2187,1194},
{2454,1122},
{2672,1026},
{3000,929},
{3196,854},
{3439,756},
{3583,681},
{3736,583},
{3899,447},
{3960,374},
{4024,266},
{4067,127},
{4079,92},
{4100,-58},
{4120,-212}
```

limited660, 2016-02-05:
I've seen mentions about R27-R29, it appears that R29 is unpopulated on my board - http://i.imgur.com/mrmh2NX.jpg  If I remember correctly, measuring the resistance of the thermistor leads at the extruder board gave me 16 mega ohms.  When I measure the E3D at roughly room temp I get around 100k and DaVinci was similar

luc-github, 2016-02-05:
what is the looks of the curve compare to ogremustcrush one : https://github.com/luc-github/Repetier-Firmware/issues/83#issuecomment-68432991

Did you measured the resistance of R29 even looks not populated ?

limited660, 2016-02-05:
Blue is #5, orange is what I have gotten so far that is pretty close.

[image]

luc-github, 2016-02-05:
do you use ?
```- M105 X0 - Get temperatures. If X0 is added, the raw analog values are also written.```

luc-github, 2016-02-05:
about table and calculation you can refer to https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/bed-extruder-tables.xlsx

limited660, 2016-02-05:
I seen the X but wasn't sure how it worked, I'll try to generate a better table.

luc-github, 2016-02-05:
generate table experimentally is not difficult - difficulty is to reproduce same measurement 3 to 4 times to verify reliability - we had some hard time previously to get reproducible measurement -  the best it to do several measure and do an average remove obvious bad measurement not on curve

limited660, 2016-02-05:
I have the thermocouple pinched between nozzle and heat block right now, that's about as close as I can get right now.

limited660, 2016-02-05:
So going off your file, the ADC values are unchanged but the temp is x8?

luc-github, 2016-02-05:
Yes as I remember adc did not changed but temp X 8

limited660, 2016-02-05:
Here is one I just made, only was able to get ADC values from 16.5C to 247.8C.  Once I get more time I am going to sample enough to even out the ADC range (some range 20, some almost 300 per value) and somehow figure out max/min.

{558,1982},
{647,1897},
{747,1818},
{851,1744},
{981,1666},
{1141,1583},
{1308,1506},
{1480,1432},
{1696,1354},
{1782,1336},
{1942,1274},
{2156,1200},
{2454,1110},
{2482,1109},
{2639,1054},
{2886,975},
{3109,897},
{3312,819},
{3494,744},
{3631,672},
{3754,598},
{3850,522},
{3934,437},
{3986,358},
{4027,265},
{4045,212},
{4063,132}

luc-github, 2016-02-06:
in this case you may use current data to add the missing points because the
trend calculation at cold temperature may give wrong values and give sensor
defect error - current low temperature at least do not give that

luc-github, 2016-02-06:
I did not check your new data but to smooth curve you can use the nice excel function that give trend
use polynomial 5 or 6 for some part of the curve make it more smooth and help heater to have smooth heating

limited660, 2016-02-06:
I've seen how to apply the polynomial trendline to  the chart but I'm not sure how to use that to change the data other than change points individually until they fit the trendline on the graph.

luc-github, 2016-02-06:
you get the function by displaying on chart
then you create a function reading the temperature  on one cellusing relative reference ,then create your range of temperature and copy the function that will give the ADC values

luc-github, 2016-02-06:
for hat I can see the polynomial 5 match very well your data and correct the little glitch in the middle

luc-github, 2016-02-06:
[image]

limited660, 2016-02-06:
Do you have example file of using polynomial equation to gather other data points?

luc-github, 2016-02-06:
no sorry I cleaned up just to put values then curve was ok
you need to convert for example : 
[image]

to an excel function using B1 cell as input for example http://www.excelfunctions.net/Excel-Power-Function.html :
```=(1E-12)*POWER(B40;5)-.....

luc-github, 2016-02-06:
sorry checking file - seems bed tab has the trend calculation - you can check it

luc-github, 2016-02-06:
any success in the smoothing ?

limited660, 2016-02-06:
I'm out of town until tomorrow night, will try then

limited660, 2016-02-08:
This one is pretty smooth along the center but the equation doesn't seem to work on the legs so I just used the old values and guessed.

```{4120,-212},
{4100,-58},
{4079,48},
{4070,132},
{4038,212},
{4021,265},
{3985,358},
{3937,437},
{3858,522},
{3759,598},
{3635,672},
{3489,744},
{3312,819},
{3105,897},
{2878,975},
{2636,1054},
{2459,1110},
{2173,1200},
{1943,1274},
{1705,1354},
{1490,1432},
{1305,1506},
{1133,1583},
{975,1666},
{850,1744},
{749,1818},
{654,1897},
{555,1982},
{526,2025},
{461,2109},
{375,2231},
{294,2383}```

luc-github, 2016-02-08:
yes would be too easy if equation work all the time ^_^

limited660, 2016-02-19:
I'm not sure if another table should be added at this point as it doesn't seem this issue is very widespread. I've just been dumping this in as table 4 but it currently seems to work for me.  Maybe this is a tolerance issue on the resistors?

luc-github, 2016-02-19:
can be - did new table improve printability in addition of user experience at room temperature ?

limited660, 2016-02-19:
Temperatures are now very close for me up to 245C, I'll test up to 300 on my other extruder when I get time.

luc-github, 2016-02-20:
I am little bit busy with my moving process right now but I think adding your table is good for 1.0A users 
Will complete the excel file and add it as option default when I have more time

MasterGsus, 2016-02-22:
how can i test this table? can someone send me the arduino file with this table included?

limited660, 2016-02-22:
MasterGsus, you can use the one on my page for now - https://github.com/limited660/Repetier-Firmware-0.92

I updated Extruder.cpp and changed temp table 4 to the new values, I also changed Configuration.h to default to table 4 for your test.  You will still need to select whatever printer you have but this build will default to the new thermistor table.

luc-github, 2016-02-22:
or you can modify configuration.h
here for 1.0A:
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L646-L647

using https://github.com/luc-github/Repetier-Firmware-0.92/issues/140#issuecomment-181364340

change NUM_TEMPS_USERTHERMISTOR0 to 32 to match content of new table

MasterGsus, 2016-02-23:
no i have flashed yours limited660. but i have at room temperature on the bed 19C and extruder 0C. is this correct?

limited660, 2016-02-24:
No it's not correct, what did your extruder read at room temperature before this firmware?

MasterGsus, 2016-02-27:
bevor with stock table i always had a difference of 10C between hotend and bed. bed 19C and Extruder 9C.
i have installed a e3d v6 but with stock extruder i hade the same difference.

limited660, 2016-02-29:
Well MasterGsus, it looks like you have a few options here... 

1. Create your own table with a thermocouple and use "M105 X0" in your host to get the ADC value.
2. Find a way to generate an accurate table with the calculators (I don't think anyone has been able to do this with the Da Vinci)
3. Swap the board to one of the many others (Ramps, Rambo, Smoothie, etc) that have known setups for thermistors.
4. There might be a way to get past all the components on the board and implement a setup similar to one of the boards in 3?

limited660, 2016-03-15:
Well Luc from what I have seen, everyone that doesn't get a correct temperature reading from the start usually doesn't end up in the same place so I'm not sure this table will help anyone but myself.  It may be beneficial to add it as yet another table that someone can choose from if they have issues but not the default.  I'll leave it up to you whether or not to add it.

closed 2016-03-15 by limited660

luc-github, 2016-03-15:
Thank you - well, we got same trouble for 1.0 /2.0, doing experimental table is not easy, and to get reproducible results across all printer even more difficult.

limited660, 2016-03-21:
Luc, I've read the pinout file and I see there are a couple ADC inputs available.  Would you happen to know if/where they are on the board?  I'm thinking of moving to a thermocouple to get around this issue and I need 2 analog inputs.

luc-github, 2016-03-21:
Well I did not need them so never search about
What you can do is to follow from the pin on MCU to see where is go but not sure it go to a connector - or on the path components may be missing 
Sorry cannot help more

limited660, 2016-03-21:
It would be even better if it was a clean input, I'll take a look when I get some time.  Thanks

limited660, 2017-05-23:
@luc-github just some more input on this.  I noticed while running the printer the other day that when the heaters kick in there is a rapid change in reported temperatures on the LCD so maybe everyone with this issue just isn't getting the correct voltage to the board and this is possibly calibrated out at the factory?

luc-github, 2017-05-24:
ye,s every printer should get some variation but I do not think this will be calibrated in factory as no variable resistor on board and no way for service to calibrate this and store some calibration infowhen they send new board if they do not have the full printer

Temperature do not need to be accurate compare to reference but proportional and reproducible.
 I do not remember Stock FW showing room temperature but just heating temperature, what is important is usage temperature
I am pretty sure XYZ filament has wider range of melted state compare to standard filament which compensate the accuracy
_just my 2 cents