luc-github/Repetier-Firmware-4-Davinci #141: Da Vinci 1.0A Motherboard ?

bucklandgate, 2016-02-05:


luc-github, 2016-02-06:
What is the issue ?

bucklandgate, 2016-02-06:
Hi Luc
Thank you for your email. The issue is that I bought a Da Vinci 1.0A a few weeks ago with the intention of flashing it and using Rep 0.92 to save costs of filament and improve print quality. I am an ex mechanical design engineer now 80 years of age and have been gagging to try 3d printing for some time. Unfortunately, in my lack of electronics knowledge I think I flashed the printer when it was switched on so it now appears unusable. When I switch on I get a humming sound and 3 LED's light up on the motherboard but I have tried installing the firmware (using Arduino Uno) using either my Mac (OSX 10.11.3) or Lenovo (Windows 10) without any success.
I am afraid I have destroyed the motherboard and am looking for advice on replacing it with anything other than another  Da Vinci board to give me what I want. I have a multimeter and various tools so I'm sure I will be able to cope.
I would be extremely grateful for any advice either on testing the current motherboard for faults or on replacement with a different board. Sorry to be a nuisance but I am determined to carry on learning about 3d printing.
Best Regards
Roy Ellis (royellis@sky.com})

luc-github, 2016-02-06:
Hi Roy,
no worry - to flash FW no need to be electronician (I am not ) - and just flashing won't brick the printer - I guess you did wrong process that is all

May I know what process you followed to reset board and flash FW ? as I do not understand why you use arduino Uno

rainierez, 2016-02-06:
I had issues like that when using arduino ide version 1.6.x but when I used 1.5.8 it works just fine.  Try using that.

bucklandgate, 2016-02-06:
Hi LucThanks for replying. I thought I was following instructions on github page "luc-github / Repetier-Firmware-0.92" and the "Installation" section. Section 3 says Use an arduino IDE supporting arduino DUE etc. so I have used my arduino UNO and tried loading both 1.5,8 and 1.6.5 with no success, the display shows no light and I get no info. I haven't yet found out how to see error messages in arduino! I tried to be careful in loading the firmware and doing the edits but no go.Is there any other way to install the firmware on the printer? ~Sorry to be a pest but I really am trying!RegardsRoy Ellis

bucklandgate, 2016-02-06:
Many Thanks Chris for the info. I will keep trying 1.5.,8 but at present I have had a few errors on 'Verification' and need to sort those our before I can try 'Upload'. I will keep you posted.RegardsRoy Ellis (bucklandgate)

luc-github, 2016-02-06:
Roy,
You do not need any Uno card just the IDE and USB cable between your printer and your PC

1 - you need to modify on your Arduino Installation files the variant.cpp and optionally the usbcore.cpp
2 - you need to open the repetier.ino from src/due of my github files, modify to set model to 1  and compile it
3 -  if ok, you must reset board with Jumper
4 - once board is reset you can upload using Arduino IDE
5 - once flash is completed - power off then power on printer
6 - go to menu printer and go to settings to load fail safe settings and save

there is a very nice tutorial here : https://www.youtube.com/watch?v=rjuCvlnpB7M

Cannonized, 2016-02-06:
Roy, I had similar issues/worries when I first did the flash. It took me 3 times to reset the board in the printer.  Had to download new USB drivers it was very frustrating to say the least. I uploaded my printer about a month ago with arduino 1.6.5 it works great. If you have any issues we're all here to help. Luc is amazing and very quick to respond even when it seems like a simple "your having issues with that?" Type of question. Everything seems daunting at first then you get past it and your amazed at how silly you feel for being so frustrated and worried.

bucklandgate, 2016-02-08:
Hi LucSorry I missed a reply to yours of 0602 but I have been trying so hard to follow instructions without any success. I must be doing something really stupid but every timeI try to upload from arduino I get a couple of screens of red error messages and eventually 'compilation failed'. I've found it very difficult to find arduino 'installation files' on my Mac El Capitan and generally  got to think that all this is too much for me. I shall keep trying different methods of getting repetier onto my printer but am beginning to think it might take me years! Your message was so detailed I honestly thought the job would be done in an hour or two but you hadn't expected such incompetance.Wish me luckRoy Ellis

luc-github, 2016-02-08:
Hi Roy, if you say you are following steps and get compilation failed it means you do not select correct board in IDE or doing wrong modification in files.
if reading the process do not work for you - I suggest we turn this in opposite view

Describe each steps you are doing as precise as possible and I will tell you what part is wrong/correct

I do not see another way to help you sorry

bucklandgate, 2016-02-09:
Hi LucThanks for your extreme patience. I appear to be going wrong at step 4 (Installation) in that I can't find arduino files (variants.cpp and USBcore.cpp) although I have found the replacements in /src. Whenever I start arduino and open repetier.ino there is no USBcore or variants! Even so I'm still doing something wrong because whatever's left won't compile! I am unwilling to use Windows  (unless ABSOLUTELY necessary) because I have a 27" Mac against an 11" Windows andat my age this is critical for clarity.Hope you don't mind me asking.Best Regards
Roy Ellis 

 

if reading the process do not work for you - I suggest we turn this in opposite viewDescribe each steps you are doing as precisely as possible and I will tell you what part is wrong/correctIf you can do on your lenovo win10 - it will be more easy than Mac I thinkI do not see another way to help you sorry —

luc-github, 2016-02-09:
ok give me time to check in what box is my mac, I am moving and boxes are everywhere.
In same time please describe all your actions with your words to duplicate the process of flashing because even without these files it should compile fine

also please describe what Version of IDE and Due module you are using and how you installed them, it can help to understand the situation

bucklandgate, 2016-02-09:
Thanks Luc
I am really sorry to interfere with your moving arrangements and please say if you'd rather leave this until you're settled. (My wife would be delighted of I spent less time "fiddling").1. On Da Vinci 1.0A - Turned DV 'OFF' shorted J37 perhaps 10 secs, then turned DV 'ON' for 10 secs, then 'OFF'.2. On Mac - downloaded "arduino- 1.6.5-r5-macosx.zip" from Arduino site and "Repetier-Firmware -0.92-master.zip' from github. Arduino.app shows in "Applications" but not RepetierFirmware, only Repetier-Host which I also installed.3. Mac USB Tree shows only "Communication Device" connected as follows Communication Device:
  Product ID: 0x6124  Vendor ID: 0x03eb  (Atmel Corporation)  Version: 1.10  Speed: Up to 480 Mb/sec  Location ID: 0x14100000 / 3  Current Available (mA): 1000  Current Required (mA): 100  Extra Operating Current (mA): 0
No sign of "Arduino" board.4. I found 'variants.cpp' and 'USBcore.cpp' in /src/arduinoDUE/etc but cannot find the arduino ones to be replaced.5. Start 'Arduino', open 'Repetier.ino' from 'Repetier-Firmwarfe-0.92-master' and found 'USBcorp.cpp' IS among the files, but NOT 'variants,cpp'. Tried to 'Upload' and, again hundreds of red lines and 'Compilation Failed"????
I know I'm doing something wrong but am totally lost as to what!Don't forget if you would like to leave this until you have more time I'll be patient!Many thanksRoy

luc-github, 2016-02-09:
ok I see you did not download Due from board manager, right?

bucklandgate, 2016-02-09:
I did. I went into Board Manager and clicked 'Arduino Boards (etc,,,,) then picked SAM and installed version 1.6.4 but may have clicked 'Install' which took a while. I've just found an error: Unable to 'copy' 15 lines but 3  main headings are:-Error downloading  http://arduino.cc/packages/package_index.json3 linesCaused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json6 linesCaused by: java.io.IOException: Received invalid http status code from server: 4163 lines
Hope that helps but I was unable to 'cut and paste' but if you need more I could type them all out, no trouble. I'm surprised I didn't notice these before!
Roy

luc-github, 2016-02-09:
seems have network issue and module may be corrupted
please uninstall it and retry installation of 1.6.4 - you must not have any
issue downloading package​

bucklandgate, 2016-02-09:
Many thanks LucI have tried downloading 1.6.4 (Mac) and when I open the arduino.app I get the same issue, 15 lines of error messages immediately. I think it may well  be caused by corrupted Java and I shall try reloading that to try again. Unfortunately it's getting late now but that means you can forget about me for a day or so until I have tried AGAIN. I strangely feel a little less bothered now I'm thinking it might not be all my fault! Sleep tight Luc and sincere thanks for all your helpRoy

bucklandgate, 2016-03-27:
Update - I decided I was getting nowhere with the Da Vinci board so I've replaced it with a Ramps 1.4/Mega 2560. I will be starting to wire it (with a bit of luck!) in a few days time when I receive some 2, 3 and 4 way PCB connectors but am having problems crimping such tiny end crimps and getting them to stay in the two PCB connectors I was able to find. I'm  considering getting insulated crimps so that I can eliminate the PCB connectors. I shall be using Marlin and an LCD2004 display. That's about it for the present. Wish me luck as I still feel determined to finish this project and get printing again.Roy

luc-github, 2016-03-27:
Ok so I close issue 

good luck then:smile:

closed 2016-03-27 by luc-github