luc-github/Repetier-Firmware-4-Davinci #142: Davinci Pro

cjsnomis, 2016-02-19:
I recently bought a XYZDavinci Pro and after quite a bit of use it seems that the base software sucks and it really needs to be flashed. What can I expect time-wise for repetier to be available and stable on the Pro?

luc-github, 2016-02-19:
Time depend of availability of skilled people having a pro to work on it.

I do not have a pro currently so cannot work on it

I planned to work on Jr when donnations reach the value  because it is cheap, but pro is more expensive so no idea sorry...

luc-github, 2016-02-20:
I think topic can be closed and will be reopen when necessary

closed 2016-02-20 by luc-github

anolis, 2020-03-05:
> Time depend of availability of skilled people having a pro to work on it.
> 
> I do not have a pro currently so cannot work on it
> 
> I planned to work on Jr when donnations reach the value because it is cheap, but pro is more expensive so no idea sorry...

Hello Luc, sorry for thread mining. I recent came across a DaVinci 1.0 Pro and can write code. What do we need to do to get the firmware supported?

luc-github, 2020-03-05:
Ho boy, yes it is old
Well a lot, even JR pins are almost now known, Pro which use same board, are definitly not confirmed, but even they are
1 - SAM4E whichis the MCU of Jr and Pro is not supported by repetier/Marlin so the easiest way would be to go porting RepRap FW, which is not the easiest to use when you do not know 3D printing (which is usualy the case if you bough a XYZ printer), I saw several people willing to rollback after flashing Repetier because they realized they are not willing to go to a learning curve and not able to calibrate properly their printer to get good print, and repetier is easier to configure than reprap when you start 3D printing. 

2 - Reprap FW has limited support of screen and keypad - so need to write the code to support both  
3 - JR board and pro boards are made with 2 MCU :1 for printer FW = SAM4E 1 for temperature monitoring and control NXP M0, no FW support this kind of configuration so need to modify printer fw to monitor/control temperature thrue serial communication between SAM4E and M0
4 -  as M0 protocol is not known to control /monitor temperature need to do reverse engineering to establish it oreven better  build a new FW for the NXP M0, I personnaly never programmed an NXP MCU so I cannot say how easy is the tool chain usage
5 - based on my experience with 1.0/2.0 1.0A/2.0A and AiO the thermistor tables (extruder/bed)  will need to be created because the electronic should not be same as usual printers

so not so easy but possible with lot of time and motivation because printer will be unavailable for a while. 

That said if it "may" worth for Jr, Pro is open Filament so gain would be in controling settings  only and may be slicer (S3D already support Pro if I remember well)

Based on my experience change the board is easier - this is what I did on the JR I got.
XYZ did the great job making the FW porting hard to do  IMHO

ammoniak, 2020-04-09:
I recently got a DaVinci Pro as well... I'm not yet sure if I should replace the mother boards with a RADDS or try to get repetier on the old board working. Could you share the port-mapping?