luc-github/Repetier-Firmware-4-Davinci #143: Manual SD Mount

Hecot, 2016-03-14:
I'm using an microSD extension cable with the DaVinci 1.0 running 0.92! I'm not sure if the board got a pin to detect the SD-card when inserted. All I know is that I need to powercycle the printer to get the card recognised, so I assume there is no such pin!??

I've looked inside the CONFIGURATION.h to active the manual mount/unmount command in the display.

#define SDCARDDETECT SDCARDDETECT // original

changed to

#define SDCARDDETECT -1 //will give compile error "1" works but makes no change!?

I also checked UI.h

There is no #define for motherboard 999 ... !?

Can't get it to work - still always need to powercycle Printer when I insert the SD-Card! Can someone please help!?

luc-github, 2016-03-14:
pin is defined here :https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/userpins.h#L94

luc-github, 2016-03-15:
```#define SDCARDDETECT -1``` do not generate compilation error for me 
what is your IDE version ?

Hecot, 2016-03-15:
Thanks for the quick reply! 

As IDE I'm using 1.5.8!

In userpins.h I find the correct define for the pin:
#define SDCARDDETECT   74

But when I change configuration.h:
#define SDCARDDETECT -1 //SDCARDDETECT

Now it will compile without error (I don't know what wa wrong last time), but there is still no option in the SD-Card menu to manually mount/unmount. I still need to powercycle the printer to get the card recognized!

In pins.h I found:
#define SDCARDDETECT       14

I set it to 74 as well but still no automatic detection. I would love to have either automatic detection what would be the best case or at least manual mount/unmount!

luc-github, 2016-03-15:
pins.h is for boards other than davinci so do not pay attention to it, only  userpins.h is for davinci boards.

Do not change ```#define SDCARDDETECT SDCARDDETECT``` in configuration.h but in userpins.h
```#define SDCARDDETECT -1 //74``` because can bring some conflict in compilation

My printer is still packed and cannot verify on menu yet sorry

About automatic, I think it may not be possible with extender cable as I suspect your extender enable detect pin mechanically by being plugged, so card looks always plugged.

Hecot, 2016-03-15:
YEAH - now it's working!!! I now have a nice option on the LCD for mount/unmount - good stuff! 

I guess you are right regarding the automatic detection - there will be an switch build into the card holder, that is not activated by the extension cable. If I find that pin on the board, I could add an external push button on the outside of the housing what would ease the exchange of the card. This will be on my list for future mods ... :) 

Thanks for your great Support!

Hecot, 2016-03-20:
Houston we have a problem :) 

Today I did some small prints and used the Manual unmout/mout of the SD-Card without power cycle the printer. The print started as normal but then after a few layers the print stops with SDcard-readError. I've done many prints with this SD-Card without any issue and all I changed was the Manual unmout/mount. So I powercycled the printer and start the same print again - no problem print all well! Then the second print this afternoon - same Problem, failed when Manual unmount/Mount but succeed without any trouble when powercycle the printer and restart the same file.

Any idea?

luc-github, 2016-03-20:
Mount/Unmount is part of code I never touched - and never used it so not sure what is the problem - need to investigate in code - could be a bad initialization

Hecot, 2016-03-21:
Hmmm ... OK! But it's strange that it happen after some printing time (20 minutes)! I will now do a powercycle to be sure that the print will not get stuck - this will put me in the same situation as in the very beginning :)))

BTW: I found that "left/down/OK" will perform a soft reset when in main screen, I haven't tried if this will cure the SD-Card error - but could be a solution for me instead of switching on and off!?

luc-github, 2016-03-21:
Yes it perform a soft restart and do same initialisation as fresh start

luc-github, 2016-03-23:
If ok please close issue

Hecot, 2016-03-25:
With a  "left/down/OK" soft reset it works fine, no SD-card read errors! I will close the issue here, but in general I think the root cause is not solved! This is something repetier should have a closer look at!?

closed 2016-03-25 by Hecot

luc-github, 2016-03-25:
well the issue is that is not the original repetier FW and may be have already some fix in 0.92.8 which is not yet integrated
And because repetier do not have Davinci board it may not able to reproduce - because they alreay have a lot in their boiling plate - I am not sure they will  have a look

I will review code and see if I can see something 
when I will release the 0.92.8 based - it would be great if you can check if issue is still there

Hecot, 2016-03-25:
GREAT - of course I'm willing to repeat the test! All the best!