luc-github/Repetier-Firmware-4-Davinci #144: no extruction in the beginning of the print, Cleaning nozzle wrong cords

immunity20, 2016-03-17:
When the printing begins (simplify3d slider)
I have put the command for cleaning nozzles. 
Problem: although T1 (right extruder) moves correctly over its trash
T0 (left extruder) moves arround 5-6cm more right (over bed) instead over its trash (left one)

Extruder will print the official pre-print line (i have added the gcode) which will not extrude abs in the first 30mm.

Next when it goes to print the object, it wouldn't extrude abs for the first around 30mm
(abs 95C bed 230C)

luc-github, 2016-03-17:
what is your printer ? 2.0 or 2.0A ?

Can you share your start GCODE ? 
the no extration of the first 30mm is usualy  due to fact the noozle is empty because of gravity,  and need to add code for preparing extraction on drip box before starting the print and after temperature is reached.

immunity20, 2016-03-17:
2.0A Duo, when bed go 95 C extruder start heating to 230C
when 230C reached, extruder will extrude 30mm abs in the home position (without moving) then he correctly clean the T1 nozzle, then faulty clean (on the air over the bed) T0.

`G90
M82
M106 S153
M140 S95
M190 S95
M104 S230 T1
M109 S230 T1
G28 ; home all axes
M100
T1
G92 E0
G1 F1800.000 E-1.00000
G92 E0
G1 Z0.350 F2700.000
G1 X10.000 Y179.988
G1 F1800.000 E1.00000
G1 X10.000 Y79.988 F270.000 E15.55131
G1 X12.000 Y79.988 E15.84234
G1 X12.000 Y179.988 E30.39364
G1 X12.150 Y179.988 E30.66284
G1 F1800.000 E24.66284
G92 E0
G92 E0
G1 E-2.0000 F9000
G1 Z1.400 F1002
; layer 1, Z = 0.4
T1
; tool H0.400 W0.500
; inner perimeter
G1 X56.750 Y91.750 F4800
G1 Z0.400 F1002
G1 E0.0000 F2700
G92 E0
G1 X93.250 Y91.750 E3.0350 F919
G1 X93.250 Y108.250 E4.4070
G1 X56.750 Y108.250 E7.4420
G1 X56.750 Y91.950 E8.7973
G1 X56.750 Y91.750 F919`

luc-github, 2016-03-17:
My printer is still packed and not operational yet - should be ok this week end

You can try this version : https://github.com/luc-github/Repetier-Firmware-0.92/tree/5cbd5e584c1bf2666750d06fb9b4f08c3d23a515

This is working on my Duo so far

immunity20, 2016-03-17:
in which file the M100 command is programmed?

luc-github, 2016-03-17:
printer.cpp

limited660, 2016-03-17:
Have you changed the x max length?  On my 2.0A, the extruders are going over until the wire is nearly pinched between the carriage and end of X axis.

immunity20, 2016-03-17:
When i flashed the rom i just fixed the variables for version and model of the printer .Where i change the max length for x axis?

luc-github, 2016-03-17:
no need to change anything about max length, default values are corrects - especially for Duo 2.0

limited660, 2016-03-17:
immunity20, the reason I asked is the M100 nozzle clean uses X Max Length as the location to go to when wiping extruder 0.  If you haven't changed that manually then it should work correctly...  Can you take a picture of where the extruder ends up during the M100

immunity20, 2016-03-17:
https://drive.google.com/file/d/0BxhVohn4h3qKWGFGNHFrWVBtNzA/view?usp=sharing

limited660, 2016-03-17:
If you manually home the printer with G28 and then send G0 X199 does it go all the way to the dump area?

immunity20, 2016-03-17:
No it stops over bed (near the position it tries to clean the extruder T0) X250 wouldn't move farther

limited660, 2016-03-17:
Can you try to flash a fresh download and make sure to only change the numbers after "DAVINCI" and "MODEL" at the top of configuration.h

immunity20, 2016-03-17:
fixed , thanks :)

closed 2016-03-23 by luc-github