luc-github/Repetier-Firmware-4-Davinci #145: How to access z-plane autolevel data in EEPROM?

Hecot, 2016-03-17:
I got a Davinci 1.0A running Repetier Firmware 0.92!

After an hour of hair-tearing and trying to get the bed leveled with the automatic funktion I changed to the manual process using a sheet of paper, that worked fine. But I did a mistake by writing data to the EEPROM in the automatic funktionby pressing the wrong button. 

On my prints I've noticed Z-plane micro movements on the stepper motor from the failed automatic calibration, trying to level the bed. I thought that there must been an place in the EEPROM where these data are stored so I tried to access them in order to set them to zero. To my surprise there are not data entries that can be modified by the user? Where are these Z-level data stored? There is also an EEPROM binary file on the SD-card - what is the meaning of this file, is it the backup or the actual EEPROM settings?

My current work-arround is to switch the autolevel function off, what will stop the Z-plane micro movements! My next thought was that it could be usefull to run an autocalibartion after the manual adjustment to get the last percents to a perfectly leveled bed!?

Thanks for your help and comments!

luc-github, 2016-03-17:
to be honest the Davinci bed leveling is not really accurate - everyone agree must be turn off and use only manual leveling.
That said I use automatic leveling to check the 3 values are in same distance to fine tune my manual leveling if needed, but never save the results.

Auto bed matrix is save in EEPROM there is a command to display it but forget which one - need to check

Hecot, 2016-03-17:
Dear Luc!

Thanks for your answer - you are really a great help!!!

About the command to display the leveling: Is this a g-code or what kind of command we talk about? What would be your recommandation to set the leveling data back to zero? If the file on the SD-Card is the virtual EEPROM I could simply erase the file and I assume that the firmware will restore it with defaults. OK, then I will need to do all the settings again ... I'm not sure what I have modified in the last weeks - was happy to find a good working setup. Best would be knowing the place in the file, but it's in binary format, not plain text, therefore it's not easy to find ...

luc-github, 2016-03-17:
erase EEPROM.bin and do M502/M500 will regenerate new EEPROM file

To reset matrix there is a command: 
M322 - Reset autolevel matrix

luc-github, 2016-03-17:
Reading the code - if you select Load from EEPROM from printer menu it should display the matrix in serial monitor

Hecot, 2016-03-17:
Awesome - you are the best!!!

luc-github, 2016-03-23:
if ok please close issue

closed 2016-03-25 by Hecot