luc-github/Repetier-Firmware-4-Davinci #146: EEPROM option not available!?

Hecot, 2016-03-17:
I posted this question also to the repetier forum, but didn't get any response so far. I know that this is a bit off-topic here, but I hope someone will answer this question:

I'm running Repetier host V1.6.1 on two computers - one desktop, one laptop - both Win7-pro 64-bit. With both computers I can send commands to the printer (DaVinci 1.0A with 0.92 firmware) - all work well with one exception: On the deskop computer I can access the EEPROM of the printer, but on the laptop this option is not available (it's just light gray and could not be selected)!?

The only difference I can see is that the laptop got the Arduino software for flashing the firmware on it, on the desktop there is only the Repetier host software installed. Could this be the reason that the Arduino software or driver settings will cause not allowing to access the EEPROM? Or is there a special option for accessing the EEPROM.

If this would be the case, what would be the countermeasure so I have access to the EEPROM also with my laptop?

Thanks and sorry for bothering you with this off-topic question!

luc-github, 2016-03-17:
No - I guess it is more a setting in repetier host
does M205 works on both ?

Hecot, 2016-03-18:
Dear Luc,

M205 is working and I can send commands to the printer ... but still can't access the EEPROM from Repetier host ... the option stays gray!

[image]

luc-github, 2016-03-18:
looks like repetier host is waiting for something enable it
I never got issue - did you try to disconnect and connect to see what display in serial log ?

Hecot, 2016-03-18:
Here is the display of the log from disconnecting and connecting again!

[image]

luc-github, 2016-03-18:
check your communication settings / or try another usb port, you may need to decrease the baud rate, seems you got "ile" istead of "file" and "wai" instead of "wait", so it is obvious you have communication issue

Hecot, 2016-03-18:
Dear Luc,

I reduced the Baud rate to lowest value 9600 ... makes no difference - I tried a different cable and USB port, but still have this lost letters in the serial port console. Once out of a sudden the EEPROM option was available but when pressing the button it took 20 seconds to load the values ... some of them became red (never seen this before). When trying a second time it's back gray again ... 

Looks like there are serious communication problems ... but beside what I have tried I see no other help! Do you have any more idea how to solve this? Strange that on my Desktop it's working fine ... maybe there is a grounding issue?

luc-github, 2016-03-18:
I think you USB has some noise - laptop USB are more sensitive because everything is smaller and close compare to Desktop and if EMI is not correct you may have noise on signal.

be sure printer and pc/host  use same speed (115200 is usually ok ) 

Some Idea : 
- check if you have some option in BIOS for USB you can play 
- Be sure you wifi / bluetooth are off and no ethernet cable around same for AC adaptor cable or even light cable, all these can bring noise

luc-github, 2016-03-23:
If no more question - please close issue

Hecot, 2016-03-25:
I tried all that come to my mind - switched off every power consuming devices like lightning, wifi etc. still have the same issue of unstable connection! Anyway - I'm closing this isssue as I think it will only regard my poor Macbook!

closed 2016-03-25 by Hecot