luc-github/Repetier-Firmware-4-Davinci #147: HAl.h file malfunctioning

iadbd100, 2016-03-26:
This hhuge error message is howing if i click "upload"

Arduino: 1.6.5 (Windows 7), Board: "Arduino Uno"

HAL.h: In static member function 'static void HAL::startWatchdog()':
HAL.h:744: error: 'WDT_MR_WDRSTEN' was not declared in this scope
   timeout = WDT_MR_WDRSTEN | WDT_MR_WDV(timeout) | WDT_MR_WDD(timeout);
             ^
HAL.h:744: error: 'WDT_MR_WDV' was not declared in this scope
   timeout = WDT_MR_WDRSTEN | WDT_MR_WDV(timeout) | WDT_MR_WDD(timeout);
                                                ^
HAL.h:744: error: 'WDT_MR_WDD' was not declared in this scope
   timeout = WDT_MR_WDRSTEN | WDT_MR_WDV(timeout) | WDT_MR_WDD(timeout);
                                                                      ^
HAL.h:745: error: 'WDT' was not declared in this scope
   WDT_Enable (WDT, timeout);
               ^
HAL.h:745: error: 'WDT_Enable' was not declared in this scope
   WDT_Enable (WDT, timeout);
                           ^
HAL.h: In static member function 'static void HAL::stopWatchdog()':
HAL.h:748: error: 'WDT' was not declared in this scope
     inline static void stopWatchdog() {WDT_Disable (WDT);}
                                                     ^
HAL.h:748: error: 'WDT_Disable' was not declared in this scope
     inline static void stopWatchdog() {WDT_Disable (WDT);}
                                                        ^
HAL.h: In static member function 'static void HAL::pingWatchdog()':
HAL.h:753: error: 'WDT' was not declared in this scope
   WDT_Restart (WDT);
                ^
HAL.h:753: error: 'WDT_Restart' was not declared in this scope
   WDT_Restart (WDT);
                   ^
In file included from Repetier.h:370:0,
                 from Commands.cpp:22:
Printer.h: In static member function 'static void Printer::disableXStepper()':
HAL.h:147: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                 ^
Printer.h:374:9: note: in expansion of macro 'WRITE'
         WRITE(X_ENABLE_PIN, !X_ENABLE_ON);
         ^
HAL.h:147: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                                                                                               ^
Printer.h:374:9: note: in expansion of macro 'WRITE'
         WRITE(X_ENABLE_PIN, !X_ENABLE_ON);
         ^
Printer.h: In static member function 'static void Printer::disableYStepper()':
HAL.h:147: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                 ^
Printer.h:385:9: note: in expansion of macro 'WRITE'
         WRITE(Y_ENABLE_PIN, !Y_ENABLE_ON);
         ^
HAL.h:147: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                                                                                               ^
Printer.h:385:9: note: in expansion of macro 'WRITE'
         WRITE(Y_ENABLE_PIN, !Y_ENABLE_ON);
         ^
Printer.h: In static member function 'static void Printer::disableZStepper()':
HAL.h:147: error: 'g_APinDescription' was not declared in this scope
 #define WRITE(pin, v) do{if(v) {g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;} else {g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin; }}while(0)
                                 ^
Printer.h:395:9: note: in expansion of macro 'WRITE'
         WRITE(Z_ENABLE_PIN, !Z_ENABLE_ON);
         ^
HAL.h:147: error: 'g_APinDescription' was not declared in this scope
[... 407 lines cut ...]
             if (!(READ(UI_BACKLIGHT_PIN))) WRITE(UI_BACKLIGHT_PIN, HIGH);
                   ^
HAL.h:145: error: 'g_APinDescription' was not declared in this scope
 #define READ(pin)  PIO_Get(g_APinDescription[pin].pPort, PIO_INPUT, g_APinDescription[pin].ulPin)
                            ^
Commands.cpp:2098:54: note: in expansion of macro 'READ'
                             if(EEPROM::busesensor && READ(FIL_SENSOR1_PIN))uid.executeAction(UI_ACTION_NO_FILAMENT,true);
                                                      ^
HAL.h:145: error: 'PIO_INPUT' was not declared in this scope
 #define READ(pin)  PIO_Get(g_APinDescription[pin].pPort, PIO_INPUT, g_APinDescription[pin].ulPin)
                                                          ^
Commands.cpp:2098:54: note: in expansion of macro 'READ'
                             if(EEPROM::busesensor && READ(FIL_SENSOR1_PIN))uid.executeAction(UI_ACTION_NO_FILAMENT,true);
                                                      ^
HAL.h:145: error: 'PIO_Get' was not declared in this scope
 #define READ(pin)  PIO_Get(g_APinDescription[pin].pPort, PIO_INPUT, g_APinDescription[pin].ulPin)
                                                                                                 ^
Commands.cpp:2098:54: note: in expansion of macro 'READ'
                             if(EEPROM::busesensor && READ(FIL_SENSOR1_PIN))uid.executeAction(UI_ACTION_NO_FILAMENT,true);
                                                      ^
'NVIC_EncodePriority' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

luc-github, 2016-03-26:
you compile for " Board: "Arduino Uno"" not for Due that is the error

iadbd100, 2016-03-26:
got it. tnx :)

iadbd100, 2016-03-26:
But now this error msg is showing

Arduino: 1.6.5 (Windows 7), Board: "Arduino Due (Native USB Port)"

arm-none-eabi-gcc: error: C:\Users\Amin\AppData\Local\Temp\build147917029105684617.tmp/core/syscalls_sam3.c.o: No such file or directory
arm-none-eabi-gcc: error: C:\Users\Amin\AppData\Local\Temp\build147917029105684617.tmp/core.a: No such file or directory
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

luc-github, 2016-03-26:
someone got this issue before - seems the path is not correct in arduino settings
which version do you use of 1.6.5 the r5 ? the installer or the zip ?

iadbd100, 2016-03-26:
using zip. version 1.6.5

luc-github, 2016-03-26:
may be it is bug of zip version as I never got it with installer version
anyway here the fix found to workaround : https://forum.arduino.cc/index.php?topic=357675.0

iadbd100, 2016-03-26:
one trouble fixed, another still exist

arm-none-eabi-gcc: error: C:\Users\Amin\AppData\Local\Temp\build5163511974471614834.tmp/core.a: No such file or directory
Error compiling.

can you give me the link of the installer version?

luc-github, 2016-03-26:
https://www.arduino.cc/en/Main/OldSoftwareReleases#previous

iadbd100, 2016-03-26:
that solved the issue. scatch is uploading, shows "done uploading" but nothing happens to the printer. it remains just it was after flushing.

iadbd100, 2016-03-26:
I think total files are not actually uploaded into the arduino, rather it stops suddenly. but shows " done uploading".

iadbd100, 2016-03-26:
Issue solved. Many many thanks for helping :)

closed 2016-03-30 by luc-github