luc-github/Repetier-Firmware-4-Davinci #148: Can any Da Vinci firmware be upgraded to repetier?

Iotatronics, 2016-03-30:
I am receiving a Da Vinci 1.0 tomorrow (Friday Apr 1) and i know from the seller that the serial number starts with 3F10A. As per [this thread on voltivo](https://forum.voltivo.com/showthread.php?tid=8281) there was some issues regarding machines with this serial number, have those been resolved and will the instructions in readme.md still apply to my machine?

Secondly, I've also read that there was issues when flashing from 1.0.x to repetier and that one must have 1.1.x or 1.2.x on their Da Vinci in order to flash it to repetier. Is this still the issue? I don't know the firmware version on my printer yet since it hasn't arrived, but the readme.md makes no mention of either the serial or firmware being a restriction. All it mentions is that you need to change some #define constants to match your machine.

luc-github, 2016-03-30:
1.0A is supported, it is mentioned in readme and it is explained in video linked in readme

About fw restriction, I am not aware of it, because you erase it when you use jumper, so it should not have impact, but again no feedback on issue linked to stock FW version.

The only failure reported were always due to people not following instructions or when board has issue

Iotatronics, 2016-03-30:
I read that 1.0A is supported, as well as 2.0A (i've read over the readme). But i wasn't away that 1.0A and serial number 3F10A are the same thing. Or does 1.0A have to do with the firmware. I was under the impression that the machines broke down into 1.0 and 1.0A and they use the same firmware but have different hardware.

luc-github, 2016-03-30:
3F10A is for 1.0A 
1.0A and 1.0 do not use same FW, hardware are different (for SDcard, drivers are soldered, etc...)
same for 2.0A and 2.0, 
Actually FW versioning is pretty messy on Davinci so not easy to follow

But you can check the feedback on 1.0A (https://github.com/luc-github/Repetier-Firmware-0.92/issues/93), the ones that are not yet supported are the 1.1plus (with touch screen), the pro, and the Jr

luc-github, 2016-04-01:
if no update I close issue

closed 2016-04-01 by luc-github

Iotatronics, 2016-04-01:
Sorry and thanks for all your help. I will try flashing it within the next few days and get back to you how it turned out

luc-github, 2016-04-01:
no problem, good luck and happy printing ^_^

Iotatronics, 2016-04-02:
So last night i flashed repetier to my new Da Vinci 1.0A with zero problems! Just followed the instructions (i chose to use arduino 1.5.8) and it worked like a charm. I had trouble erasing the firmware initially since there is not jumper pins, but holding a piece of solder between the pads worked great.

My Da Vinci had serial number starting with 3F10A, a micro SD card with jumper J37 below it to erase the firmware. It was on firmware version 1.0.1 when i erased it and flashed repetier.

Thanks for all your effort making this great firmware. I have some work to do to get the printer working nicely as im getting large stripe marks on the outside surface of my prints, but i think that just has to do with settings to adjust.

luc-github, 2016-04-02:
good, thanks for sharing :smile:

roytoy, 2016-04-02:
hi i have a davinci 1.0a flashed it with ardrino 1.5.8 repetier 92 the light wont come on i can scroll through menu select print file it will move extruder no fillament will come out and the table will stay down any help would be apreciated

luc-github, 2016-04-02:
did you do the M502/M500 command ?

roytoy, 2016-04-02:
yes sent from repetier host and also did it from printer

luc-github, 2016-04-02:
you print from SD or from host ?

roytoy, 2016-04-02:
sd card table wont move when printing no filament comes out no lights i can move my xyz axis manually

luc-github, 2016-04-02:
you print what from SD ? davinci sample ?

roytoy, 2016-04-02:
yes wat was on there if i print from host same problem   extruder moves no print no table if i look at printer screen it say def/ 0c
                                  28/0c
                                  z: 0.00
                              printer ready

luc-github, 2016-04-02:
def/ 0c means defect on extruder you should have the message on if you connect host 
based on assumption you did not changed your hot end/ thermistor you may check your thermistor connection / cable and also check main board has the 3 lights on to confirm not short damaged the board

luc-github, 2016-04-02:
also do not use davinci sample it won't work

roytoy, 2016-04-02:
if i go to printer menu i can select things but does nothing exept i can move my axis i have 3 lights on my extruder was fine i just printed with the stock firmware n hour ago i just got tired of the cartridges there so much unused filement i cant use

roytoy, 2016-04-02:
how do i get the light on

luc-github, 2016-04-02:
when extruder defect you cannot heat - you must restart printer to reset the safety

in menu settings you have light on/off

luc-github, 2016-04-02:
check what is extruder temperature on host

roytoy, 2016-04-02:
how do i restart power switch?

roytoy, 2016-04-02:
there is a dry run setting that wont turn off

luc-github, 2016-04-02:
restart = power off then power on, this is the only way to reset dry run

luc-github, 2016-04-02:
what is extruder temperature on host ?

roytoy, 2016-04-02:
ya for some reason it wont reset i kill powe unplug cord even nothing no light no print extruder on host none of my temps are reading

roytoy, 2016-04-02:
did i install something wrong

luc-github, 2016-04-02:
it won't reset because extruder is detected giving wrong temperature for some reason

luc-github, 2016-04-02:
what version display in setting menu ?

roytoy, 2016-04-02:
it is a davici 1.0a what it says now davinci  1.0 1.15-10-30-1.1 repetier 0.92mod thank you for helping me

luc-github, 2016-04-02:
ok seems you did not flash with correct settings you select 1.0 instead of 1.0A

be sure you flash with MODEL 1 not 0 

then version will be 1.0 1.15-10-30-1.1A

luc-github, 2016-04-02:
that is why table is not correct and light is not working

roytoy, 2016-04-02:
ive tried both i have refashed twice same results i will flash back to 1.oa right now but i have same problems

roytoy, 2016-04-02:
there are no other manual commands to imput is this correct

luc-github, 2016-04-02:
no - that is why I have added this check with version number to ensure people give true answer about FW version used

luc-github, 2016-04-02:
same behavior can be due to different root cause - extruder defect error can be due to several root causes

luc-github, 2016-04-02:
temperature of extruder is in log of host when you connect as well as the error message of defect extruder
in terminal log

roytoy, 2016-04-02:
ok system version 1.15-10-30-1.1 is what it says now still same problems just reflashed it i entered 1 for 1.0 and 1 for j37 board

roytoy, 2016-04-02:
i have all 3 lights on in back to

roytoy, 2016-04-02:
is it supposed to say it a 1.0 or a1.oa

luc-github, 2016-04-02:
if system version do not end with A it means not correct for 1.0A
close arduino IDE
open configuration.h with notepad
be sure :
#define DAVINCI 1 
#define MODEL  1
save file
open repetier.ino and flash
version should finish by A

roytoy, 2016-04-02:
could it be posible for you to email me a saved file of this

luc-github, 2016-04-02:
just download the file from github and change MODEL 0 to MODEL 1 no other change is necessary

luc-github, 2016-04-02:
it is late for me now (almost 1AM) so I let you redo

roytoy, 2016-04-02:
ok im gonna erase the file i have download the other one thankyou for your help ill let you know

roytoy, 2016-04-02:
its 649 pm here

roytoy, 2016-04-02:
thank you so much for your help it must of been a corupted file everything is working i absolutely love it