luc-github/Repetier-Firmware-4-Davinci #149: remaining print time in hh:mm instead of %!?

Hecot, 2016-03-31:
Dear Luc,

You where such a great help in the past, so I have another question - sorry if this is a bit off-topic!

I'm wondering if there would be the possibility to change the remaining print display from % to hh:mm!? 

This would make more sense to me as I don't have any print server or webcam running at the moment to check from a distance.

Thanks!

limited660, 2016-03-31:
If I recall from researching this in the past, you need a way to parse the g-code to generate the time.  We would need a standard g-code that would set the time and be something that all slicers would use.  I don't know if some type of timer can be added to the firmware but if it can then it should be pretty easy to count down from a g-code provided time.

luc-github, 2016-03-31:
Progress is done by host if printing from host via M117
If printing from SDCard progress is calculated by position in file versus file size so nothing related to time ...
parsing comments looking for slicer info about total estimated time is pretty random as every slicer do not add comment and format may vary

Got same analysis to know the current layer and display it...

Hecot, 2016-04-01:
OK - got it!

closed 2016-04-01 by Hecot