luc-github/Repetier-Firmware-4-Davinci #150: Adjustment of stepper motor current?

Hecot, 2016-04-01:
Dear Luc!

Currently I'm printing transparent PETG! This filament will look much nicer when printed at lower temperatures but when I will go below 200°C my extruder stepper motor will not always be able to generate the necessary force and I will get some "klicking noise" that will result in imperfections in the print (weak spots).

I've learned from another forum, that the DaVinci board is using digital motor control so it should be possible to tweek the extruders motor driver to generate higher currents to the stepper to get a higher torque? 

Thanks for your help!

luc-github, 2016-04-01:
hi about digipot the answer is no I checked before : https://github.com/luc-github/Repetier-Firmware-0.92/issues/76

About clicking the reason is just you temperature is too loo for the melting point - and do some blob on print

Hecot, 2016-04-02:
OK - Thanks! I add this to my hardware mod list for the mainboard! As an interims solution I found that the autospeed function in slicer do a great job by keeping the extruded volume constant and adjusts the speeds accordingly. No more clicking at low temps, it seams that volumetric peaks caused this!

closed 2016-04-02 by Hecot