luc-github/Repetier-Firmware-4-Davinci #151: Repetier on Da Vinci 1.0 PRO (OctoPrint/OctoPi)

BradenJM, 2016-04-01:
I have a Da Vinci 1.0 Pro running the latest software and firmware versions. For a while, I have been attempting to use OctoPi. I had an issue [that you can find here (https://github.com/guysoft/OctoPi/issues/218 "GitHub Serial Issue"). guysoft told me this was because you can't use octopi on the stock Da Vinci firmware. If this is true, is it worth installing (or even possible) repetier on the pro? From what I know, the main appeal of repetier is to allow 3rd party filament and the ability to use other programs such as Cura or Slic3r. 

Since the Pro already allows all of this, do you think its worth it (or again, even possible) to install repeteir solely for octopi? Thanks.

luc-github, 2016-04-01:
repetier fw is not compatible with pro currently see also my answer here : https://github.com/luc-github/Repetier-Firmware-0.92/issues/142

about the question : does it worth to have a FW which allow you to tune your printer to perfection and allows printer be used with any host ?
My answer would be : Yes of course :wink:

BradenJM, 2016-04-01:
Sorry, I must have accidentally had the 'Open' Filter when I searched for Da Vinci Pro threads. 

So, would there be any way to get octoprint to work with the PRO that you know of? Thanks.

luc-github, 2016-04-01:
I do not think so - Davinci does not really use a host, XYZ host actually copy gcode file on temporary location of SDCard then printer use it to print, that is why XYZ is not necessary to be connected when printing unlike normal host, and host does not use gcode to communicate with printer but DAVXXX I forget the syntax but it is on voltivo forum if I remember well

So for octoprint with stock FW - sorry :disappointed:

BradenJM, 2016-04-05:
Just realized I never replied. Well, I appreciate your rapid answers. Thank you.

closed 2016-04-05 by BradenJM