luc-github/Repetier-Firmware-4-Davinci #152: 2nd fan support

Hecot, 2016-04-19:
Dear Luc,

I know that this is discussed before (https://github.com/luc-github/Repetier-Firmware-0.92/issues/64) but the issue is partly private and can not be reopend - so I did a new one.

I have a Davinci 1.0a and found that all the parts to control a second fan are already on the main board.There is also a cable running to the extruder for the second fan (blue cable).

[image]

On the Extruder PCB there is a connector called Fan B where a second fan can be soldered to. I've checked all the wirings with an ohmmeter and found it all working. 

[image]

The Fan A that will do the hotend cooling is connected to an transistor that can be controlled by Pin 13 on the Arduino MCU
The Fan B that could do a layer cooling is connected to an transistor that can be controlled by Pin 129 on the Arduino MCU

I'm not 100% sure on the pinning could be +/- 1 pin as the ohm meter probes are quite big - but should be correct!
I don't have a fan yet - but could connect an LED to the Extruder PCB to test. Can you please help me to get this working in the code!?

luc-github, 2016-04-19:
Fan is pin 25 (13 on mcu) and Fan2 should be pin 85 (129 on MCU)

so change ORIG_FAN2_PIN in userpins.h (https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/userpins.h#L64-65)
```
#define ORIG_FAN_PIN        25
#define ORIG_FAN2_PIN       85
```

set REPURPOSE_FAN_TO_COOL_EXTRUSIONS to 1 in configuration.h (https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L26)

```#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0 ```

change  ```#define EXT0_EXTRUDER_COOLER_PIN -1``` to ```#define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN ``` here : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1594

It should to the trick (in theory)

Hecot, 2016-04-20:
Dear Luc,

Thanks for the instructions - it's WORKING!!!

There is only one thing I needed to change: because of defining 

`#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS  1`

the fan definitions swaped and I needed to "double invert" to get the correct assignment: [https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1594](url)

`#if REPURPOSE_FAN_TO_COOL_EXTRUSIONS==1
#define FEATURE_FAN_CONTROL 1
 #define FAN_PIN **ORIG_FAN2_PIN**
  #if DAVINCI==2 || DAVINCI==3
      #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
      #define EXT1_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
    #else //DaVinci 1.0
        #define EXT0_EXTRUDER_COOLER_PIN **ORIG_FAN_PIN** //Warning need to add a permanent fan with power supply to cool extruder
    #endif`

luc-github, 2016-04-20:
or you could invert the pins  assignement  :wink:
happy for you it works - happy printing :smile:

luc-github, 2016-04-26:
I close it if no update

Hecot, 2016-05-12:
Dear Luc,

Yesterday I received the second fan and tried to get it working - since the last flashing of the software I did no print - just tried the M104 command to switch the second fan on/off - this is working great. 

When starting to print the same g-code I still had on my SD-Card the extuder pushed the filament backwards out of the printer before start to print!? I never had this before! Then I tried to undo all the changes by overwriting all the files and just swop the pins as you suggested. This solved the problem with the push-back of the filament but also leads to massive errors - the printer starts printing and then after a few commands it is resetting by it's own. I've flashed a couple of times - no errors with programming and memory comparing. 

Finally I returned to the standard version and just set "repurpose fan" to 0 and the printer is working as usual!

I assume this configuration is not intended to use? Maybe you have ideas what to do or some else have experience on how to use the second fan with the onboard parts!?

luc-github, 2016-05-12:
well repurpose is working on 1.0 and 2.0 and used , but not by me.
changes do not affect loading unloading in theory, just fan pins assignment, loading unloading is handled by slicer, I am out now so hard to review code, you may check if auto retract is enabled

luc-github, 2016-05-12:
one thing I would suggest is to revert all changes and connect your second fan and do a print test to see if is ok
if all ok then enable the second fan using the M42 command and redo another test print

if it is ok it means electronique is ok but code is not

luc-github, 2016-05-24:
hi I may have a clue of the issue
please in configuration.h 
change ORIG_FAN_PIN/ORIG_FAN2_PIN by their numerical values - I think it is the problem

luc-github, 2016-06-05:
Any update on this issue ?

Hecot, 2016-06-06:
Dear Luc!

Currently my Davinci is in parts (I'm in the middle of a hardware modification, changing to a bouden extruder and E3D hotend). 
But the second fan issue is still on my radar and I will check when the mod is finished. Should be done within the next days, I'm only waiting on a missing part ... !

Talk to you soon!

Hecot, 2016-06-14:
Dear Luc!

I now have time to check the 2nd fan issue. 

When reading your instructions again I didn't understand what you mean by "change ORIG_FAN_PIN/ORIG_FAN2_PIN by their numerical values" in configuration.h ... I see no definitions of the pins in configuration.h!? Therefore I changed the Pins in userpins.h - I think this is what you had in mind!? In userpins.h it is now:

``//fan PINS
#define FAN_BOARD_PIN -1
#define ORIG_FAN_PIN        85  //25
#define ORIG_FAN2_PIN       25  //4

I can control the fan now and everthing is working fine - but I still have these random resets I had before??? Even if I'm not printing every few minutes the printer is reseting itself by it own! 

Please let me know if I did it wrong and maybe you can explain a bit more in Detail!? THANKS!!!

Hecot, 2016-06-14:
Dear Luc,

Usually this is the compilers job to replace all the defines with the actual values ... !? 

Anyway  - I did what you said and replaced the definitions with the numerical values in configuration.h. But unfortunatly it makes no difference! The printer is still resetting itself after a few minutes - totally random! 

Any other idea what I can try? Is there any way of debugging - like a crashfile or simular?

luc-github, 2016-06-14:
Yes but need some order (file with define is used by another one, not each file use each others)  and arduino got several preprocessor bugs in each release - so was just a guess
  
about test did you do these (https://github.com/luc-github/Repetier-Firmware-0.92/issues/152#issuecomment-218716063) to see if the issue is hardware or software ?

you also can disable watchdog to see if issue is due to watchdog somehow

Hecot, 2016-06-14:
The printer is working without any issue when I use it without the repurpose function, so I'm sure there is no hardware issue ...

luc-github, 2016-06-14:
without the repurpose, it only use one fan, so you cannot be sure... - the tests I gave you help to make both fans working without using repurpose, to validate definitly it is not hardware and isolate some software part
also several people used this repurpose on 1.0 and 2.0 without issue, so I also can say I am sure it is not FW issue :winkle:
The purpose of the tests I suggested is to verify all situations hard and soft - could have many root causes from both parts

Hecot, 2016-06-16:
I got it WORKING!!!

Here is what I did: As it seams that the repurpose feature causing the trouble I set repurpose to 0 in configuration.h
[image]

changed back the order in userpin.h
[image]

And added these defines to configuration.h
[image]

Now all is working fine - thanks for your Support!!!

closed 2016-06-16 by Hecot

luc-github, 2016-06-16:
Ok so the only reason I see are these lines :
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L241-L250
```
// You can use either PWM (pulse width modulation) or PDM (pulse density modulation) for
// extruders or coolers. PDM will give more signal changes per second, so on average it gives
// the cleaner signal. The only advantage of PWM is giving signals at a fixed rate and never more
// then PWM.
#define PDM_FOR_EXTRUDER 1
#if REPURPOSE_FAN_TO_COOL_EXTRUSIONS
#define PDM_FOR_COOLER 0
#else
#define PDM_FOR_COOLER 1
#endif
```

rlagg, 2016-08-30:
Thanks for the great work Hecot! and Luc, of course.

Just a note that the pin outs annotated on the photo of the Extruder PCB (top of page) are inaccurate.
Pin 3 (marked as NTC) is GROUND
Pin 4 (marked as GND) is the 5V supply for the fans.

Found this while designing a board to control 12V fans from the existing 5V logic, fortunately BEFORE any smoke was generated.

biggreenogre, 2016-09-09:
A note on the pinouts annotated on the picture of the extruder PCB above which I felt might save someone some grief.

*   **PIN 4** (marked "GND") is actually the **5V supply** for the fans.
*   **PIN 3** (marked "NTC") is **GND**

Found this while tracing the fan circuits during design of a board to control the 12V (E3D) fan from the original 5V circuit. Fortunately, I found this *before* any smoke got out!
 
Thanks for the work @Hecot (and @luc-github of course)! I've been upgrading my 1.0A and will be adding a second fan for cooling. This looks like the best approach with both fans under f/w control.

nicksears, 2016-10-05:
I'm trying to figure a little more out about the pins on the MCU and how they correlate to the pins arduino uses. Any suggestions? I posted [here ](https://forum.voltivo.com/showthread.php?tid=8144&pid=18131#pid18131) with what I've found so far, but it seems like the pins I want to use (that the davinci is natively able to use) are not available. 

For instance, I'm using these diagrams [(due schematic)](https://www.arduino.cc/en/uploads/Main/arduino-Due-schematic.pdf), [(due pinout0](http://www.robgray.com/temp/Due-pinout-WEB.png) and [(sam3x datasheet)](http://www.atmel.com/Images/Atmel-11057-32-bit-Cortex-M3-Microcontroller-SAM3X-SAM3A_Datasheet.pdf) but when I trace back the pin we're using in this post (arduino pin 85), I get pin 129 on the MCU. This is the UOTGID pin on the Arduino Due (no idea what that is other than I see it on the schematic) but here is says it's pin 85. Can anyone help?

luc-github, 2016-10-05:
check the pinout file : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/pinout.xlsx
you have used pins number with mcu pins number

nicksears, 2016-10-05:
so awesome, the amount of work on this project never ceases to amaze. Is there any good info out there on how the pins are defined in the first place? I understand the numbering of the MCU, but how do we actually tell arduino that MCU pin is now arduino pin 85? is it something in (Repetier-Firmware-0.92-master\src\ArduinoDUE\AdditionalArduinoFiles\1.5.8\hardware\arduino\sam\variants\arduino_due_x) or is it something more fundamental? Thanks!

luc-github, 2016-10-05:
yes it is defined in variant.cpp , using pin descriptor and eventually specific peripheral

DaVincIan, 2017-04-01:
Hecot

Thanks for the work and Thingiverse part for this mod. I did my first print today, using your E3D extruder mount. I find the extruder a little low, so I'm raising it by 3 mm.

As for the 2nd fan software changes. On 16 June 2016 your commented:

And added these defines to configuration.h
  #endif
#else
  #define FAN_PIN ORIG_FAN2_PIN
  #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN_PIN
  #define FEATURE_FAN_CONTROL 1
#endif

In the Da Vinci firmware "variant" I have, the line numbers do not correspond with your line numbers for the configuration.h file. I listed the full code segment below. Did I make the changes in the correct place?

/** Should support for fan control be compiled in. If you enable this make sure
the FAN pin is not the same as for your second extruder. RAMPS e.g. has FAN_PIN in 9 which
is also used for the heater if you have 2 extruders connected. */
#if REPURPOSE_FAN_TO_COOL_EXTRUSIONS == 1
#define FEATURE_FAN_CONTROL 1
#define FAN_PIN ORIG_FAN_PIN
  #if DAVINCI == 2 || DAVINCI == 3
      #undef EXT0_EXTRUDER_COOLER_PIN
      #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
      #define EXT1_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
  #else //DaVinci 1.0
        #undef EXT0_EXTRUDER_COOLER_PIN
        #define EXT0_EXTRUDER_COOLER_PIN -1 //Warning need to add a permanent fan with power supply to cool extruder
//        #define EXT1_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
  #endif
#else
//  #undef  FAN_PIN
//  #define FAN_PIN -1
//  #define FEATURE_FAN_CONTROL 0
  #define FAN_PIN ORIG_FAN2_PIN
  #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN_PIN
  #define FEATURE_FAN_CONTROL 1
#endif

luc-github, 2017-04-06:
this configuration use the one pin to control both extruders fans (on/off at 50 degres) to cool down the extruders
and one pin to contol the third fan to help to bridge when necessary using M106/M107 commands

as far I know on Davinci only one pin support PWM which will allow to control fan using M106/M107 and also control the speed with S parameter

nicksears, 2017-04-06:
If you want to use another fan, the easiest thing to do is hijack the pin
output to the cartridge. Can't remember the number, check the pin sheet.
Unless you have a 1.0a I believe, it won't have an extra plug wired up to
the extruder. Even then it's not used you'd have to populate the board with
a 2n222 transistor.

DaVincIan, 2017-04-06:
My aim: I want to have an extruder fan for cooling my E3D and an extrution fan. From what I gather Hecot achieved this.

**With the firmware code:**

#if REPURPOSE_FAN_TO_COOL_EXTRUSIONS == 1
#define FEATURE_FAN_CONTROL 1
#define FAN_PIN ORIG_FAN_PIN
  #if DAVINCI == 2 || DAVINCI == 3
      #undef EXT0_EXTRUDER_COOLER_PIN
      #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
      #define EXT1_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
  **#else //DaVinci 1.0 With **#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 1** 
        #undef EXT0_EXTRUDER_COOLER_PIN
//        #define EXT0_EXTRUDER_COOLER_PIN -1 //Warning need to add a permanent fan with power supply to cool extruder
        **#define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN**
  #endif**
**_#else //With **#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0** 
  #undef  FAN_PIN
//  #define FAN_PIN -1
//  #define FEATURE_FAN_CONTROL 0
  **_#define FAN_PIN ORIG_FAN2_PIN
  #define FEATURE_FAN_CONTROL 1_**
#endif_**

With **#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 1** 
  #else //DaVinci 1.0 excetutes and I can control FAN 1 using M106

With **#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS 0** 
  the last #else // in the code segment above excetutes and I can control FAN 2 using M106

_(the two cases might be swapped around, I'm not 100% sure I remember it correctly)_

M106 is equivalent to using the fan control in Repetier. 

But in both cases above, I cannot turn the "other" fan on. I only have control of a fan at a time. Hence I do not know how Hecot does it. Does the extruder fan get turned on by the FW once a print is in progress and the hot end heats up? If so, then it is fine. I was just hoping to test both fans (on together) before I print.

FYI, I did have PWM control on both pins (85 & 25), but only one at a time unfortunately.

luc-github, 2017-04-06:
the extruder fan is controled by temperature - starting on at 50 degres / off under 50 degres
just heat your extruder a little and fan should start when temp is over 50 degres

Hecot, 2017-04-06:
Sorry for the late reply! I was moving last few days - now back online :)

On my Davinci 1.0A I got the second fan working with PWM and can control it separetly via Repetier remote software and with the control knobs on the front panel. The slic3r software will add the necessary g-code to turn it on when printing bridges etc. 

Please let me know how I can support best or what should I do to help you get this working! My firmware version may be older than yours and it is some time ago so I can not remember all details. Maybe it would help if I send you my working firmware version so you start from there? Please let me know

DaVincIan, 2017-04-07:
Thanks for everyone's participation. I managed to solve the riddle.

My "suspicions" were correct. For anyone else that plans to do this worthwhile modification, it works like this (with the mod implemented and FW updated as described above): You can test/control Fan 1 with the M106 gcode whisch is equivalnet to using the fan slider in Repetier/Manual Control tab. on the same tab, when you turn on the extruder , Fan2 will run.

I'm busy finalizing my upgrade, based on the excellent wok by Hecot et al. Once done, I'll post the details.

For now, if you want to use a 12 V fan...all you need to do is change the 5V supply pin and  wire it to the 12 V supply. If you are not electronic savvy, wait for my post...I tested it and the current is <= 200 mA. The transistor can handle this easily.

Hecot, 2017-04-07:
Great to hear you got it working! Happy printing!!!

wesk1, 2017-10-24:
@davinclan, I need some help with this, can you guide me to get this working or if you can share your configuration and userpins file.

Thanks

DaVincIan, 2017-10-24:
Good day Wesk1.

I made quite a bit of changes, but all essentially as per the post. It is
also very easy to choose between 5 V or 12 V fan. How far are you i the
project? What are you struggling with?

Did you get the hardware changes done/correct?

I'll need to look at home for the detail and get back to you. I have the
one fan cooling my E3D extruder and the secong cooling the print...very
needed for small PLA prints. This is controlled vis slicer settings (and
can be done from the front panel software).

I'll try and respond by tomorrow, or even tonight, once I understand your
request a bit better.

(I just looked at my last post again...I mentioned most of this in there).

Regards
Ian

wesk1, 2017-10-24:
@davinclan, hello mainly the firmware code I see different suggestions but I’m not sure which one works, I do want a similar setup like how you have it, i will try the suggested code in your last post. Thanks

DaVincIan, 2017-10-24:
Yes, it is confusing...but I'll get mine to you ASAP.

wesk1, 2017-10-26:
@davinclan, thanks

DaVincIan, 2017-10-26:
Hi Wesk

Sorry I had a busy day...

If I remember correctly, one need to make 2 changes in the
"Configuration.h" file. I added my complete arduino code in the zip file.
all of the best.

My next step is to add auto leveling. I have some hardware ideas...some
work, some not so good...

The first is "early" in the code:

// ################## EDIT THESE SETTINGS MANUALLY ################

#define DAVINCI 1// "0" if not DAVINCI, "1" For DAVINCI 1.0, "2" For
DAVINCI 2.0 with 1 FAN, "3" For DAVINCI 2.0 with 2 FAN, 4 for AiO (WITH NO
SCANNER SUPPORT)
#define MODEL  1//"0" for first generation (jumper JP1 to reset ) , "1" for
new generation   (jumper J37 to reset)
#define REPURPOSE_FAN_TO_COOL_EXTRUSIONS* 0* //Setting this to 1 will
repurpose the main Extruder cooling fan to be controlled VIA M106/M107
                                           //Warning: for DaVinci 1.0 need
to add a permanent fan with power supply to cool extruder

I know want want to set it to "1", but this needs to be 0.

Second change/update...search for "REPURPOSE_FAN_TO_COOL_EXTRUSIONS" it's
about line number 1868. The code is changed as below.

/** Should support for fan control be compiled in. If you enable this make
sure
the FAN pin is not the same as for your second extruder. RAMPS e.g. has
FAN_PIN in 9 which
is also used for the heater if you have 2 extruders connected. */
#if REPURPOSE_FAN_TO_COOL_EXTRUSIONS == 1
#define FEATURE_FAN_CONTROL 1
#define FAN_PIN ORIG_FAN_PIN
  #if DAVINCI == 2 || DAVINCI == 3
      #undef EXT0_EXTRUDER_COOLER_PIN
      #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
      #define EXT1_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
  #else //DaVinci 1.0
        #undef EXT0_EXTRUDER_COOLER_PIN
//        #define EXT0_EXTRUDER_COOLER_PIN -1 //Warning need to add a
permanent fan with power supply to cool extruder
        #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN //Warning need to
add a permanent fan with power supply to cool extruder
  #endif
#else
  #undef  FAN_PIN
//  #define FAN_PIN -1
//  #define FEATURE_FAN_CONTROL 0
  #define FAN_PIN ORIG_FAN2_PIN
  #define FEATURE_FAN_CONTROL 1
#endif

I hope this help.

The extruder fan is controlled permanently. It turns on when the extruder
is on. The second fan is controlled in the slicer settings, and can be
controlled on the GUI.

wesk1, 2017-10-26:
@davinclan, no worries thanks for your help, i will try this.

bogaskilop, 2018-06-19:
Hi all 

I recently get one of that DaVinci 1.0 printers   and I'm looking for fast upgrade  just to change firmware and add print cooler fan.  Is is posible to add this future to DaVinci 1.0 ?  PCB on extruder looks different that from first post

so repetier will turn on and off print cooling fan ?

Hecot, 2018-06-19:
@bogaskilop : My printer is a Davinci 1.0A, so when the PCBs are diffrent I guess you have Version 1.0 - without A!? The older version also have some parts on the main PCB that were not assembled (one transistor + some Rs and Cs. I remember a post in a different forum explaining how to populate the missing parts. 

About the other question: The slicer software will add the necessary g-code to turn it on/off when printing bridges etc.

bogaskilop, 2018-06-20:
yesterday i manage to install repetier and was printing ok  but today it stops mid print  or mostly at Z9.60 LOL can not find whats wrong . some times when i send gcode from S3D it heat up and nothing happening then ...

Hecot, 2018-06-20:
It looks like there is a transmitting issue - I would try printing from SD-card and check the g-code file with a text editor to check if there is any g-code above Z9.60. Also try a fresh SD-card - I had a broken SD-card some times ago, working 10 times and then failing in the middle of the print ...

bogaskilop, 2018-06-20:
today i put new sd card 16 gb sandisk into davinci  but if those problems will appear in next few dyasi will think about swapping mainboard for some MKS .

Is there and option to connect BLtouch to Davinci 1.0 board  to any empty pins ?
I just dont want to check bed level any more on that printer :)

luc-github, 2018-06-20:
you changed SD, but you still print from S3D? or your print from SD?

Hecot, 2018-06-20:
A new borosilikat glas fixed all my bed leveling issues so I deactivated the bed-leveling completly. I would suggest you open a new topic for BLtouch as this becomes OT.

bogaskilop, 2018-06-20:
s3d ,  waiting for sd extension

luc-github, 2018-06-20:
Printing from USB on Davinci is really subject to several random issues - USB cable is very sensitive to any electric disturbance