luc-github/Repetier-Firmware-4-Davinci #153: X offset and Y offset values are not stored

szlovak, 2016-05-02:
Hello

Do I have to hard code general extruder position? I have Da Vinci 2.0, single fan and print starts always too much on y and too less on x .When I put X offset 9 mm and Y offset -3 mm everything is fine till next turn on of the machine. I'm not talking about ext 1 vs ext 2 position - it's fine.

luc-github, 2016-05-02:
Xmin / Ymin / Zmin are position at home = offset, they are modifiable in EEPROM from repetier or from menu printer: settings/position, value on 2.0 are correct as I have a 2.0 single fan that is my reference.

I suggest you to review your host/slicer settings that usually cause this kind of issue = bed not centered

szlovak, 2016-05-02:
I've just changed X min and Y min and it was printed at wrong place. Mine printer has maybe misplaced sensors? Maybe different series has some hardware differences?

Before I could correct it in Simplify 3D, but I don't have it now, there was an option for it. I went back to Cura engine and there isn't such setting, so I figured out that X and Y offset in printer puts it back where it should be, but these values are not stored. I mean settings> position> x and y offset. I stored values to EEPROM, back and forth, these values are intact till next re-powering of printer. In Repetier host application there are no such values when I browse EEPROM content through USB

luc-github, 2016-05-02:
if you store to EEPROM and they are not saved it may mean your SDCard has issue of EEPROM.bin  file is locked in some way, to released the file :
1- power down  printer and unplug power cable
2 - wait 5 ~10 min
3 - replug cble and power on
4 - do a load fail safe from settings

for S3D there is still settings for offset on 3.0.2
[image]

szlovak, 2016-05-02:
I meant that there is no such field as "X offset" and "Y offset" at all in EEPROM.bin so it can't be stored, or at least RH doesn't recognize this field or displays these values. 

I did what you asked for, X offset nor Y offset is stored at all.

Exactly I set "origin offset" when I was using S3D

luc-github, 2016-05-02:
in EEPROM it is called X Home pos and Y Home pose
[image]

szlovak, 2016-05-02:
I don't have X home nor Y home in RH 1.6.1 (the newest) 
[image]

But anyway it is the same as X min and Y min, that  I set through buttons on machine that didn't work for me.. Her are values 0.7 and 0.7  for a test.

luc-github, 2016-05-02:
X min pos and Y min pos are the same as X home pos and Y home pos so you can change the value in EEPROM

You may need to delete EEPROM.bin file on SDCARD if you cannot modify any value in EEPROM
when you change light state to Off - EEPROM keep value after a restart ?

szlovak, 2016-05-02:
I can change X min , but it has no influence on print position. Yes, no problem with any  other settings, at least which one I was using, like acceleration, lights. Can you save this X offset value to some value and have it keep after machine re - power?

luc-github, 2016-05-02:
yes it is supposed to work Xmin and Y min are saved and read from EEPROM at boot - I will check on my printer tomorrow if I can reproduce your issue

luc-github, 2016-05-02:
when you change  X min to 9 mm and Y min to  -3 mm in EEPROM using repetier and restart printer 
Xmin and Ymin are back to 0 ?

szlovak, 2016-05-02:
x min and y min are zeros all the time or the values I set them to manually, but like I said they don't do much, setting them sets only limit for something else I think like the object had maximum dimensions of bed, then they would limit the extruder to not to go over the edge of bed I think.

luc-github, 2016-05-03:
sorry I am confused:
your issue is not clear, seems you can change Xmin and Ymin (which are same as X offset and Y offset) in EEPROM - but you say you cannot can you clarify ?

Also I see you have in EEPROM X max length set to 150 and  Y max length set to 200 why ? these settings are not bed dimensions but extruder movement distance, mainly to reach wipe blade. 
why did you changed default values ? these ones are not corrects, I think you mess up between FW settings and slicer settings, FW settings no need to be changed unless you do hardware modifications.

szlovak, 2016-05-03:
I'm talking about changing parameters in printer. I can change X min and Y min, which are stored fine in eeprom.bin. Print starts anyway without respecting these values. I cannot store X offset and Y offset , but they do work when I want to shift print on the table . I think offset and min values are not doing the same thing, are they? I can't see it

I've restored again to factory settings to revert to original settings for X max and Y max as you say they are not good these I set, OK. Saved to eeprom.bin, started to print and position is still not in the center. Here is my startup gcode:

G28 ; Home extruder
M107 ; Turn off fan
G1 Z0
G90 ; Absolute positioning
M82 ; Extruder in absolute mode

{IF_EXT0}M104 T0 S{TEMP0}
{IF_EXT1}M104 T1 S{TEMP1}
{IF_BED}M190 S{BED}
G92 E0 ; Reset extruder position
{IF_EXT0}M109 T0 S{TEMP0}
{IF_EXT1}M109 T1 S{TEMP1}
{IF_EXT0}T0
{IF_EXT1}T1

Here are settings of RH :+1: 
[image]

Probably if I would touch it , it could center the print , like does description says clearly, but I want to do it in printer, to have it stored permanently.

luc-github, 2016-05-03:
Ok so as said it is slicer / host issue
1 - Xmin and Ymin values must be identical in FW and repetier settings, they are the ones for reference
2 - X Offset setting and Y offet setting are for extruder2 position compare to extruder 1

[image]

now that said - I just realize you wrote you use 0.92.6 - this one is experimental and still under code review - and I am late in test -  will check this behavior using this branch as I was checking stable one for your issue, my bad
 
Please have a check with stable branch (based on 0.92.3) = master to tell me if you reproduce your issue

szlovak, 2016-05-04:
Putting some X min and Y min in RH didn't affect the gcode, after it was saved, RH restarted..., compared it gives identical result. So it limits probably output code to not go over these X min and Y min values.

Now talking about hardware. Master version is now in machine, properly configured in configuration.h (the default values actually are set for da vinci duo one fan). I set X min to 9mm , print was too much to right when I expected to be moved to the left, even more than before (so it was wrong direction), so I put negative value and it was even worse, X motor started to skip on belt's teeth. It's because the default position extruders are is the home position  and extruders can't be moved to negative values. This is forced by setting negative X min in that case. Y min behaves I think differently, maybe there is some more space, I set only 3mm definitely will not risk damaging belts again, it can happen . I think I'll will play now with extruders positions that is calculated in steps.

luc-github, 2016-05-04:
Well I have tested on both FW version on my Davinci 2.0 single fan and cannot reproduce your issue

Position is correct - not sure what can be the root cause of your problem

szlovak, 2016-05-04:
I found out that in printer extruder's position is adjusted in mm , not in steps, and then it's stored in eeprom.bin in steps. Anyway I moved right extruder a bit, it was more like 7.5mm , -35.65mm I lowered to -43.15mm, and Y from 0.15mm to 4mm, but in Y direction I can't center it completely, although I did it in the right direction, no matter how much I increase this value. I'm missing something about 2.5mm-3mm, but you can live with it. Second (left) extruder I corrected by subtracting initial, default values from these one I set for right extruder. Now it's much better, especially X is adjusted and don't have to do anything no more in any slicer. Of course it's for pure perfection, only once I was on the very edge of the bed :)

luc-github, 2016-05-04:
Left extruder is Extruder 1= T0 , right extruder is second extruder= T1
But it is good if you solved your issue

luc-github, 2016-05-07:
if no update I close issue

closed 2016-05-07 by luc-github

szlovak, 2016-05-07:
OK, please close