luc-github/Repetier-Firmware-4-Davinci #156: DaVinci AiO Feedback (no scanner)

luc-github, 2016-05-11:
the continuation of #80

The-Experimentalist, 2016-05-11:
ahhhh, that's better. 
Which commit should I use if I move to 1.6.7? 
What are the advantages of using 1.6.7 ?

luc-github, 2016-05-11:
well https://github.com/luc-github/Repetier-Firmware-0.92/tree/0.92.6-sync
is still experimental and not fully modified
Due core 1.6.7 bring some fixes - but IDE 1.6.8 has some issues that 1.5.8 do not has

Unless you want to test experimental branch no need to change
the 2 key points to move to 0.92.9 are :
1 - Multi language support
2 - better bed leveling - still working on this part to give same user experience than before

if you do not care these 2 points no need to change - I will update 0.92.3 based FW with new AiO pins in same time as 0.92.9 based

The-Experimentalist, 2016-05-11:
With point 2 I presume you mean auto levelling?

luc-github, 2016-05-11:
yes

luc-github, 2016-05-11:
https://www.repetier.com/documentation/repetier-firmware/z-probing/

The-Experimentalist, 2016-05-11:
I will have a look at that link. I have not updated since this commit, where I should have forked:

2015-09-27 13:45:24
Parent: 0f5907a998cf230e118cbd741a688c47c7970ab2 (Update README.md)
Child:  54e94776ce84007dc882a02adb12328e30a64fee

zeus10, 2016-05-16:
hello j'ai un autre pc peut être mettre arduino 1.6 et refaire les testes

zeus10, 2016-05-19:
hello plus personne ici

luc-github, 2016-05-19:
Salut - Desole pas vu la notification

luc-github, 2016-05-19:
Je suis deborde avec d'autres projets - je vais faire une branche de test avec les modifications de setup pour les 3 pins aujourd'hui sans fautes - desole du retard

Je n'ai pas reussi a installe le 1.5.8 pour verifier le pb et maintenant on est en 1.6.9 pour l'IDE (1.6.8 pour la DUE)

luc-github, 2016-05-19:
po nur le support du plateau, faire tourner le moteur en soit ce n'est pas difficile - mais le firmware n'est pas prevu pour ca, sans parler de l'interface - il faut ajouter le support d'un moteur qui n'est ni xyz ni Extrudeur a cause du control de la temperature

Le plus simple je pense est d'utiliser le code de BGM (je sais je suis faineant..., mais bon a quoi bon reinventer la roue quand cela fonctionne ?) https://github.com/bgm370/dav10/blob/master/dav10test.ino
pour faire tourner le plateau - comme en principe il faut 180 steps pour faire un tour a multiplier par la valeur de microsteps qui doit etre de 16

luc-github, 2016-05-19:
je suggere de faire une command M101 dans un premier temps avec le sens et la vitesse 
M101 S0 P30

pour lancer la commande : move(int forward, int steps, int del)
ou 0 est le sens des aiguilles d'une montre, steps = 200 X 16 =3200, pas besoin de parametre si c'est pour une rotation complete, del est la pause entre chaque step pour controler la vitesse, plus del est eleve plus c'est lent

zeus10, 2016-05-19:
oui mais si je trouve déjà la Command pour faire tourner ce plateau c'est déjà pas mal ..
sympa merci pour les conseils je te tiens au courant

luc-github, 2016-05-19:
je vais ajouter la commande pour faire tourner le plateau dans la branche test ce soir aussi

zeus10, 2016-05-19:
[image]
[image]
ca c'est chouette  j'ai mis deux images et elle ne sont pas la ???

zeus10, 2016-05-19:
REGARDE FAIS un print ecran caar après je retire

zeus10, 2016-05-19:
je sais que tu vas encore rigoler mais 
dans arduino moniteur serie  j'ai envoyé ca   move(int forward, int steps, int del) M22 S0 P30  il ne ce passe rien

luc-github, 2016-05-19:
je viens de me mettre a mon PC je met les commandes de desactivation du plateau / laser 1 / laser dans le setup et je fait la commande M101 dans la foulee - ca ne devrait pas etre long

luc-github, 2016-05-19:
enfin en theorie - le clonage de la branche a l'air de prendre des plombes car mon pc fait des mises a jour en meme temps....

zeus10, 2016-05-19:
loll ok alors je te laisse tranquille a tantot

luc-github, 2016-05-19:
la pin 16  est la pin de chauffage de l'extruder donc ca ne peut pas commander le plateau

zeus10, 2016-05-19:
driver a4988

pin 16 step qui vas sur la pin 1 mcu
pin 19 dir qui vas sur la pin 68 mcu
pin 2 enable qui vas sur la pin 14 mcu

zeus10, 2016-05-19:
regarde image que j'ai mis plus haut

luc-github, 2016-05-19:
ok 
#define LASER1_PIN			101 //Low = 0ff
#define LASER2_PIN			2	//Low = off
#define PLATE_STEP_PIN     22
#define PLATE_DIR_PIN      13
#define PLATE_ENABLE_PIN   26 //HIGH = Disabled

luc-github, 2016-05-19:
la branche de test est la : https://github.com/luc-github/Repetier-Firmware-0.92/tree/test-aio

luc-github, 2016-05-19:
l'initalisation est faite laser1/2 = off et plateau libre
je vais regarder la commande moteur

zeus10, 2016-05-19:
ok merci pour tous après je retire tous ce que j'ai mis ici qui na rien a voir avec ton sujet

zeus10, 2016-05-19:
je peut envoyé avec arduino 1.5.8
Firmware-0.92/tree/test-aio

luc-github, 2016-05-19:
tu peux essayer mais moi j'utilise 1.6.9 vu que je n'arrive pas a installer 1.5.8
je viens de corriger une erreur de compilation aussi

zeus10, 2016-05-19:
ok je vais essayé et je dis te quoi

luc-github, 2016-05-19:
Ok command M103 S<sens> P<speed>

M103 fait S=0 et P=50
je ne peux pas tester donc ....

zeus10, 2016-05-19:
j'ai envoyé nouveau FW et tout c'est bien passer magnifiquement  passer les lasers sont bien désactivé et le plateau est libre il tourne a la main donc n'est plus en état de maintien .

luc-github, 2016-05-19:
cool
plus qu'a tester la commande M103 :wink:

zeus10, 2016-05-19:
arduino mode serie M103 S0 P50 il ne se passe rien

luc-github, 2016-05-19:
c'est que la pin step n'est pas la bonne

luc-github, 2016-05-19:
que la pin dir ne soit pas bonne n'est pas grave ca tournera que dans un sens

zeus10, 2016-05-19:
mmmmm oui pour dir mais j'ai testé moi meme avec suiveur trace tu a vu sur la photos plus haut et agrandir pour voir detaille

zeus10, 2016-05-19:
oui mais enable doit être dans quel état

luc-github, 2016-05-19:
le plateau est libre durant le mouvement et fixe a la fin

luc-github, 2016-05-19:
je viens de rajouter le parametre R pour la pin step - tu peux faire plusieurs essais 
M103 P19 utilisera la pin 19 au lieu de la 22

zeus10, 2016-05-19:
M103 S0 P50   réponse   ok 0 
je vais comparer les tension sur les autre driver pour comparer et revérifier si c'est la pin step n'est pas la bonne et je te dirais quoi mais je crois pas pour la pin mais qui sait .....

luc-github, 2016-05-19:
malheuresement je ne peux pas plus aider desole

zeus10, 2016-05-19:
oui c'est bien vrai ce que tu dis et je te dis encore grand merci pour tout mais je vais trouver le probleme si tu est sur de la commande que tu a dis M103 S0 P50

luc-github, 2016-05-19:
M103 suffit en fait

zeus10, 2016-05-19:
oui vu tu la mis dans fw je suppose loll

luc-github, 2016-05-19:
M103 R<pin> te permettra de tester plusieurs pins sans recompiler

luc-github, 2016-05-19:
je viens de mettre le code a jour

zeus10, 2016-05-19:
OK je vais mettre new fw alors

zeus10, 2016-05-19:
VOILA FW remplacer et j'ai envoyé  sur port serie arduino M103 R
 et la j'ai eu un brui du plateau il c'est mis en état de maintien (donc bloqué) ne peut plus le faire tourné a la main logique ou pas

zeus10, 2016-05-19:
je te dis bonne nuit car ici déjà minuit demain mes idées seront plus claire merci pour ton aide a demain

luc-github, 2016-05-19:
oui c'est normal mais il faut mettre un numero de pin apres le R
 genre M102 R19 ou M102 R22

luc-github, 2016-05-20:
Une solution une peu bourrine mais qui permet de verifier que ce n'est pas le code le probleme:
Changer les pins du moteur X par le pins du plateau et lance un home de X  si les pins sont OK ca doit tourner 
changer ici dans userpins.h : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/userpins.h#L4-L8

changer en 
```
#define ORIG_X_STEP_PIN     22
#define ORIG_X_DIR_PIN      13
#define ORIG_X_MIN_PIN      11
#define ORIG_X_MAX_PIN      -1
#define ORIG_X_ENABLE_PIN   26
```

c'est juste un test de plus

zeus10, 2016-05-20:
[image]

zeus10, 2016-05-20:
bonjour regarde ceci impossible erreure mais je vais cherché voir si pas autre probleme

luc-github, 2016-05-20:
essaie ma suggestion de changer les pins du moteur X

zeus10, 2016-05-20:
oui voila je suis de retours evidenment que je vais essayé loll

zeus10, 2016-05-20:
j'ai retirer connecteur moteur x et j'ai mis celle du plateau et fait un home x et 
le ------plateau tourne

zeus10, 2016-05-20:
et en temps normale sans changer les connecteurs  ont a l'état de maintien donc le driver 4988 fonctionne sinon pas d'état de maintien

luc-github, 2016-05-20:
changer les connecteurs = utiliser les pins de x = normal que ca marche

ce que tu veux c est verifier les pins pour le plateau, pas verifier les pins du moteur X

ou j ai manque un truc ?

zeus10, 2016-05-20:
MOI j'ai cru que tu voulais contrôler le moteur du plateau

[190 more replies]

closed 2016-06-20 by luc-github