luc-github/Repetier-Firmware-4-Davinci #157: Filament Sensor do not pause printer if printing from host

luc-github, 2016-05-15:
hi luc
I restarted my DaVinci and I noticed that if there is no filament during printing, the printer asks to load a new filament but bug. The bed down and then up but does not charge the filament and printing failed.

luc-github, 2016-05-15:
RequestPause: No Filament command is well handled by Host 1.6.1 from test
Will check complet behavior with printer

Doudou14, 2016-05-15:
The pause works but the printer down and up the bed and move the extruder without loading the filament. Then it crashes.
Orders on repetier do not react Printer

luc-github, 2016-05-17:
I reproduced issue with IDE 1.6.9 / Due 1.6.8
message show - saying no filament press ok - menu show then skip and bed down - and additionnnaly if z is not homed it can go down and hit

Seems menu is skipped for some reason - will check

luc-github, 2016-05-17:
if I do a 
```G1 E10```
everything work as espected for menu - but bed do not revert to correct position

luc-github, 2016-05-17:
seems is lost even there is a home done

luc-github, 2016-05-17:
Seems Resuming from pause do not set the position when pause occur unlike SD print.
Simple solution is just to not change the Z position but put X Y out of bed as X Y positions are send in every command they will be restored naturaly

Doudou14, 2016-05-18:
So I have to remove the Z command in the code. I must look

luc-github, 2016-05-18:
I did already, check last commit

luc-github, 2016-05-18:
here the commit : https://github.com/luc-github/Repetier-Firmware-0.92/commit/a442ed9e3bdeea73cde793324a4ff8e60f20cfbf

luc-github, 2016-05-18:
You are welcome - sorry got delay again with the tests and auto bed leveling UI finalisation due to others projects keeping me busy ...

closed 2016-05-18 by luc-github