luc-github/Repetier-Firmware-4-Davinci #158: Add start and end sequence g-code to documentation

joelmeijering, 2016-05-19:
First of all, thanks for all the work, the firmware has been working great and I have been using it for about a year now.

However, I had to spent some time to get the start and end g-code right. Since it took quite a lot of time, I think It could be added to the documentation. Unless the issues I was having were specific to me.

The problems I was experiencing:
- z axis was too high (I had to set a negative z-offset in slic3r of -3.4mm)
- printhead was hitting the print bed screw at the start of the print
- printer would first heat the nozzle, after this was completed it would start heating the bed. After this, it would start homing the z-axis, instead of doing all three at once.
- printer does not perform the printhead cleanup operation after heating the nozzle

I wrote some g-code that fixes all this. (when slicing I put the g-code int the slic3r g-code field). For me, it prevents the printhead from hitting screws, speeds up heating and homing by doing it all at once and it performs a cleanup by moving the nozzle over the cleaning brush (for lack of a better word).

Start g-code:
```
  ; Heat up
M117 Heating...
M140 S120 ; Set bed temp, no wait.
M104 S230 ; Set nozzle temp, no wait.

; Init position
M117 Homing...
G21 ; set mm units
G28 ; home all axis
G90 ; set absolute coordinates
G1 Z20 F300 ; move platform down 20mm

M116 ; Wait until target temperatures for extruder and heated bed are reached

; Clean nozzle
M117 Cleaning...
G1 Y-10 F1500 ; G1 Y5 F1500
G1 X-13 F1500 ; G1 X22 F1500
G1 X-33 F1500 ; G1 X2 F1500
G1 X-13 F1500 ; G1 X22 F1500
G1 X-33 F1500 ; G1 X2 F1500
G1 X-13 F1500 ; G1 X22 F1500
G1 X-35 F1500 ; G1 X0 F1500
G1 Y5 F1500 ; G1 Y20 F1500
G1 Y-10 F1500 ; G1 Y5 F1500
G1 Y5 F1500 ; G1 Y20 F1500
G1 Y-10 F1500 ; G1 Y5 F1500
G1 Y5 F1500 ; G1 Y20 F1500
G1 Y100 X100 ; G1 Y100 X100

; Safely position nozzle to start printing
M117 Positioning...
G1 X100 Y100 F3000 ;move to front right corner
G1 Z[first_layer_height] F200 ;move platform close to nozzle
G92 E0; reset extrusion distance
M117 Printing...

```

End g-code:
`G1 Z190 F300 ; lower platform to bottom`

PS Maybe I just overlooked some builtin features of slic3r or this firmware distribution, and I'm doing things in a more complicated way than I have to? Nevertheless, maybe this can save people some time.

luc-github, 2016-05-19:
Hi @joelmeijering  thanks a lot for sharing,
well start / end GCODE actually depend of your slicer so this one will work with Slic3r but
```
G1 Z[first_layer_height] F200
``` will not do well with S3D neither Cura and even crash the printer, for craftware, kisslicer

about clean nozzle you have M100 command that do the trick

I would like also to suggest you to:
1 -  add some extrusion before clean noozle, when heating the filament inside nozzle will melt and fall just by gravity, not a lot but a little, the result is that the nozzle chamber is empty when heating is done, and so first extrusion just fill the empty nozzle before being extruded

2 -  to switch off heating your bed/extruder for safety purpose :smile: 

I shared some start/end GCODE for S3d here : https://forum.voltivo.com/showthread.php?tid=8522&pid=5032#pid5032

and for cura here : https://forum.voltivo.com/showthread.php?tid=8200&pid=2056#pid2056

I did not for slic3r because was not happy with the result so did not used long time

all these to say - not easy to do a doc that cover all :wink:

luc-github, 2016-05-19:
to finish to answer your suggestion there are 2 ways to do documentation
1 - create an issue in FAQ format for specific question like :
how to low down the bed ?- what need to do if I hit bed when Z is homed ? 
like here : https://github.com/luc-github/Repetier-Firmware-0.92/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3AFAQ+

2 - create a page in Wiki 

you are welcome to contribute to any

joelmeijering, 2016-05-22:
Thanks for the clarification. And also thanks for the advice :)

I think the following needs to be added to the README.md. Maybe you can add a section "Notes" after the section "Installation". But only if you consider it useful to people who want to try out this firmware. I think its useful, because these were issues I ran into and it was not always obvious how to fix them. A few of these notes can save people a lot of time. I suggest you add the following:

## Notes:

- With the repetier firmware the printer no longer automatically executes the nozzle cleaning procedure. Add the g-code M100 to the custom start g-code section in your slicer to make the printer clean the nozzle".
- The printer will likely need a z-offset adjustment in order to get the nozzle to start printing at the right height. This can be configured in most slicers under "z-offset" 
- Depending on the starting point of the print, the nozzle might hit the print bed screw while moving to the starting point. This can be prevented by adding custom start g-code in your slicer, like so:
``
; Safely position nozzle to start printing
G28 ; home all axis
G90 ; set absolute coordinates
G1 Z20 F300 ; move platform down 20mm
G1 X100 Y100 F3000 ;move to front right corner
``

luc-github, 2016-05-22:
normally you should not have any zOffset if you do proper bed leveling, and I think it is same root cause of your point 3

z offset should be 0, what is your printer model?

joelmeijering, 2016-05-22:
That explains the problem, I have never run the bed leveling feature.

The bed leveling feature has never worked for me (with the repetier), it was probably a metal conduction issue or I was using firmware too old (march 2015). Also the stock probe is kind of useless because it does not measure the glass, it measures the metal clips.

I have a DaVinci 1.0.

I recently installed a custom z-probe (just a simple switch), wired to the same pin that the stock probe is wired to. Today I am testing this new probe and it seems to be working so far.

The benefit of a switch type z-probe instead of the stock metal conduction based one, is that I could measure the whole bed and use the repetier distortion map. But this only works well in 0.92.9 (according to repetier docs). So I'm trying to get 0.92.9 so I can use the distortion mapping. But for now Im gonna test wether or not manual levelling assisted by having the printer measuring 3 points using the new probe works.

luc-github, 2016-05-22:
auto level is buggy that is true, 0.92.9 is supposed to have correct one in gcode
in menu I am still working pn rewrite it and I am late, but next week I will speed up

Manual bed level do not use probe but screws and paper sheet, it is the best when it is done properly

luc-github, 2016-06-05:
I have implemented Zmin calculation using probe - will finish autolevel this week I think as they share same process

luc-github, 2016-06-20:
If no update I close ticket

closed 2016-06-20 by luc-github