luc-github/Repetier-Firmware-4-Davinci #159: DaVinci AiO Feedback (no scanner)

luc-github, 2016-05-23:
because previous thread is too long

greymfm, 2016-10-23:
OK, here's some feedback on 1.0 AiO (built on 2014/09/19) :-) I'm trying non-stock firmware for the first time. At first, everything looked fine, it prints! But after some printing time (1-10 minutes), machine restarts. Happens on both USB and SD card printing. I opened the machine, measured 3.3v power supply, everthing stable. 

What would you suggest to debug this? :-) I'm using master branch ( commit on Oct 8, 2016 ).

luc-github, 2016-10-23:
code did not really changed for a while -  so I would say issue is a short or a problem with the model.
is reset random or after same period of time ?
does reset happen if you just heat extruder and bed?
does reset happen with several objects or always same model ?

greymfm, 2016-10-23:
Reset seem to be fully random:
- heating extruder and bed (240/90) for 15 minutes: no reset
- printing different models over and over: random reset (currently testing again and yet no reset since 40 minutes)
Also, there was never any reset with stock firmware.
Kind of error/bug that we do not want to debug right... ;-) 

What could trigger a software-reset, the watchdog function?

luc-github, 2016-10-23:
you can disable the watchdog to see if it is the one which reset printer. 
in configuration.h line 1577
https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Configuration.h#L1577
set `#define FEATURE_WATCHDOG 0` and reflash firmware

greymfm, 2016-10-23:
Thanks, I'll try - the reset seem to have something to do with the USB plugging. If I restart the machine without USB, not reset. If I start using USB and remove USB for SD card printing, random reset...

luc-github, 2016-10-23:
ho I see
so it means the USBCore.cpp patch is not set - without my patch if serial communication is set and you unplug USB it will trigger the watchdog and if you disable the watchdog and this happen you my damage the printer
Any loose of connection with froze printer unplug cable/ PC go to standby / etc

greymfm, 2016-10-23:
OK, it looks like that was the trick. I'm used (the Davinci way) to transfer models from my laptop to the printer's SD card, initiate printing on my laptop, then unplug USB cable so I can move away laptop for the printing time. Thank you very much for clarifying the USB issue to me :-)

luc-github, 2016-10-23:
the only way to speed up is to use wifi SDCard or to use an extender to easily plug/unplug SDCard but you then need to unmount/manually the SD card as extender block the SD plug detection as always connected, no issue if you use Wifi SD card

greymfm, 2016-10-23:
Hmm, it's MicroSD in the latest machines (not sure if micro Wifi exists) - BTW: does the repetier-firmware protocol require to confirm each single G-code sent over the line or only certain chunks of G-code? (confirming only chunks could speed up the transfer)

luc-github, 2016-10-23:
you need extender micro SD on one side and normal SD on other side to use Wifi SD - several people did it
no, repetier read each GCODE one by one, check integrity of each command, then convert them in binary to save on SD - it is slow like hell

greymfm, 2016-10-24:
I would like to send you a small donation to honor your efforts on this work (I think it was a lot of work) - Do you have a PayPal account? Thanks, Alexander

luc-github, 2016-10-24:
Hi thanks for your support - I just add back to end of read me the donation button which was removed when I merged 0.92.10 based version

luc-github, 2018-06-02:
Close as now it is outdated

closed 2018-06-02 by luc-github