luc-github/Repetier-Firmware-4-Davinci #160: M355 gcode and lights connection

10009921, 2016-06-03:
After long searching I found out about the M355 gcode command to switch the case lights on and off.

What I cannot find is where and how the case light(s) connection must be made.

Or am I misunderstanding the concept and the the "case lights" is referring to the LCD's lights instead of external lighting for a cabinet in which the printer might me placed???

luc-github, 2016-06-03:
this switch on/off the led strip on Davinci 
depending on model you have pin is different : 
```
#if MODEL==0
#define LIGHT_PIN    85
#else
#define LIGHT_PIN    17
#endif
```
what davinci do you have ?
what firmware version do you use ?

luc-github, 2016-06-07:
If no update I close issue

closed 2016-06-07 by luc-github