luc-github/Repetier-Firmware-4-Davinci #161: finer graduation for load filament feedrate?

Hecot, 2016-06-09:
Dear Luc!

I've changed to a E3D V6 bowden extruder on my Davinci and now do a lot of trials for new temperature and volumetric setting. For these trials I found the "load filament" function very usefull as it works without any external computer - but the downside is that I could only adjust the feedrate in full digits in the settings menu! 
Is there a way to change these settings to one or better two decimal places in the firmware?
Thanks

luc-github, 2016-06-09:
[image]

use EEPROM editor of repetier and you have access to it

Hecot, 2016-06-09:
Dear Luc,

Thanks for the hint, but you may remember that I have problems with adjusting the EEPROM settings [https://github.com/luc-github/Repetier-Firmware-0.92/issues/146]. Therefore I would love to use the buttons on the printer to adjust the feedrate ... all I would need is a finer pitch (one or two decimal places) in the control menu.

luc-github, 2016-06-09:
yes using arduino serial is possible , you can connect with arduino serial to printer ?

Hecot, 2016-06-09:
Yes - I can connect to the arduino board with USB cable for programming the firmware. It would be great if you could show me what to change in the code to get finer resolution of the feedrate in the settings menu!

luc-github, 2016-06-09:
everything is in ui.cpp not difficult whenyou do more long to explain
so I suggest you use arduino serial or any serial terminal and use GCODE
```
M206 T<type> X<value> P<position in EEPROM> Set EEPROM value

#define EPR_TYPE_BYTE		0
#define EPR_TYPE_INT		1
#define EPR_TYPE_LONG		2
#define EPR_TYPE_FLOAT		3

#define EPR_LOAD_FEED_RATE 1179
#define EPR_UNLOAD_FEED_RATE 1183
#define EPR_UNLOAD_LOAD_DISTANCE 1187
```
for load feedrate 2.2
M206 T3 X2.2 P1179
for unload feedrate 4.2
M206 T3 X4.2 P1183
for load/unload distance 80
M206 T3 X80.0 P1187

luc-github, 2016-06-09:
easy ?

luc-github, 2016-06-09:
you also can try M205 command to show all EEPROM - it may unlock your grey item ?

Hecot, 2016-06-10:
Yes - getting a finer resolution of the feedrate was easy ... 

[image]

but still strugling to get the digit after the point displayed on the LCD. So far I didn't understand what cuts off the digits from beeing displayed ... any idea!?

luc-github, 2016-06-10:
because UI and action are separated
https://github.com/luc-github/Repetier-Firmware-0.92/blob/0.92.6-sync/src/ArduinoDUE/Repetier/ui.cpp#L1737-L1739

```
            else if(c2 == 'l') addFloat(EEPROM::loading_feed_rate, 5, 0);
            else if(c2 == 'u') addFloat(EEPROM::unloading_feed_rate, 5, 0);
            else if(c2 == 'd') addFloat(EEPROM::unloading_loading_distance, 3, 0);
```

Hecot, 2016-06-10:
Hmmm ... I've tried 1 and 2 digits (see below) but the LCD is still showing no digits!? 

[image]

I don't know how to debug the addfloat function - I assume it is working ... !? I will check if there is another float number with digits shown in the Display to prove the function is working.

luc-github, 2016-06-10:
did you tried this ?
```else if(c2 == 'l') addFloat(EEPROM::loading_feed_rate, 3, 1);```

Hecot, 2016-06-11:
YES - you are right I need to get some clearance for the additional dot and digit ... now it's working great!

[image]

Thanks for your help - you are GREAT!!!

I think about making an extra menu for adjusting the feedrate and temperature in one menu to ease the work to profile the filament ...

closed 2016-06-11 by Hecot