luc-github/Repetier-Firmware-4-Davinci #162: da Vinci AIO black bars won't go away.

allenlorenz, 2016-06-11:
On my Da Vinci AIO, I have followed the instruction given in, the readme.
1 ) shorted j37.
2 ) downloaded and updated the files...
       you can add a Note for Linux are located in your home directory in directory .arduino15
3) used 1.6.9 ide with 1.6.8 files. updated files which compiled with a warning.

In file included from sketch/Configuration.h:182:0,
                 from sketch/Repetier.h:200,
                 from sketch/Commands.cpp:22:
sketch/pins.h:1210:140: warning: narrowing conversion of '128' from 'int' to 'const int8_t {aka const signed char}' inside { } [-Wnarrowing]
                         HEATER_0_PIN, HEATER_1_PIN, HEATER_2_PIN,FAN_PIN, E0_PINS E1_PINS E2_PINS TEMP_0_PIN, TEMP_1_PIN, TEMP_2_PIN, SDSS }
                                                                                                                                            ^
sketch/Commands.cpp:24:41: note: in expansion of macro 'SENSITIVE_PINS'
 const int8_t sensitive_pins[] PROGMEM = SENSITIVE_PINS; // Sensitive pin list for M42
                                         ^

Sketch uses 204,012 bytes (38%) of program storage space. Maximum is 524,288 bytes.

Next the files loaded  ( output is )
Atmel SMART device 0x285e0a60 found
Erase flash
done in 0.013 seconds

Write 206836 bytes to flash (808 pages)
[==============================] 100% (808/808 pages)
done in 6.395 seconds

Verify 206836 bytes of flash
[==============================] 100% (808/808 pages)
Verify successful
done in 4.871 seconds
Set boot flash true
CPU reset.

**However the CPU does not reset, and if I power cycle I still see the black bars.    Any ideas ?**

luc-github, 2016-06-11:
Looking at the report you are using 0.92.6-sync branch, right ?

this error is typical of reset is not complet or variant.cpp is not replacing current one.

redo a J37 jumper long time (5s) and doing twice :
1 - power off
2 - jump j37 and keep it jump
3 - power on wait 5s
redo
1 - power off
2 - jump j37 and keep it jump
3 - power on wait 5s

Check on your home the hidden directory: .arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x

and verify the file variant.cpp has a line ending with 
```
// PIN 125
```

allenlorenz, 2016-06-11:
I tried under windows, and got it to work ( after making several dumb
mistakes ).    After I learn what I am doing with Repetier ( host,
firmware, and pritning).  will go back and investigate this.

Quck note,  the default value for
DAVINCI  is 4 for the AIO,
however the
MODEL 0       ( which I changed to 1, ( since J37)     is this just to make
sure people are paying attention ?

Nice Job...

think you need to put the file stuff up front in the readme in bold...

luc-github, 2016-06-11:
Davinci 1 and 2 can be MODEL 0 or 1, but AiO is only model 1 so actually if DAVINCI is 4 then I force it to be MODEL 1 later just in case, so it is OK 

0.92.6-sync branch was recently for testing mostly new AiO feature (scanner function) so default is for AiO

FYI scanner software porting is still in early stage- Firmware is ready but not host - I do not have AIO so it rely on people having it and working on this porting

allenlorenz, 2016-06-11:
don't have a lot of "free" time but will keep you updated on what I find!

luc-github, 2016-06-11:
thanks - happy printing :smile:

closed 2016-06-11 by luc-github

allenlorenz, 2016-06-12:
Well did first print looks good.  I am also impressed the code look clean.
So I had to hit donate button for you !

luc-github, 2016-06-12:
Thanks a lot :+1:

SASpirit, 2016-10-05:
Hi Guys,

I have run into the EXACT same issue today and for the life of me cannot figure out why the printer will not update.  Every thing seems to run through just fine but the printer never makes any sounds and the black bars remain.  It is the AIO model and I am using 0.92.10

Any idea what could be going on?

SASpirit, 2016-10-05:
Actually just figured it out.  You have to power down and then on again for the screen to change and reflect that the software has loaded.  Then proceed to load the fail safe.

Tks

luc-github, 2016-10-05:
yes this is point 9 in readme 😀