luc-github/Repetier-Firmware-4-Davinci #164: Da Vinci Pro Support

mitchhh22, 2016-06-23:
Is there a working version that works with the 1.0 Pro Model?

luc-github, 2016-06-23:
Hi got already the question : https://github.com/luc-github/Repetier-Firmware-0.92/issues/142
Not from me - but what is the chip on the pro board ?

geoffsim, 2016-06-23:
I am ready to work on getting repetier on my pro.
Chip is ATMEL ATSAM4E8E.

luc-github, 2016-06-23:
ATSAM4E8E is same as JR - I bough one with the donation but I am late to this project,  I am so busy with others projects that the printer is waiting on table....

What you need to understand is this porting is not trivial  : 
* First you need to have arduino environnement supporting ATSAM4E8E which is first step before porting. Bootloader and register address are completely different than current Davinci chipset so we cannot rely on it. It is not only variant.cpp/.h but the complete package. I am supposed to work on this with the owner of this github : https://github.com/aethaniel/ExperimentalCore-sam which is exactly what we need - but the guys is also busy, I and I do not have the skills to do it alone - I am still learning and even do not know how to start, so I wait for him as he is an expert.

* Then we can start the "porting" which cover the pinout and the thermal sensor tables creation

The second part is not difficult if first part is reliable.

I hope it clarify -

geoffsim, 2016-06-23:
I am a software engineer by profession, so I hope I can help.
Please let me know what I can do.

luc-github, 2016-06-23:
Of course any help is welcome ^_^
here the data sheet :
http://www.atmel.com/Images/Atmel-11157-32-bit-Cortex-M4-Microcontroller-SAM4E16-SAM4E8_Datasheet.pdf
here the header:
 for https://github.com/aethaniel/ExperimentalCore-sam/blob/master/module/tools/CMSIS_Devices/ATMEL/sam4e/include/sam4e8e.h
for your reference.

May be you can use another approch than me at first: use Atmet studio to program the ATSAM4E8E 
Developement board is : http://www.atmel.com/tools/ATSAM4E-XPRO.aspx so you may use code sample for this board to succeed.

**But be aware if you reset your printer board you have no way to flash stock FW back - so during the period you won't be able to use your printer anymore**

geoffsim, 2016-06-23:
I was hoping to find JTAG support on the motherboard, as that would make it easier.
It might also allow me to extract the existing firmware.

I remember you were talking to someone else some time ago regarding a simple program to write to the printer for a simple test. I can't find that thread anywhere. Can you point me to it?

luc-github, 2016-06-23:
this was before I look into the bootloader issue : http://www.soliforum.com/topic/13769/davinci-jr-repetier-port/

geoffsim, 2016-06-23:
I have created some high res photos of the motherboard.
I have also identified the erase switch/jumper as SW2. I have not tried it!

luc-github, 2016-06-23:
nice, you can share here

geoffsim, 2016-06-24:
Here are the images of the top and bottom layers of the board. The bottom is mirrored to make it easier to trace tracks. For example, it is quite easy to see that pin 87 (ERASE/PB12) of the Amtel chip goes to the solder pad marked SW2.

[image]
[image]

luc-github, 2016-06-24:
Wow nice work!

geoffsim, 2016-06-24:
Any idea what U8 and U15 are for?

[image]
This is a 32mb serial flash rom.

[image]
This is a 32-bit ARM Cortex-M0 micro-controller.

luc-github, 2016-06-24:
I just can guess -  recovery system ? independant flash / mcu to flash ATSAM4E8E in case update failed / or someone erase the main flash 

need to check what are the connection between the 2 mcu

geoffsim, 2016-06-29:
I am not taking this any further at the moment as I am now quite happy with the current Simplify3D support for this printer, and I can't see any immediate benefit seeing as it already allows 3rd part filaments to be used.
I hope the images I have posted, and the identification of the "erase" jumper location, goes some way towards deploying Repetier to the DaVinci 1.0 Pro. I will keep watching...

luc-github, 2016-06-29:
If you do not need repetier and happy with S3D it is good - thanks for the pictures :smile: 
Happy printing - S3D rocks

luc-github, 2016-07-09:
if no update I close topic

closed 2016-07-09 by luc-github