luc-github/Repetier-Firmware-4-Davinci #165: Second Fan Pin on 1.0A

DecisiveMocha, 2016-06-26:
I added the transistor and load resistor on the main board to activate the second fan header on my 1.0A that's unused from the factory. I connected the extruder fan to the second header to test it, but it's not working.

When I short collector to emitter on the transistor, the fan turns on, so it seems the electronics are sound but I'm not getting the current to the base of the transistor to switch it. I show continuity from the base to the current limiting resistor, but the other end of the current limiting resistor seems to have continuity to physical pin 129, not physical pin 137 as in pinout.xlsx and the pin configuration in the firmware. I'm not sure how to switch the firmware to use pin 129 - it's digital pin PB11 and seems to be normally used by the Arduino for USB.

I don't think this is the issue, but here's the firmware change I made in Configuration.h to make this work:
```
/** Should support for fan control be compiled in. If you enable this make sure
the FAN pin is not the same as for your second extruder. RAMPS e.g. has FAN_PIN in 9 which
is also used for the heater if you have 2 extruders connected. */
#if REPURPOSE_FAN_TO_COOL_EXTRUSIONS==1
#define FEATURE_FAN_CONTROL 1
 #define FAN_PIN ORIG_FAN_PIN
  #if DAVINCI==2 || DAVINCI==3
      #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
      #define EXT1_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
    #else //DaVinci 1.0
        #define EXT0_EXTRUDER_COOLER_PIN -1 //Warning need to add a permanent fan with power supply to cool extruder
    #endif
#else
  #define FAN_PIN -1
  #define FEATURE_FAN_CONTROL 0
#endif
```

to this:

```
/** Should support for fan control be compiled in. If you enable this make sure
the FAN pin is not the same as for your second extruder. RAMPS e.g. has FAN_PIN in 9 which
is also used for the heater if you have 2 extruders connected. */
#define FEATURE_FAN_CONTROL 1
#define FAN_PIN ORIG_FAN2_PIN
```

DecisiveMocha, 2016-06-26:
I think I figured out what's going on. As you can see in the picture at http://www.soliforum.com/post/124542/#p124542 , I added a transistor and resistor to activate another pin on the same "Fan1" header that the 1.0A already uses. ORIG_FAN2_PIN AKA pin 137 seems to correspond to a pin on the "Fan2" header that I still haven't populated.

Is it possible to use pin 125 from the firmware?

luc-github, 2016-06-26:
Hi, 
Can you have a look here : https://github.com/luc-github/Repetier-Firmware-0.92/issues/152

seem all hardware is available, can you confirm you do not have the reset issue when enable the repurpose fan?

As the ways to get repurpose fan on 1.a and AiO controling 2 fans and not only one as currently, is not yet clear for me, it is not yet implemented in FW

so every feedback is important
please review 
https://github.com/luc-github/Repetier-Firmware-0.92/issues/152 and comment here thanks

luc-github, 2016-06-26:
About you question for PB11 pin has 2 channels :l A for usb and B for A19
entry for A19 is not in variant.cpp need to add a line like 

```{ PIOB, PIO_PB11B_A19, ID_PIOB, PIO_PERIPH_B, PIO_DEFAULT, PIN_ATTR_DIGITAL,NO_ADC, NO_ADC, NOT_ON_PWM,  NOT_ON_TIMER }, // ID  130```

if you use 0.92.10 version you need to add definition of new pin in fastio.h also

DecisiveMocha, 2016-06-27:
I guess the board must be using PB11 channel A because switching ORIG_FAN2_PIN to 85 as in #152 did the trick. I can now control a fan through that extra wire.

It's notable that my board differs from the one pictured in #152 - I have the same blue wire but I had to solder on my own transistor and resistor.

Using that fan with PWM (and not PDM) did not cause any resets for me. However, I'm currently testing this setup by connecting the extruder fan to the second fan output. I'll be getting the actual cooling fans on Wednesday - I'll see what happens with PDM vs PWM and whether I hit the reset issue then.

Thanks!

luc-github, 2016-06-27:
Thanks for sharing - so it seems 1.0A has several level of populated components - good to know

Happy to see you do not have any issues, waiting for your final confirmation

DecisiveMocha, 2016-07-12:
I just realized that what I should really try is switching the pin configuration around so the cooling and extruder fans are swapped. That way I can run through these same tests while varying the speed of the extruder fan, which is switched directly by its motherboard NPN transistor. I'll have to do that tomorrow, though.

DecisiveMocha, 2016-07-13:
I swapped the pins around in userpins.h and ran the extruder fan through the settings, but I still couldn't get a reboot to occur. Can you think of anything else I should try out?

luc-github, 2016-07-13:
I think it is ok - thanks a lot :smile: it may be an hardware issue as your board is different than the one which has issue 
Can you share your configuration.h and userpins.h ?
Just to see if we missed nothing and create a FAQ to do your setup
Thank you again for your tests and feedback

luc-github, 2016-07-22:
@DecisiveMocha any update ?

luc-github, 2016-08-08:
Ok If no update I close topic

closed 2016-08-08 by luc-github

DecisiveMocha, 2016-08-08:
Sorry, those files were on a laptop where I don't normally read my mail, so I kept forgetting to fire it up and post them.

Here they are: [ConfigurationAndUserPins.zip](https://github.com/luc-github/Repetier-Firmware-0.92/files/407254/ConfigurationAndUserPins.zip)

luc-github, 2016-08-10:
You send me communication.h not configuration.h ^_^