luc-github/Repetier-Firmware-4-Davinci #167: M302 description incorrect

thinkyhead, 2016-07-06:
The source code for `M302` states:
```cpp
case 302: // M302 S<0 or 1> - allow cold extrusion. Without S parameter it will allow. S1 will disallow.
    Printer::setColdExtrusionAllowed(!com->hasS() || (com->hasS() && com->S != 0));
    break;
```
- "Without S parameter, it will allow." **Correct:** `!com->hasS()`
- "S1 will disallow." **Wrong:** `com->hasS() && com->S != 0`

According to the code, `S1` will actually _allow_ cold extrusion, not disallow it.

I'd like to see this command applied consistently, if possible. In Marlin 1.0.x the behavior is:
- `M302` by itself allows cold extrusion (it sets the min extrusion temp to 0)
- `M302 S0` allows cold extrusion (by setting the min extrusion temp to 0)
- `M302 S180` prevents cold extrusion below 180 (celsius only in 1.0.x)

I'm proposing to extend Marlin's `M302` (https://github.com/MarlinFirmware/Marlin/pull/4222) with the following changes:
- `M302` by itself displays the current cold extrusion status (RepRapFirmware).
- `M302 P1` allows cold extrusion (RepRapFirmware).
- `M302 P0` disallows cold extrusion below the current min extrusion temp (RepRapFirmware).
- `M302 S180 P1` sets the min extrusion temp to 180, but allows cold extrusion.

luc-github, 2016-07-06:
good catch 
actually come from repetier : https://github.com/repetier/Repetier-Firmware/blob/development/src/ArduinoDUE/Repetier/Commands.cpp#L2057-L2058

you can open ticket to them or will do - let me know

About improvement of command I also suggest to propose to repetier original FW for consistency

thinkyhead, 2016-07-06:
Oops, I did mean to post it on the main page, but I was poking around here.

https://github.com/repetier/Repetier-Firmware/issues/548

closed 2016-07-06 by thinkyhead