luc-github/Repetier-Firmware-4-Davinci #168: DaVinci 1.0 Print shift

gamecompiler, 2016-07-07:
Hi, since id compiled the current branch with its Latest commit e809348 on 7 Jun, the print is shifted sometimes. Sounds crazy, but id attached a picture what happen sometimes.

The gcode is here:
[composition_export.zip](https://github.com/luc-github/Repetier-Firmware-0.92/files/352629/composition_export.zip)

[image]

Im not sure how thas can be happend. Maybe its not a software problem.
This error only occurs when im away(Thats no helpfull info i know).

You may have tips how to solve this problem ?
Sorry for my english.

gamecompiler, 2016-07-08:
I checked the printer for emi electromagnetic interference. Everything is fine there.
The endstops also work perfect. But id cleaned them with a toothbrush now.
It seems that it is a mechanical problem with the cables. 
It can happen that the X Axis- cable will be "pulled out" sometimes or damaged because of heavily movement while going back and forward.

**The fix was adding a little bit silicon on the connector and the first 1cm of cable.**

Other people also may want to make twisted pair cables for each coil for every stepper. Thats may not needed, but its a improvement for that people who may have problems with EMI.
(My dont have twisted pair cables, and i dont have problems with EMI)

luc-github, 2016-07-08:
yes connectors are weak on Davinci, this also generate decouble errors with sensors connectors loose
many people remove connector and sold wires directly

gamecompiler, 2016-07-08:
Hmm interesting, id tested the wire problem with an other print with any problems.

Id printed the old gcode of my design again now.
The shift problem occurs at the same layer (129/130)as before on this gcode and also when i generate a new gcode on the same object. 

Id saw that my design is not manifold. I will fix that first and try to reprint it.
But i think that shouldnt be the problem because while watching the gcode in Repetier, the print should be correct...

How can i identify where the gcode for layer 129/130 is ? 
There is a problem that let the X-Axis move X value + (1/2) * object size.

Now i dont think that it is a hardware problem because this error seems to be reproducible on software side.  Im using skeinforge for slicing.

Could be interresting if this error occurs also when using an other slicer like Mattercontrol on the same object. There is something strange wrong.

Id attached my design stl just as refference. Id rotate it X -> 270 and Z->90.

[HaloGunV2.zip](https://github.com/luc-github/Repetier-Firmware-0.92/files/354292/HaloGunV2.zip)

Here are also my skeinforge settings, i used the 0.25 profile WITHOUT support.

[skeinforge.zip](https://github.com/luc-github/Repetier-Firmware-0.92/files/354295/skeinforge.zip)

Please try my settings and object (Ink. rotating) on your printer. I would like to now if you have the same problem then.

gamecompiler, 2016-07-08:
Now im sure that it is a software problem.
Id printed my design sliced with CuraEngine. It printed without any problems.
Then id printed the design again sliced with Skeeinforge. Same problem as before.
There is happening something strange.

luc-github, 2016-07-08:
all slicers are not same - I know cura is good slicer but I never used skeinforge, it may be more sensitive if your object has not manifold area

gamecompiler, 2016-07-08:
Could be, i will try this gcode tommorow on a friends printer.
He has the same nozzle width and bowden extruder.
When the error do not occur on his printer, we will now it is may a DaVinci Repetier firmware related bug. This will be awesome interesting Oo.

The good thing is that at this time we have a reproduceable bug.
Nothing than just words from people using forums.

If the error do not occure, it "cant" be skeeinforges fault then. I guess there is a g command with some serious bugs inside this firmware, frustrating many people Oo.
The hard part will be to figure out what happend and why did it happen.
The most people say that something is a hardware fail, but the most time its not

(Id heard that really often in my dads Company he works, always it was first "his fault" and it was never, as it turned out in the end.)

I like bug hunting ^^

luc-github, 2016-07-08:
has said earlier can have several root cause for a shift.
I always check if my model is manifold and my slicer (S3D) fix automatically such issue when detected.

On my side, I never got it on my Davinci but got some on another printer with marlin FW - issue was EMI related, using armored wire for end stop solved the issue.

gamecompiler, 2016-07-11:
Got it, it is a crazy bug which also appears on other printers. 
Seems to be Skeeinforge only....
So this problem "isnt a problem" (related to this firmware).

Youd say that your are using S3D. Do you use a DaVinci 1.0 ? 
If yes, could you share your settings please ?

luc-github, 2016-07-11:
>So this problem "isnt a problem" (related to this firmware).

Yes I know :wink: 

My printer is Duo 2.0 with one fan - I use S3D default settings with some fancy music I think

S3D is compatible Stock FW and repetier FW, with some limitation with stock FW (no Jog and limited usage of temperature if I remember well)

luc-github, 2016-07-13:
if no issue I close topic

closed 2016-07-13 by luc-github