luc-github/Repetier-Firmware-4-Davinci #169: Pt100 and analog pinouts on a Davinci AIO

Lennie1972, 2016-07-07:
I'm trying to install a PT100 sensor ( w/ amplifier) to my Davinci AIO. My Problem is that I can't figure out which one is the analog pin on my board. Could you by chance help me with this. Any other help to get this up and running would be greatly appreciated

luc-github, 2016-07-08:
did you check the pinout.xls ?
variant.cpp list all analog pin also

Lennie1972, 2016-07-16:
I'm sorry it's taken me so long to respond back. I haven't been able to match up the pin number with the one in the back all file. I'm not sure what I'm doing wrong. Thank you for your patience and time.

Lennie Craig

luc-github, 2016-07-16:
ok, may I know the purpose of your pt100? 

replace current thermistor?

Lennie1972, 2016-07-16:
Yes. I want to change Rhine on my Davinci AIO so that I can print harder filament

Lennie Craig

luc-github, 2016-07-16:
so in this case you just use same cable as current thermistor 
```#define TEMP_0_PIN      13 // ADC channel #, not a PIN #```

Or I misunderstood something ?

Lennie1972, 2016-07-17:
That's correct. I'm trying to change my e3d heat block and thermistor to my pt100. I also purchased the amplifier to the pt100 because I was told that I would it as well.
So now I'm trying to install on my Davinci. Do I need the amplifier?
Lennie Craig

luc-github, 2016-07-17:
yes better to use amplifier but it just need a ground available on extruder and 5v (you need to add)

I guess you already reworked your board as mentioned here :  https://github.com/luc-github/Repetier-Firmware-0.92/issues/104

Lennie1972, 2016-07-17:
No. I haven't done the rework. Looks like I have a bit of studying to do

Lennie Craig

luc-github, 2016-07-22:
@Lennie1972 any update ?

luc-github, 2016-08-08:
Ok if no update I close topic

closed 2016-08-08 by luc-github

Lennie1972, 2016-08-08:
I'm sorry. I've been working a lot and it's been taking up a lot of my time. 

Lennie Craig

Lennie1972, 2016-08-08:
I hadn't reworked my board as you mentioned and in all honesty I was was a bit overwhelmed by the instructions on how to do so.

Lennie Craig

luc-github, 2016-08-10:
I did not said you need to rework the board - just to use same connector(s)

NP happy printing