luc-github/Repetier-Firmware-4-Davinci #171: New alpha version based on 0.92.10 Compatible with the Scanner?

Lehmii, 2016-07-10:
I read here that the repeating 0.92.10 Compatible with the Scanner from the AiO can now be used ...

New alpha version based on 0.92.10 is ready for test here

     Multilangue: globe_with_meridians:
     AiO support for scanner: camera: <------------
     Better auto leveling: tractor:

luc-github, 2016-07-10:
Hardware is supported laser / led / turntable 
https://github.com/luc-github/Repetier-Firmware-0.92/tree/0.92.6-sync
AiO scanner support is present in FW but scanner software support is currently basic, horus is a good candidat, feel free to help

Lehmii, 2016-07-10:
Can I take the scanner software of XYZ?

It surprises me only that Arduino is not supported?
[URL=http://www.directupload.net/file/d/4412/6upctc2f_png.htm][IMG]http://fs5.directupload.net/images/160710/temp/6upctc2f.png[/IMG][/URL]

luc-github, 2016-07-10:
no - XYZ software is not supported - @zeus10 is working on Horus support (https://github.com/bqlabs/horus) which is the candidate currently

luc-github, 2016-07-10:
what are these links ? - cannot see  
what means : 
>It surprises me only that Arduino is not supported?

luc-github, 2016-07-10:
Ok saw the picture :
yes 4 for AiO (WITH NO SCANNER SUPPORT) because support is not 100% because no software so better to warn people who read the readme too fast

but what means :

>It surprises me only that Arduino is not supported?

Lehmii, 2016-07-10:
Thanks for the info, I'm still waiting for the FW.

luc-github, 2016-07-10:
ok so I close topic

closed 2016-07-10 by luc-github