luc-github/Repetier-Firmware-4-Davinci #173: Question: Reset necessary for updating firmware changes or upgrade?

opensourcefan, 2016-07-17:
Sorry, I feel this is too simple of a question but I can't find the answer on any installation page. 

Do I need to perform a reset via the jumpers to flash an update or can I simply flash over top of an already Repetier flashed Da Vinci? I want to make a quick configuration change (re-purpose fan).

luc-github, 2016-07-17:
jumper is only to wipe xyz fw, after no need anymore.

opensourcefan, 2016-07-17:
Thank you very much for the quick reply. Could that be added to the first line of the installation instructions maybe?

opensourcefan, 2016-07-17:
I can submit it if you prefer.

luc-github, 2016-07-17:
I just did on master - please confirm it is clear for you so i will apply on devt also

Thank you

opensourcefan, 2016-07-17:
Yes, that is very clear, thank you.

luc-github, 2016-07-17:
thank you for pointing this out ^_^

opensourcefan, 2016-07-17:
Anytime, thank you for all your hard work.

luc-github, 2016-07-17:
thank you - just do not forget to load fail safe or M502/M500 after update, as EEPROM content may change due to update

opensourcefan, 2016-07-17:
Ah, yes, that can be easily forgotten. ;)

luc-github, 2016-07-22:
I close issue if no more update

closed 2016-07-22 by luc-github