luc-github/Repetier-Firmware-4-Davinci #174: Question: Extruder fan power trigger when original re-purposed.

opensourcefan, 2016-07-22:
I re-purposed my original fan to be layer cooling fan. I'm wiring up a 12v fan to the extruder which will always be on when powered on. I had an idea to use a relay triggered from the led light power that would at least turn the fan off when the machine goes into standby mode, ie the lights turn off. I would rather the fan not be on all the time if possible. 

Does anyone know of a better solution? Doesn't need to be software controlled but would be nice.  

Da Vinci 1.0a, motherboard does NOT have the components for Fan 2 installed.

luc-github, 2016-07-22:
did you checked:

https://github.com/luc-github/Repetier-Firmware-0.92/issues/165  ?

opensourcefan, 2016-07-22:
Thank you, yes I have seen that but was hoping not to mess with the motherboard if I don't have to. I might end up doing that though.

luc-github, 2016-07-22:
another solution could be to use the cables of cartridge eeprom but you need 5v fan

opensourcefan, 2016-07-22:
Hey that's a good idea, I don't mind using a 5v fan. Which cables are those exactly?

luc-github, 2016-07-22:
be careful no to do a short, there 3 cables, 1gnd, 1vcc, 1pin, check pinout.xls to see which pin

eeprom connector is easy to remove, or you can connect directly use connector on main board

opensourcefan, 2016-07-22:
I'll look at the motherboard and see what blank spots I have. I think I'd rather add the missing fan 2 components if I'm going that deep into the board. It's important though that whatever I connect to turns the fan off when in standby mode because I can easily connect to a always on power rail, 5 or 12v.

luc-github, 2016-07-22:
well  the EEPROM reader is no use do it is free:
on my DUO on screw for the 2 cartridges:
[image]

and the 4 cables : GND VCC pin 1 pin 2 have a connector (the grey cables)
[image]

So normally with that you do not need to mess up with your board - you even can control a 12v circuit or anything as you have a free pin

opensourcefan, 2016-07-22:
That's excellent, thank you very much for the pictures, I understand now. I totally forgot about that thing down there as I never used xyz filament. I'm about to pull it all apart now, we'll see what happens.

opensourcefan, 2016-07-23:
So for the time being I hooked up to the 12 volts from the power supply. The previous owner of my machine cut that and the bed heater connectors off and put them back together so poorly that after I removed the black tape the connectors fell off. So I took advantage and added my 12v connection there. I used PC power supply fan connectors so I can piggy back more connections (low current stuff) if needed. The previous owner also broke a few of the motherboard connectors so when I get those I'll most likely add the fan 2 components. 

I do like the idea of using the unused eeprom cartridge line and will keep that in mind for any 5v mods that I'll be doing in the future. 

[image]

wahooper1, 2016-07-23:
I'm hoping to do the same thing. Could you point me in the right direction as to the pin numbers and how to go about programming that?

opensourcefan, 2016-07-23:
There's a few different ideas here, not quite sure which one you would like to do.

I re-flashed this firmware with re-purpose fan enabled, moved that fan (extruder fan) to my layer cooling  fan duct and added a always on fan back into the extruder fan location.

wahooper1, 2016-07-23:
I currently have my printer with the re-purpose turned off. I just finished printing a layer cooling duct, and i have a 5v fan on the way here from amazon. 

So i'm going to turn on re-purpose and use the original extruder fan pins as the layer cooling. 
But i need help programming the new extruder fan to run off of the EEPROM reader (pin 115?)

Sorry for my poor wording. I'm not really sure how to explain.

opensourcefan, 2016-07-23:
No worries at all, I understand now. I currently just have my fan on the 12v rail and haven't done the eeprom reader mod yet. Under the code tab you find the pinout xls file. It should give you the pinouts. If not it won't be too hard to ring them out with a multimeter.

wahooper1, 2016-07-23:
I'm just not sure where to add it in the arduino files.

opensourcefan, 2016-07-23:
I don't think you need to, it's simply just using the power. Luc may be able to correct me but I don't think you will actually be able to "control" your fan, just power it.

wahooper1, 2016-07-23:
Correct. I dont want to control the extruder fan like a layer fan. I just want it to come on like it should when the extruder comes on and go off when the extruder turns off. I don't want it on all the time like how you have yours.

wahooper1, 2016-07-23:
I'm thinking if I change

 #define ORIG_FAN_PIN        25
#define ORIG_FAN2_PIN       115 <---pin from EEPROM

and

#if DAVINCI==3
#define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
#else
#define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN <----EEPROM pin
#endif

this should in theory make it work like im saying.

opensourcefan, 2016-07-23:
You are way ahead of me, give it a shot but please report back with the results.

wahooper1, 2016-07-23:
I would love for luc to chime in and tell me if im waayy off. I'm use to programming lasers not 3d printers. hahaha

opensourcefan, 2016-07-23:
This could be an excellent mod if you are in the right track. And if it was built in we'd be able to turn it on and off just like the fan re-purpose.

wahooper1, 2016-07-23:
Honestly hoping i'm on the right track.
Also changed

#if REPURPOSE_FAN_TO_COOL_EXTRUSIONS==1
#define FEATURE_FAN_CONTROL 1
 #define FAN_PIN ORIG_FAN_PIN
  #if DAVINCI==2 || DAVINCI==3
      #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
      #define EXT1_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN
    #else //DaVinci 1.0
        #define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN2_PIN < ------defining EEPROM as extruder cooler

opensourcefan, 2016-07-24:
Just printed a Marvin with a perfect keychain loop, love the automated cooling now. I'm not bothered by the fan being on all the time, it's about the same noise level as the power supply fan so can't really notice it.

wahooper1, 2016-07-24:
So it works but i'm only getting 3.3 volts

opensourcefan, 2016-07-24:
So what was you final code change then? Curious why it's only 3.3 volts. That's pretty deliberate voltage number, is it possible that VCC is only 3.3 or did you confirm the 5v before hand?

wahooper1, 2016-07-24:
I know that the fan is supposed to get 5v but what i dont know is if that pin (115) can supply 5v or if it is limited to 3.3v

opensourcefan, 2016-07-24:
Oh and does your new code simply tell it to power that specific pin instead of the original? Meaning the fan still turns on a 50c and goes off after cooling to 50c. That's what my 1.0A did out of the box.

wahooper1, 2016-07-24:
Yes. the pin goes 3.3v+ above 50C and turns off below 50C

opensourcefan, 2016-07-24:
Well done with the code then, that's perfect. 

It's logical that when reading for the eeprom it could use 3.3, most of my eeprom devices are 5 though. Pretty sure Luc thought it was 5v, I haven't checked myself.

wahooper1, 2016-07-24:
I'm sure somewhere (that i cant find) in the arduino code we could set that pin to 5v.

opensourcefan, 2016-07-24:
That would be cool. Thanks for putting your skills to this.

wahooper1, 2016-07-24:
I'll be happy to hear from luc on this. Will post the code changes as soon as i get it completed and luc says its not going to hurt anything.

wahooper1, 2016-07-24:
BTW with the changes ive made so far i AM getting a random reset.

opensourcefan, 2016-07-24:
I'm sure Luc will know what's happening.

luc-github, 2016-07-24:
Hi sorry just morning here - Ok about 3.3v this is the normal current output for the pin - as due is not an arduino so no 5v but 3.3v -  so should be ok for 5v fan no ? or you can add a circuit to control a 12v fan like here: http://www.instructables.com/id/Use-Arduino-with-TIP120-transistor-to-control-moto/

about reset, seems happened to some people and solved, see here  : https://github.com/luc-github/Repetier-Firmware-0.92/issues/152

opensourcefan, 2016-07-24:
If 3.3 is all were going to get from that I would rather use the 3.3 to trigger a relay to power a 12v fan. I think the extruder fan shouldn't be underpowered especially for PLA on the Da Vinci. 12v fans are also easy to find and come in all different types.

wahooper1, 2016-07-24:
the circuit you linked to has a  pwm 5v pin going to the base of the transistor. We dont have that luxury. Any idea how to make this work?

opensourcefan, 2016-07-25:
Well wahooper1, if you can figure out the code minus the resets, one day I'll use the 3.3 volts with a boost converter to my 12v extruder fan. Being that the fan only draws 70ma (100ma on startup) it'll be no problem at all for the boost converter. It's only one extra component, not a big deal IMO. 

Something similar to this, I have a few laying around for such an occasion. http://www.ebay.com/itm/1PCS-XL6009-DC-DC-Adjustable-Step-up-Boost-Power-Converter-Module-Replace-LM2577/301410297753?_trksid=p2047675.c100005.m1851&_trkparms=aid%3D222007%26algo%3DSIC.MBE%26ao%3D1%26asc%3D37429%26meid%3D3d6a0bdad35c4319b3f8306abe793791%26pid%3D100005%26rk%3D2%26rkt%3D6%26sd%3D291664700368

luc-github, 2016-08-08:
Any update on this topic ?

luc-github, 2016-08-14:
If no update I close topic

closed 2016-08-14 by luc-github