luc-github/Repetier-Firmware-4-Davinci #175: DaVinci 1.0 1st Gen board, black bars and flashing Light after upgrade

613codyrex, 2016-08-08:
My DaVinci, after multiple attempts to try to get the new .92 firmware to work has failed. The printer after the flash with arduino (using all supported versions) to .92 will only show black bars, and the internal LEDs will turn off and on.

I have attempted to try to switch laptops, trying USB 2.0 instead of 3.0, trying to use different arduino versions but I continue to get this issue. The only thing to have worked has been using .bin files of .91 to upload through the use of command Prompt. 

I can add more detail (i have a video of the printer doing this) if you need it.

luc-github, 2016-08-08:
what version of ide do you use ?
did you check the FAQ ?
https://github.com/luc-github/Repetier-Firmware-0.92/issues/172

613codyrex, 2016-08-09:
I have used 1.5.8, 1.6.5 and 1.6.8. 

I have made sure and done the whole process multiple times without any luck. I've attempted to do this on 3 Win 10 laptops and a IMac but no luck unfortunately.

luc-github, 2016-08-09:
Well printer is Ok as you can flash the 0.91
and black bars are typical of bad variant.cpp, also depending of IDE version location on computer is different

Can you give me the path of the variant.cpp you replaced ? and version of ide you now use ?

613codyrex, 2016-08-14:
Sorry for the long wait. I forgot. 

Anyway the locations of the variant.cpp for the IDE version 1.5.8 in the (program files x86) hardware/Arduino/Sam/variants/arduino_due_X/variant.cpp but when I do this I copy the whole hardware file from the GitHub download and just paste it to Arduino location. Could this be the issue? 

Thanks for the reminder.

luc-github, 2016-08-14:
yes could be, did you verified the file variant.cpp has been properly replaced ?

613codyrex, 2016-08-15:
So i tried to individually move the USB_core.cpp and the variants.cpp to the proper places as defined in the instructions on the main page and not just copying the whole hardware file and pasting it into the arduino folder , but it did not work and its doing the same thing.

this was a free and clean install of arduino 1.5.8 and a clean download of repetier .92 from the github.

[IMG_9219.zip](https://github.com/luc-github/Repetier-Firmware-0.92/files/417593/IMG_9219.zip) This is what it has been doing for me right now (sorry for the weird video in a zip file)

luc-github, 2016-08-15:
Ok let's do another way 
Use my developement branch that use ide 1.6.9

install IDE 1.6.9    
then go to board manager and install Due 1.6.8   
[image]   
then go to : `C:\Users\<your username>\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x`    
you should see the variant.cpp to replace, if you do not see it, you are not in r1ght place   
[image]

then use the one present on git version in developement it has only one so cannot do mistake 
(https://github.com/luc-github/Repetier-Firmware-0.92/tree/devt/src/ArduinoDUE/AdditionalArduinoFiles/Arduino%20-%201.6.9%20-Due%201.6.8/Arduino15/packages/arduino/hardware/sam/1.6.8/variants/arduino_due_x)   
to very it is properly replaced - open it and check it has this :
[image]

then  edit Configuration.h to be sure it match your printer 
```
#define DAVINCI 1// "0" if not DAVINCI, "1" For DAVINCI 1.0, "2" For DAVINCI 2.0 with 1 FAN, "3" For DAVINCI 2.0 with 2 FAN, 4 for AiO (WITH NO SCANNER SUPPORT)
#define MODEL  0//"0" for first generation (jumper JP1 to reset ) , "1" for new generation   (jumper J37 to reset)
```

then compile and flash, when done restart the printer and reset EEPROM

613codyrex, 2016-08-15:
Great news. The firmware is working on my printer right now, except its now in svenska and the display was weird after the flash.

Thank you for bearing with me and helping me with this.

closed 2016-08-16 by luc-github

goldx5, 2016-09-01:
I use 1.6.9 DUE 1.6.8
this is my error message

done in 5.137 seconds

Verify 205956 bytes of flash

[                              ] 0% (0/805 pages)
[==                            ] 9% (80/805 pages)
[=====                         ] 19% (160/805 pages)
[========                      ] 29% (240/805 pages)
[===========                   ] 39% (320/805 pages)
[==============                ] 49% (400/805 pages)
[=================             ] 59% (480/805 pages)
[====================          ] 69% (560/805 pages)
[=======================       ] 79% (640/805 pages)
[==========================    ] 89% (720/805 pages)
[============================= ] 99% (800/805 pages)
[==============================] 100% (805/805 pages)
Verify failed
Page errors: 805
Byte errors: 203350

goldx5, 2016-09-02:
Why Verify failed?

Verify failed
Page errors: 800
Byte errors: 202054

goldx5, 2016-09-02:
Why Verify failed?
Verify failed
Page errors: 809
Byte errors: 204313

Wingeddonkey, 2016-09-02:
Having a similar issue. 
Had an older version of .92 installed, and after a move, pulling my printer out its been heating element disc. and X travel issues, so I wanted to start from ground up. 

Getting a seemingly succesful flash with Verify Successful. 
Seeing the printer on com ports as a Bossa program port
Using 1.6.11 then 1.6.9, installed DUE (native USB), then after seeing this replaced variant and USB core. 
During install get a "Hardware disconnected tone" but no further response from printer. 

Looking for other ideas, (Noticed you changed the variant files ~4 days prior to this, maybe an issue there?)
Going to go back and try my old 1.5.X version I used originally.

Wingeddonkey, 2016-09-02:
Successful reinstall of 1.5.8 version, took a reboot for it to display properly. Going to try the newer version again, and if no luck stick with my 1.5.8.

Wingeddonkey, 2016-09-02:
Successful 1.6.11 install. Not sure what the final result to fix the issue was. But up and running again.

goldx5, 2016-09-02:
Sorry my English not good
Arduino: 1.5.8 (Windows 7), Board: "Arduino Due (Native USB Port)"

In file included from Repetier.h:562:0,
                 from BedLeveling.cpp:107:
HAL.h: In static member function 'static void HAL::serialSetBaudrate(long int)':
HAL.h:700:14: error: 'class UARTClass' has no member named 'setInterruptPriority'
       Serial.setInterruptPriority(1);
              ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

goldx5, 2016-09-02:
"Extruder.cpp" contains unrecognized characters. If this code was created with an older version of Arduino, you may need to use Tools -> Fix Encoding & Reload to update the sketch to use UTF-8 encoding. If not, you may need to delete the bad characters to get rid of this warning.

Where can I change UTF-8?
thanks.

goldx5, 2016-09-02:
I use 1.6.11 & Due 1.6.8

[==========================    ] 88% (720/809 pages)An error occurred while uploading the sketch

[============================= ] 98% (800/809 pages)
[==============================] 100% (809/809 pages)
Verify failed
Page errors: 809
Byte errors: 204313

goldx5, 2016-09-02:
Oh... I finally know, when I upload, I must to take off the jumper JP1.
I finally successful!!!!!!!!!!!!

alexmurillos, 2016-11-16:
@luc-github please help!
I been trying to put the repeater software on my Da Vinci 1.0A printer. I have flashed already, and I am having trouble uploading the code in arduino. BTW, I'm using the code from GitHub (https://github.com/luc-github/Repetier-Firmware-0.92).
First I tried using the Arduino IDE 1.5.8, but there was a compilation problem.

error: 'class UARTClass' has no member named 'setInterruptPriority'
Serial.setInterruptPriority(1)

So I upgraded the version of the IDE to 1.6.9, 1.6.12, and I compiled without any errors, but now the error appears while uploading, always in 69%. Any ideas, I'm stuck with my printer's LCD two bars, right after flashing it.

I did put the files as you explained before in this forum. I do not know what else to do!!!!

luc-github, 2016-11-16:
This issue is closed please open new issue if necessary 

did you check the FAQ ? https://github.com/luc-github/Repetier-Firmware-0.92/issues?utf8=%E2%9C%93&q=is%3Aclosed+label%3AFAQ+

and especially : 
Compilation Error: https://github.com/luc-github/Repetier-Firmware-0.92/issues/201
Verification Failed: https://github.com/luc-github/Repetier-Firmware-0.92/issues/185
Black bars on LCD after flashing : https://github.com/luc-github/Repetier-Firmware-0.92/issues/172