luc-github/Repetier-Firmware-4-Davinci #176: Newbie help needed please!

juri74, 2016-08-12:
hello all, i decided to update my davinci 1.0 firmware to repetier.
i do not know arduino very well, i had version 1.5.8 so i uninstalled it and installed version 1.6.5r5 so i started doing the instaall procedure point by point.
unfortunately i had to stop at point 4:
"Update arduino files (variants.cpp and USBCore.cpp) with the one(s) present in src\ArduinoDUE\AdditionalArduinoFiles\1.5.8. or in src\ArduinoDUE\AdditionalArduinoFiles\1.6.5 according your IDE version
NOTE: You do not need to compile arduino from source these files are in the arduino directory structure. On Mac you will need to right click on the Arduino.app to Show Package Contents."

i do not know where to put such files, i searchet trough all arduino install directory and i wasn't able to find variants.cpp nor USBCore.cpp.
how can i upgrade the two files?
i found a directory called "variants" that contain a bunch of subdirectories.. but they contains only ".h" files not ".cpp"
please help me!
thanks for any help
Juri

juri74, 2016-08-12:
little update:
i found USBCore.cpp under "C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" (i was searching usbcores.cpp) but the directory "src\ArduinoDUE\AdditionalArduinoFiles\1.6.5" do not exist, only 1.6.9 and it doesn'c contain USBCore.cpp
:(

luc-github, 2016-08-12:
hi, sorry on my phone so answer will be short

go to readme and check the FAQ, have your answer

juri74, 2016-08-12:
hello i completely miss the FAQ section, my fault! i'm now installing Arduino ide 1.6.8 :)

juri74, 2016-08-12:
hello again, i setted up all, shortened jp1 on my davinci 1.0 and completely erased the firmware,
the printer was recognized with another name, it was so fast i wasn't able to read it, but it was configured on com 23
so i proceeded, loaded repetier.ino looked into configuration.h but i leaved unchanged ad then crossed my finger and clicked upload button.
sketch compiled without error, but when uploading started this orange message appeared:
"bossac.exe: extra arguments found
Try 'bossac.exe -h' or 'bossac.exe --help' for more information"
arduino ide still showing "uploading" so i leaved it alone for at least 30 minutes.. still showing uploading...
what i'm doing wrong?

luc-github, 2016-08-12:
which ide version are you using, which due version? which branch of the firmware are you using?
what is your operating system ?

juri74, 2016-08-12:
hello, after removing all, deleting all arduino related files,  i installed ide 1.6.9, due version 1.6.8, and i have windows 7 64 bit.
i copied the variants.cpp to appdata and overwritten the old one, now it compile and upload but after finishing uploading, i get "unknown device connected to usb port" and black bar on display, i read faq and this should be about variant.cpp but i copied it.. now i'm completely lost, with a non working printer :')

juri74, 2016-08-12:
firmware 0.92.0 alpha

luc-github, 2016-08-12:
Blacks bars are the issue of variant.cpp
can you give me the complete path of variant.cpp ?
yes FW is 0.92.3 or 0.92.10, which branch master or devt ?

juri74, 2016-08-12:
C:\Users\Juri\AppData\Roaming\Arduino15\packages\arduino\hardware\sam\1.6.5\variants\arduino_due_x\variants.cpp
i downloaded "repetier-firmware-0.92-dept.zip"

juri74, 2016-08-12:
the path report 1.6.5 but i actually have installed 1.6.8, i do not know why path do not updated
[image]

luc-github, 2016-08-12:
OK you said you installed IDE 1.6.9 so path you give is not correct for 2 reasons wrong App data folder (it is local no roaming, arduino change location time to time) and wrong Due module path (you use 1.6.8 but it show 1.6.5) 
should be :
`C:\Users\<user>\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.8\variants\arduino_due_x`
please check this comment - it describe how to open the right folder:
https://github.com/luc-github/Repetier-Firmware-0.92/issues/114#issuecomment-231949044

juri74, 2016-08-12:
is the firmware "dept" correct? or i have to use another branch, like "testing"?
i checked and yes, the path you gave me is correct and full of shit, deleting and reistalling all :D again
thanks

luc-github, 2016-08-12:
Yes devt is the most updated and multilanguage (well for Italian lot of translations are missing - I had not time for that sorry )  - it is the development one because got latest update

juri74, 2016-08-12:
i heard the music :D
finally it is working :)
i noticed the translation file, let's see if i can contribute ;)
thank you soooooo much!

luc-github, 2016-08-12:
Thank you ^_^ - Happy printing

closed 2016-08-12 by luc-github