luc-github/Repetier-Firmware-4-Davinci #179: Davinci Duo 2.0a Wifi issue

DocMcC, 2016-08-21:
Hi,

When I enable the wifi option, the next time I switch on the printer I get 2 black bars.  I have to remove the SD card, then Repetier boots with SD card error.  put the card back in until next time I boot.

If I switch off the Wifi and save the settings it works fine, can reboot etc.

I noticed once on the ESP9266 web page that it showed an error message "Checksum error" - guess this might have something to do with it.

I tried the DEVT firmware fork and it works fine.

If you need anything further let me know.

W.

luc-github, 2016-08-21:
You mean Devt branch do not have the issue ?

DocMcC, 2016-08-21:
Yes,

Going to reflash with the latest version and see what happens.  Will keep you posted.

W.

luc-github, 2016-08-21:
Devt version has a different virtual EEPROM management - so this could make sense 
I never had issue before on master branch, but I am on devt version for 6 month now, without issue 
this actually would be a good time/opportunity  to integrate devt changes to master branch

DocMcC, 2016-08-21:
Hi,

Also, M112 (Emergency stop) crashes the printer – screen goes all garbled – and kills the eeprom file.

Will try it on the DEVT version tomorrow (printing something currently).

Which is the more upto date version?

Thanks,

W.

luc-github, 2016-08-22:
the devt branch is the more updated

DocMcC, 2016-08-22:
Great,  That is the one I am currently using.

Much appreciated,

W.

luc-github, 2016-08-22:
And you still have EEPROM.bin corruption ?

DocMcC, 2016-08-23:
Hi,

Checked today, using DEVT, no EEPROM corruption and when I hit emergency stop the printer resets so those problems are sorted..

Have noticed the following:

1) the menu system on the printer can be a bit "jumpy" - not sure if this is caused by setting the ESP8266 to 160Mhz or something else but the SD card option comes and goes if you are in the menu and as you scroll down entries appear and disappear.

2) When loading unloading filament it shows a message "press ok to start" when it has heated the extruder but then starts anyway (I think it is the same for unloading).

3) When printing if I set it to wait until temps are stable it heats the bed before heating the extruders (just a niggle but would be nice if they all started heating at the same time).

Otherwise a fantastic piece of work - makes the printer much more useful - Much appreciated,

W.

W.

luc-github, 2016-08-23:
Hi good,
1)No, menu is not related to ESP at all, the SD entrie may disappear if you are doing something in same time on printer, there is an issue for that : https://github.com/luc-github/Repetier-Firmware-0.92/issues/24

2)If sensor is enable then message disappear, this is normal, sensor overide it

3) this part is slicer responsability :smile:  check your slicer settings , in your start code you should use the commands to heat both without wait and add a M116 to wait all is hot before continuing

luc-github, 2016-08-29:
Ok I have no idea why such issue with 0.93.3 but as 0.93.10 does not show the problem, I will close issue as fixed with 0.92.10

closed 2016-08-29 by luc-github