luc-github/Repetier-Firmware-4-Davinci #180: Two black lines possible bricked da Vinci 1.0a

Chromewrecker, 2016-08-25:
I have tried for hours to fix this with no positive results. So here's the story I bought a printer and started the repetier reflash yesterday first attempt everything work saved the eeprom and started the bed calibration got that and shut the printer off had other stuff to do came back to start printing. I turned the printer on there was two black lines and the printed was flashing on and off. The port under device manager showed com9 with a yellow !. So I wiped the printer and tried again with no luck still two black lines and the printer still flashing on and off but now the port says unknown device. I tried a different port and that did not make a difference. I used ardunio 1.5.8 and repetier .092 from gethub and sent the hardware file to update the variant.cpp and the usb code I changed the h configure to 1 and 1 since I have j37 and native port but not sure what the programmer should be set to. Spent 8 hours searching forums last nite with no luck. If it matters serial number starts with 3f10a not sure what update was on the printer to begin with.

The printer curintly shows unknown device and I have tried every programmer setting this is my printer and I am noob thanks

thevisad, 2016-08-25:
Does it actually get to the point where you can flash it? Does the connected device change after you bridge the jumpers? Did you use the board manager and install all the drivers? This is a DUE so the drivers are not installed.by default and you have to use the board manager.

Chromewrecker, 2016-08-25:
I believe it's to the point of flashing per the ardunio it says load seccsefull.when I bridge the jumpers the machine has no interior lights and stops turning its self on and off it is also pretty quite at that time.not sure what the board manager is I am using arduino 1.5.8 and I don't think it is installing all the drivers as the port will now read unknown device. What is board manager like I said I am new to this so I apologize for my lack of knowledge and information thanks for the help

thevisad, 2016-08-25:
Open arduino and go to the menu for selecting the board type. At the top should be an option for boaed manager. Choose this and it will load a gui that will present all of the board drivers. Type DUE and it should show you the drivers module to install.

Chromewrecker, 2016-08-25:
Ok cool now will I have to wipe the printer again and reflash it or is this just for the computer and if I need to reflash the printer what programmer setting should I be using and again thanks for the help

thevisad, 2016-08-25:
Yes you will need to reflash once you have the right drivers installed. All of the settings should be selected by default once you have chosen the board. I am not near my computer, so this is all from memory.

Chromewrecker, 2016-08-25:
And sorry to be dumb but the arduino board is the native use port at the bottom of the list correct? I was using this in the previous attempts

thevisad, 2016-08-25:
I used 1.5.8 to flash originally, but I have all versions installed. I am pretty sure its the native port. It should error out if you have the wrong port.

Chromewrecker, 2016-08-25:
I was using the native port last nite. now question if it worked the first time and then lost the driver when I shut it of and turned it back on would that be a download issue I'm running in to or a issue with the printer or I'm I thinking to deep in this

Chromewrecker, 2016-08-25:
And if I have the arduino duo native port would that mean that I have the arduino due board or no?

thevisad, 2016-08-25:
The drivers appear not to be installed (yellow esclamation point), I assume it actually didnt flash the printer due to this. If you didnt install the xyz software you would have to install the drivers. There are two different drivers, one for the original and one for the due in programming mode (bossa). Thats why I asked if the device in device manager changes when you bridge the jumpers.

Chromewrecker, 2016-08-25:
Ok I understand now I had the xyz ware installed for a couple of days printed spool holders and other stuff getting ready for this update so the printer did function and the drivers were installed now under device manger when the pins are bridged bossa does appear then when I flash it it goes away and the printer turns on and off and the port will also flash showing Unknown device

thevisad, 2016-08-25:
Yeah, sounds like you just need to insert the proper drivers now for it to detect once flashed. It *should* work if the printer switches from bossa. The bothersome aspect is that there are still black lines on the printer after the flash. It could be you just need the right drivers for that aspect.

Chromewrecker, 2016-08-25:
Ok I'll check it out I get how around 4 my time I'll update this wether it works or not thanks for the info. Would the black lines be the driver or a lack of files

thevisad, 2016-08-25:
Drivers could cause it, basically its a failure to properly flash the printer with the firmware.

Chromewrecker, 2016-08-25:
O and I just thought of this if this helps anything I tried to use the 1.6.5 hardware files in the 1.5.8 and the printer interior lights came on and the printer would stay on and the port would say arduino com8 when it was plugged in com6

Chromewrecker, 2016-08-25:
It still had the black lines but it didn't flash on and off

thevisad, 2016-08-25:
Remove all arduino files and use only the supported files provided by the board manager and this repository. You can't really use the 1.6.5 inside the 1.5.8 since there were changes made in the 1.6 series.

luc-github, 2016-08-25:
@Chromewrecker never mix the files version - that is why they are separate 

Please check this : https://github.com/luc-github/Repetier-Firmware-0.92/issues/175#issuecomment-239762985

It should be clear enough to redo all process and it sum up what @thevisad already told you

TGMods, 2016-08-25:
One thing you have to do is update the variants.cpp file. On my PC it is found here: **C:\Users\Sally\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.9\variants\arduino_due_x**

3. Use an arduino IDE supporting arduino DUE, 1.6.9 with Due 1.6.8 module from board manager.
4. Update **variants.cpp** arduino file with the one present in **src\ArduinoDUE\AdditionalArduinoFile**
    according your IDE version

This worked for me.

luc-github, 2016-08-25:
@TGMods your path says DUE 1.6.9 should be 1.6.8 but you wrote you use DUE 1.6.8, I think you have updated your DUE module - I did not test the changes of DUE 1.6.9 yet

TGMods, 2016-08-25:
@luc-github it works with DUE 1.6.9 as that's the version I used and my Davinci 2.0a is working perfectly.

thevisad, 2016-08-25:
@tgmods it may work, but its not tested for deployment or development

TGMods, 2016-08-25:
@thevisad I'm only using it for private, non-commercial use

thevisad, 2016-08-25:
@tgmods oh I completely understand, if it works, it works. However as support/developer @luc-github has to keep things tied to items he has tested. This way we eliminate any item that is not part of a tested process, etc

luc-github, 2016-08-25:
Arduino release a lot and not always time for good testing, IDE 1.6.10 was terrible and 1.6.11 is mostly a revert of changes coming from 1.6.10, so that is why better to stick to 1.6.9 IDE which seems stable.   
For Due I agree 1.6.9 would work fine, but unfortunately  I have no time for deep test for each  release, and even less to collect result for each board so unless there is an identified bug I stick to due 1.6.8 for new people 

Do not misunderstood - if it is fine for you to use 1.6.9 or up I am fine too, and thanks for sharing this - it is good to know - I just try to limit the version number for new comer for first flash to be sure no side effect and no confusion when they do first install

Chromewrecker, 2016-08-25:
I can't seem to find app data under the c drive and user

thevisad, 2016-08-25:
Its a hidden folder, you can either type it in the url bar after your username like (c:\users\username\appdata) or click on view folder options and enable viewing hidden files and folders

https://support.microsoft.com/en-us/help/14201/windows-show-hidden-files

Chromewrecker, 2016-08-25:
When I click on drive then users then my name there are no folders called appdata am I in the wrong spot

luc-github, 2016-08-25:
there is a way to open the folder from arduino ide : https://github.com/luc-github/Repetier-Firmware-0.92/issues/114

Chromewrecker, 2016-08-25:
Ok I found it thanks thevisad now I copy the new files under cores and variants?

Chromewrecker, 2016-08-25:
So I changed the us core and variants and now I have an error compiling the sketch

luc-github, 2016-08-25:
did you followed : https://github.com/luc-github/Repetier-Firmware-0.92/issues/175#issuecomment-239762985 ?

Chromewrecker, 2016-08-25:
Yes and it's says error compiling for board arduino due native Ian port

luc-github, 2016-08-25:
It would help if you share what is the error

Chromewrecker, 2016-08-25:
The error says error compiling for board arduino due (native sub port)

Chromewrecker, 2016-08-25:
Exit status 1

luc-github, 2016-08-25:
it show what kind of error in the log, take screenshot of IDE

Chromewrecker, 2016-08-25:
[image]

Chromewrecker, 2016-08-25:
[image]

Chromewrecker, 2016-08-25:
[image]

luc-github, 2016-08-25:
Ok I guess I know the problem 
may I have the screenshot of top ide ? :smile:

Chromewrecker, 2016-08-25:
[image]

Chromewrecker, 2016-08-25:
this is the very top

Chromewrecker, 2016-08-25:
so what do you think the issue is something easy...

luc-github, 2016-08-25:
Yes
What I can see: 
1- you use Due 1.6.9, not 1.6.8 as recommended - which should work according @TGMods - but I never tested
2 - all errors come from missing definition coming from variant.cpp, so you did not replace using correct variant.cpp or you put in wrong place or you deleted it

Chromewrecker, 2016-08-25:
[image]

Chromewrecker, 2016-08-25:
this is where i put it off the link that was provided for #175

luc-github, 2016-08-25:
yes and you copied a shortcut to variant.cpp not the file itself 
file is 23K and your shortcut is 1k

thevisad, 2016-08-25:
You created a shortcut to the file instead of copying the file to the location. See it says internet shortcut?

luc-github, 2016-08-25:
so actually there is no variant.cpp in your directory

Chromewrecker, 2016-08-25:
ok do i need to down load a different arduino also

Chromewrecker, 2016-08-25:
[image]

Chromewrecker, 2016-08-25:
can how do i geet the file form the website i dont see a download

thevisad, 2016-08-25:
download a zip file of the repository and extract it from the zip file, then copy it

Chromewrecker, 2016-08-25:
the like that is provided in #175 dosent have a zip

Chromewrecker, 2016-08-25:
[image]

Chromewrecker, 2016-08-25:
i once again appologize for the confusion the videos didnt make it seem this difficult

luc-github, 2016-08-25:
it is not, you just did not copy the variant.cpp but copy a shortcut - just copy variant.cpp properly and it will work

luc-github, 2016-08-25:
should be cpp file not internet shortcut when you check files

luc-github, 2016-08-25:
[image]
[image]

[12 more replies]

closed 2016-08-26 by luc-github