luc-github/Repetier-Firmware-4-Davinci #181: Extruder hitting the glass

Legendaire, 2016-08-30:
I broke down and flashed my 1.0A to Rep .92.  After sweating if it would work everything went without a hitch until I tried to print.  Now my nozzle is scrapping across the glass.  The extruder motor is clicking until I move the build plate down.  At that time the extruder works fine.  Obviously the extruder nozzle is too close to the build plate.  How does one fix this?
I have tried RepHost 1.6.2 and Slic3r.  I'm pondering just buying Simplify3D to see if that fixes my issue.

luc-github, 2016-08-30:
you need to redo bed levelling if noozle is too close, use paper and bed screw

Legendaire, 2016-08-31:
I used the leveling which used the probe.  I really, really like this feature of the printer.  Does this not work in this firmware?

luc-github, 2016-08-31:
yes it is, but first you must do a manual bed leveling, autolevel is a fine tuning, references are differente between stock and repetier

luc-github, 2016-09-04:
If no update I close topic

closed 2016-09-04 by luc-github