luc-github/Repetier-Firmware-4-Davinci #182: Unable to upgrade from Repetier 92 to 92.10rc1

d3d3d, 2016-08-31:
Settings used for all attempts to flash 92.10rc1:

Davinci 1.0A (J37) type motherboard.
Merge Arduino 'hardware' folder with one supplied with 92.10rc1 replacing files if needed.
configuration.h settings: 1,1,0 (1=Davinci 1.0, 1=board has (J37), 0=no repurposed fan.
Board: Arduino Due (Native USB Port), also selected correct COM port.
Using OS X Yosemite (10.10.5) Windows XP, Windows 7 Pro and Windows 10 (using multiple PC's or laptops with various USB cables) same result.
Result: After compile, upload and verify success - power off/on leaves the printer in reset mode or sometimes not identifiable by OS with bars on display.

Fails upgrade from 92-master(1.15-10-30_1.1A 0.92Mod) to 92.10rc1. (not using J37)
92.10rc1 compiles, uploads and verifies using Arduino IDE 1.69 with module SAM set to 1.68.
Power off/on leaves the printer in reset mode with bars on display.

Using J37 to reset and upload: 92.10rc1 compiles, uploads and verifies using Arduino IDE 1.69 with module SAM installed and set to 1.68
Power off/on leaves the printer in reset mode or sometimes not identifiable by OS with bars on display.

Note:
Can go back to older 92-master(1.15-10-30_1.1A 0.92Mod) using OS X (10.10.5) with Arduino IDE 1.58Beta. Older 92-master has worked every time - have gone back to it four or five times.

Am I missing something?

d3d3d, 2016-08-31:
Found the answer in:
https://github.com/luc-github/Repetier-Firmware-0.92/issues/114#issuecomment-239762985

This information perhaps should be in the instructions since Arduino 1.5.8 can not be used with latest 92.10master.

Thanks for the great work you are doing luc

closed 2016-08-31 by d3d3d

luc-github, 2016-08-31:
Hi happy you solved your issue - well readme mention to use IDE 1.6.9/DUE 1.6.8 and additional files are only for this combination