luc-github/Repetier-Firmware-4-Davinci #183: Latest version

DocMcC, 2016-08-31:
Hi,

latest version not working well with the DUO 2.0a:

1) Eprom issue that was resolved in DEVT version is still in the new master version
2) Printing - prints raft and then nothing comes out - either extruder if printing at 0.1mm - sometimes might work for a period then stops extruding
3) Fans - not working correctly - left extruder on, right fan, right extruder, right fan!

Have had to revert to the previous main version - where issue 1 exists but other 2 do not..

Any help appreciated,

W.

luc-github, 2016-08-31:
Latest master is actually the Devt branch no change but added italian language - they are identical

DocMcC, 2016-08-31:
Hi,

That is my point - I have had printing problems with the DEVT branch but thought it was me - levelling, temperatures etc but after trying everything I went back to the previous Master and it is printing fine.

With the latest Master version I have the EEprom saving issue I reported before  if I turn on WIFI.

W.

luc-github, 2016-08-31:
Sorry you did not mentioned before - devt has same issue ?
>Fans - not working correctly - left extruder on, right fan, right extruder, right fan!

DocMcC, 2016-08-31:
Yes but I have been battling with the dev version to get a print (as said thought it was how I had the printer setup)

But the big issue is that it is not printing properly, the raft comes out good but the print itself either does not come out at all (jam?) or stops extruding during the print.

From what I can work out is that it might have something to do with the feed speed or temperature of the head not being measured accurately (but this is just a guess).

As said, I have gone back to the previous master and it is printing ok at the moment (touch wood) :)

W.

DocMcC, 2016-09-07:
Hi,  

Found that I have to push the temperature to 260 degrees to get it to print (at 230 it will print the bed ok but anything finer it clogs - with the original firmware it used 190 degrees.  Do I need to use a different extruder temperature table for the Duo 2.0a?

Thanks,

W.

luc-github, 2016-09-08:
No, correct table for 1.0A/2.0A is automaticaly set when you set MODEL to 1

260 is definitely too high - 230 is normal temperature for ABS, you may try to print too fast ? that would explain why you need to heat so much
50 mm/s is the maximum for stock extruder - most of people use 20 to 30

DocMcC, 2016-09-10:
Hi, latest is that since I printed at 260 degrees it is now printing fine at 220 - must have been something in the heads that was "burnt" off at the higher temperature - I have no idea what though.

I am using Simplify 3d as my slicer and reinstalled that so maybe it had some wrong settings - though I did check the speed and it was fine (1800 mm / min).

Anyway it is a mystery as to what was going on.

One other question (though this is about the printer) - I found that the left head was 0.21 mm shorter than the right head so have put in some Shims to level them out.  Is this normal or did I just get a bad printer?

Thanks

luc-github, 2016-09-10:
no it is common, hard to get perfect identical position
On mine I compensate on bed leveling

luc-github, 2016-09-13:
I guess your original issue is due  to dirt in extruder who was burned with your 260 degres

I close issue - please open a new one is needed

closed 2016-09-13 by luc-github